arduino/ArduinoCore-mbed #7: arm_math.h functions not included in core

janjongboom, 2020-06-30:
@facchinm we have the following code which compiles fine against the nRF52 core (Arduino nRF528x Boards (Mbed OS) v1.1.4), but not against Portenta (Arduino mbed-enabled Boards v1.2.0). I think the CMSIS-DSP functions are not compiled into this core...

```
#include <Arduino.h>
#include "arm_math.h"

void setup() {
   float v1[] = { 3.21f, 1.27f };
   float v2[] = { 8.21f, 3.33f };
   float dest[2];
   arm_add_f32(v1, v2, dest, 2);
   Serial.print(dest[0]);
   Serial.print(dest[1]);
   Serial.println();
}

void loop() {
  
}
```

facchinm, 2020-06-30:
The omission is due to this commit (https://github.com/ARMmbed/mbed-os/commit/50455c0aa412ed102b8431553fde3b85420daaea) which removed the `unsupported` folder from mbed tree. I can restore and precompile `cmsis-dsp` but probably it's a better idea to provide it as a downloadable library.
What do you think?

janjongboom, 2020-06-30:
@facchinm Ah... It's weird though, the headers are still there, but the implementation is not. Are you going to be updating the nRF52 core to Mbed OS 6.0.0 as well? If so, we can just ship the source with the library (as we do for non-Arduino targets). Now it's complicated, with some targets having CMSIS-DSP sources, and some not.

facchinm, 2020-06-30:
As soon as this repo is fully validated the idea is to "deprecate" the old one and switch 100% here for nRF52 too.

janjongboom, 2020-06-30:
@facchinm It's a weird choice on the Mbed OS side though, because the arm_math library is still in the core:

```
janjongboom:~/Library/Arduino15/packages/arduino-beta/hardware/mbed/1.2.0/cores $ find . -name "arm_math.h"
./arduino/mbed/cmsis/TARGET_CORTEX_M/arm_math.h
```

And also in Mbed OS 6, but the actual implementation has disappeared.

sabas1080, 2020-09-04:
Hi @janjongboom 

I have the same problem, could you solve it?

Soulkrown, 2020-09-10:
Hi @janjongboom @facchinm 

I have the same problem with https://github.com/DaleGia/nano-33-sense-serial-example, could you solve it?

janjongboom, 2020-10-14:
We've 'solved' it like this:

1. Header that defines whether to load CMSIS-DSP based on Mbed headers, then detects Mbed OS >6 and sets the `EIDSP_LOAD_CMSIS_DSP_SOURCES` macro. https://github.com/edgeimpulse/inferencing-sdk-cpp/blob/67f085eb39033edf80e1ea8e41f5c089b65187e3/dsp/config.hpp#L26
2. On Arduino: guard all CMSIS-DSP .c files with this macro: https://github.com/edgeimpulse/firmware-arduino-nano-33-ble-sense/blob/b4f9e3ffc136c7b3f9ce66bbe16a0d166fdcb38c/src/edge-impulse-sdk/CMSIS/DSP/Source/TransformFunctions/TransformFunctionsF16.c#L1

dbuggz, 2021-05-03:
Still an issue, this time in,  BLESense_dash_sketch on Arduino Create.

> /tmp/082760084/BLESense_dash_sketch/BLESense_dash_sketch.ino:23:10: fatal error: arm_math.h: No such file or directory
> 
> #include <arm_math.h>
> 
> ^~~~~~~~~~~~
> 
> compilation terminated.
> 
> exit status 1

hussain2603, 2021-05-16:
This is still an issue, the arm_math.h file is missing.

facchinm, 2021-05-17:
@janjongboom @hussain2603 @dbuggz we just published https://github.com/arduino-libraries/Arduino_CMSIS-DSP in the library manager; since mbed now doesn't ship anymore `arm_math.h` you can safely add that library (to the `dependency` field of your `library.properties` or just installing it manually) and it will be picked up.
The library is just a repackaging of the official CMSIS-DSP from https://github.com/ARM-software/CMSIS_5