arduino/ArduinoCore-mbed #8: Declaring abs define in Arduino.h breaks compatibility with TensorFlow Lite

janjongboom, 2020-06-30:
Why is `abs` defined in the Arduino.h for this core?

https://github.com/arduino/ArduinoCore-mbed/blob/682a054fb9769066571ab046b200b2a30b18c11a/cores/arduino/Arduino.h#L70

This conflicts with `std::abs` usage f.e. in TensorFlow Lite. I think this has happened before in other cores, as there's a provision in TF already for this:

https://github.com/tensorflow/tensorflow/blob/845adc40a653b311f4df2c273b22de7e175a93a1/tensorflow/lite/type_to_tflitetype.h#L18

But it still required us to patch this undef in at a later point as well when porting EI to Portenta.

facchinm, 2020-06-30:
Removing `abs` override works for me, but I'm wondering which version of TF you are compiling... 2.1.0-alpha works just fine on the Portenta using this core

hpssjellis, 2020-08-03:
Any update on this. I can't get the TensorflowLite Hello_World code to run on the PortentaH7. It compiles and uploads, but then the onboard LED flashes red and destroys it's COM port connections on windows.