arduino/ArduinoCore-mbed #9: ADC and DAC not working

preynolds101, 2020-07-03:
I made a post on the Arduino forum regarding an issue that I cannot get the DAC to work (see [here](https://forum.arduino.cc/index.php?topic=693551.0)).

I also noticed another post that a user is unable to get the ADC to work, which I have also verified (see [here](https://forum.arduino.cc/index.php?topic=693481.0)).

So, both ADC and DAC seem to be not working at the moment...

facchinm, 2020-07-03:
Hi @preynolds101 ,
DAC is not currently supported but it will be very soon; ADCs instead are working fine. Can you provide your sketch so I can test locally?

preynolds101, 2020-07-03:
To keep things as simple as possible for testing purposes, I copied the code from the analogRead page of the Arduino reference pages... Sketch attached. When I run it I just get zeros - for testing I'm connecting the 3.3V output to A0.

[sketch_jul03a.zip](https://github.com/arduino/ArduinoCore-mbed/files/4868384/sketch_jul03a.zip)

facchinm, 2020-07-03:
You are totally right :flushed: This commit is the offending one f0f4a88344b12a316e614e1396a62d3edb1156f8 ; I'll try to see what is going on and report back soon.
Thanks for the heads-up

closed 2020-07-03 by facchinm

facchinm, 2020-07-03:
The master branch is fine now; for sake of completeness, I'm writing down what happened.

During mbed-6.0 merge window a couple of commits slipped through (https://github.com/arduino/mbed-os/commit/538552adead88a1b1b2c573afd22140d93fe96d4 and https://github.com/arduino/mbed-os/commit/b289d5a08f632db1a14d1a9570d29b147a158fbc in particular).
Without that patches ALTC pins (A0 to A3) were ignored and the call to `analogRead()` returned 0. All other analog pins were fine.

So, recompiling the  mbed library on https://github.com/arduino/mbed-os/commit/6db3edc0ef77bddb6281b4d872addd02201117b8 fixed the issue.

hpssjellis, 2020-07-25:
@facchinm  @preynolds101  Can we re open this issue briefly. I think if you check you will find that A5 was also missed. 

I can run every analogRead from A0 to A6 and they all work except A5, it completely kills the serial monitor.

hpssjellis, 2020-07-25:
I have posted on the Arduino Thread  [here](https://forum.arduino.cc/index.php?topic=693481.msg4686648#msg4686648)

ubidefeo, 2020-07-25:
hi @hpssjellis 
I saw your comment in the forum and came here to check it out.
I will test with the Portenta I have, but if you have this issue I doubt J can get a different outcome.
Let's see what @facchinm has to say about it.
Keep you posted.

Also for @facchinm : is analog reference supported? (same Thread in the forum)

facchinm, 2020-07-27:
`analogReference` is not yet supported on the Portenta. 
About A5, I pushed 900f71751729a2e45816da33bee09cada6314ccf to fix the crash and regain the functionality, but it needs some investigation since the chosen mux used to be ok.