arduino/ArduinoCore-mbed #15: WiFiUDP Doesn't Compile

jwestmoreland, 2020-07-13:
To reproduce, just instantiate an instance:

WiFiUDP Udp;

....ino:145:9: error: cannot declare variable 'Udp' to be of abstract type 'arduino::WiFiUDP'
 WiFiUDP Udp;
         ^~~
In file included from C:\...\AppData\Local\Arduino15\packages\arduinobeta\hardware\mbed\1.2.1\libraries\WiFi\src/WiFi.h:32:0,

I see there's a pull request posted as well.

Thanks,
John W.

facchinm, 2020-07-14:
To be fixed by https://github.com/arduino/ArduinoCore-mbed/pull/14

jwestmoreland, 2020-07-14:
I tested that pull request and it didn't work.  It compiles, connects to the server, but I'm not seeing packets yet.

To know for sure I would either need to be able to get the debugger working with the Portenta or have some confirmation multicast is working or both.

Regards,
John

facchinm, 2020-07-14:
To debug any user code you can use `ThreadDebug` library (documentation is coming soon but the example should be a good starting point).
Otherwise, SWD pins are not exposed on MKR header but only on the high density, so it's not super easy to attach a physical debugger without a carrier / expander.
Anyway, if you already managed to wire the correct pins, just take inspiration from "Burn Bootloader" commandline (it should resemble something like this)
```
~/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino13/bin/openocd -d2 -s ~/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino13/share/openocd/scripts/ -f interface/stlink.cfg -f target/stm32h7x_dual_bank.cfg
```
and then attach via gdb.

jwestmoreland, 2020-07-14:
OK Martino - thanks!
John

jwestmoreland, 2020-07-15:
Martino,

I was able to get the command line version of gdb going - and it appears I'm not receiving multicast packets.

Is there a debug mode that you know of for the WiFi I/F that I could put it in to see packet data coming across?

Thanks,
John

jwestmoreland, 2020-07-16:
I may be getting data now - got this:

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000000
R1   : 00010000
R2   : 0000FFFF
R3   : 24010C20
R4   : 2400122C
R5   : 24001225
R6   : 00007375
R7   : 00000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 080573B9
SP   : 2400DB00
LR   : 08040691
PC   : 08040A10
xPSR : 810A0000
PSP  : 2400DA98
MSP  : 2407FF78
CPUID: 411FC271
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000000
AFSR : 00000000
BFAR : 00010000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x8040A10
Error Value: 0x24005598
Current Thread: main Id: 0x2400DB4C Entry: 0x8055EFD StackSize: 0x8000 StackMem: 0x24005B28 SP: 0x2400DB00
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=PORTENTA_H7_M7
-- MbedOS Error Info --

Plus red LED blinky.

sbhklr, 2020-07-20:
Hi @jwestmoreland ! Would you mind sharing your sketch so we can try to reproduce?

sbhklr, 2020-07-20:
@jwestmoreland I've pushed another commit to https://github.com/arduino/ArduinoCore-mbed/pull/14 which might fix your issue. Can you give it another try?

jwestmoreland, 2020-07-20:
Hello Sebastian,

OK - I'll give this a try and let you know.

Thanks,
John

jwestmoreland, 2020-07-20:
Hello Sebastian,

Yes - it works - here's some data that came through:

Connecting to THEMIS_WS2

WiFi connected
IP address:
192.168.4.3

Starting connection to server...
connected to server
ATf2=(10)AT83=(0,0,EFGH9876)ATd0=()AT27=(0)AT29=(16,32,5,5,1)AT20=()ATd0=()ATef=(96)AT81=(0,3,Time:)ATef=(111)AT81=(1,0, HH:MM:SS)ATef=(3)AT81=(2,0,GPS Atomic)AT81=(3,0,   Time)ATef=(111)AT81=(1,0, 03:55:30)AT81=(1,0, 03:55:31)AT81=(1,0, 03:55:32)AT81=(1,0, 03:55:33)AT81=(1,0, 03:55:34)AT81=(1,0, 03:55:35)AT81=(1,0, 03:55:36)AT81=(1,0, 03:55:37)AT81=(1,0, 03:55:38)AT81=(1,0, 03:55:39)AT81=(1,0, 03:55:40)AT81=(1,0, 03:55:41)AT81=(1,0, 03:55:42)AT81=(1,0, 03:55:43)AT81=(1,0, 03:55:44)AT81=(1,0, 03:55:45)AT81=(1,0, 03:55:46)AT81=(1,0, 03:55:47)AT81=(1,0, 03:55:48)AT81=(1,0, 03:55:49)AT81=(1,0, 03:55:50)AT81=(1,0, 03:55:51)AT81=(1,0, 03:55:52)AT81=(1,0, 03:55:53)AT81=(1,0, 03:55:54)AT81=(1,0, 03:55:55)AT81=(1,0, 03:55:56)A

Good job!

Thanks!
John

sbhklr, 2020-07-20:
@jwestmoreland Great, thanks for testing!

jwestmoreland, 2020-07-20:
Sebastian,

No problem!

Best Regards,
John

closed 2020-07-22 by facchinm

jwestmoreland, 2020-07-22:
Thanks Sebastian and Martino!