arduino/ArduinoCore-mbed #16: MKRRGB Doesn't Compile

jwestmoreland, 2020-07-13:
To Reproduce:

In an portenta_m7 sketch:

#include <SPI.h>
#include <ArduinoGraphics.h>
#include <Arduino_MKRRGB.h>

Error Messages:
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:25:29: error: 'sercom0' was not declared in this scope
 static SPIClass SPI_MATRIX(&sercom0, A3, A4, A3, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_0);
                             ^~~~~~~
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:25:29: note: suggested alternative: 'strcoll'
 static SPIClass SPI_MATRIX(&sercom0, A3, A4, A3, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_0);
                             ^~~~~~~
                             strcoll
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:25:50: error: 'SPI_PAD_0_SCK_1' was not declared in this scope
 static SPIClass SPI_MATRIX(&sercom0, A3, A4, A3, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_0);
                                                  ^~~~~~~~~~~~~~~
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:25:50: note: suggested alternative: 'SPI_CFG2_SP_1'
 static SPIClass SPI_MATRIX(&sercom0, A3, A4, A3, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_0);
                                                  ^~~~~~~~~~~~~~~
                                                  SPI_CFG2_SP_1
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:25:67: error: 'SERCOM_RX_PAD_0' was not declared in this scope
 static SPIClass SPI_MATRIX(&sercom0, A3, A4, A3, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_0);
                                                                   ^~~~~~~~~~~~~~~
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp: In member function 'virtual int RGBMatrixClass::begin()':
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:74:21: error: 'PIO_SERCOM_ALT' was not declared in this scope
   pinPeripheral(A3, PIO_SERCOM_ALT);
                     ^~~~~~~~~~~~~~
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:74:21: note: suggested alternative: 'PIN_SERIAL_TX'
   pinPeripheral(A3, PIO_SERCOM_ALT);
                     ^~~~~~~~~~~~~~
                     PIN_SERIAL_TX
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:74:3: error: 'pinPeripheral' was not declared in this scope
   pinPeripheral(A3, PIO_SERCOM_ALT);
   ^~~~~~~~~~~~~
\Arduino-Pro-Sketch\libraries\Arduino_MKRRGB\src\MKRRGBMatrix.cpp:74:3: note: suggested alternative: 'pinmap_peripheral'
   pinPeripheral(A3, PIO_SERCOM_ALT);
   ^~~~~~~~~~~~~
   pinmap_peripheral
Compilation error: Error: 2 UNKNOWN: exit status 1

Regards,
John W.

facchinm, 2020-07-14:
`Arduino_MKRRGB` library only targets SAMD boards (MKR) and the SPI pins it uses (A3 and A4) are not SPI capable in PortentaH7. We'll check for a solution but it's not sure we can find one that doesn't involve moving the pins.

jwestmoreland, 2020-07-14:
@facchinm ,

OK - but Potenta has "MKR" compatible pin assignments and MKR compatible connectors, correct?

I've been trying to get that to work - and maybe if this isn't 100% compatible it could explain some issues I've seen with Serial1.

Thanks,
John W.

facchinm, 2020-07-14:
The MKR connector are compatible on the 1st level of functionality (SPI, I2C, Serial, analog and gpios), not on the additional mux functionality that depend on the particular microcontroller in use.
Which kind of issues are you experiencing on Serial1?

jwestmoreland, 2020-07-14:
@facchinm ,

From the ST datasheet - it looks like A3 can be configured as a SPI output:

[image]

For serial1 - I have a sketch where this worked fine:

// command to write string to Lumex LDM-6432 display
static void Write_AT_Command(char *string)
{
#ifdef USING_LUMEX_DISPLAY
  Serial1.print(string);
  while (Serial1.read() != 'E') {}
  delay(2);
#endif

Now the function looks like this to get it not to time out or get hung:
#define SERIAL_CTR_TIME_OUT 100

#ifdef USING_LUMEX_DISPLAY
// command to write string to Lumex LDM-6432 display
static void Write_AT_Command(char *string)
{
static int counter_lum = 0;
 //    Serial1.flush();
 //  (Serial1.available()) {
do {
    counter_lum++;
    delay(1);
  } while ( !(Serial1.available()) && ( counter_lum < SERIAL_CTR_TIME_OUT) );
counter_lum = 0;

 //   Serial1.write(string);
 //   Serial1.read();
    Serial1.print(string);
 // while (Serial1.read() != 'E') {

do {
    counter_lum++;
    delay(1);
  } while ( ( Serial1.read() != 'E') && ( counter_lum < SERIAL_CTR_TIME_OUT) );
// counter = 0;
//     delay(1);
  delay(1);
 // }

}
#endif

At least I had to do that to make any progress - Serial1 was getting hung.  This worked fine with the previous MKR boards - Vidor, Nano 33 IoT, WiFi 1010.

Regards,
John

facchinm, 2020-07-14:
None of the pins wired to the RGD shield are capable of outputting an SPI clock, which is needed for the functionality.
About Serial1 I probably need a complete and minimal sketch demonstrating the issue.

facchinm, 2020-07-14:
This sketch works fine in my configuration
```
#define SERIAL_CTR_TIME_OUT 100
#define USING_LUMEX_DISPLAY

// command to write string to Lumex LDM-6432 display
static void Write_AT_Command(char *string)
{
  static int counter_lum = 0;
  do {
    counter_lum++;
    delay(1);
  } while ( !(Serial1.available()) && ( counter_lum < SERIAL_CTR_TIME_OUT) );
  counter_lum = 0;

  Serial1.print(string);

  do {
    counter_lum++;
    delay(1);
  } while ( ( Serial1.read() != 'E') && ( counter_lum < SERIAL_CTR_TIME_OUT) );
  delay(1);
  Serial1.println(counter_lum);
}

void setup() {
  // put your setup code here, to run once:
  Serial1.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Write_AT_Command("test");
}
```

It prints
```
test100
test100
test100
test100
```
if I press any key except 'E' and
```
test46
test3
test45
```
if I keep pressing it.
Is it the expected output?

jwestmoreland, 2020-07-14:
@facchinm ,

Do you happen to have one of these by any chance:?
[https://www.lumex.com/ezDisplay-Dot-Matrix-Indoor.html](url)

I can provide the demo sketch for that since I'm communicating with one of those.

Regards,
John

jwestmoreland, 2020-07-14:
OK - I'll run your sketch and report back then.

For the LDM-6432, it will send an 'E' command when a string command is completed; just so that's clear.

jwestmoreland, 2020-07-14:
Does this work OK for you:

// command to write string to Lumex LDM-6432 display
static void Write_AT_Command(char *string)
{
#ifdef USING_LUMEX_DISPLAY
  Serial1.print(string);
  while (Serial1.read() != 'E') {}
  delay(2);
#endif

Thanks,
John

facchinm, 2020-07-14:
> Do you happen to have one of these by any chance:?
https://www.lumex.com/ezDisplay-Dot-Matrix-Indoor.html

Unfortunately not; but using the second function you sent me I get a print of "test" every time I press 'E' (which is the expected behavior reading the sources)

jwestmoreland, 2020-07-14:
@facchinm ,

OK - I hope something isn't going on electrically so to speak with the Portenta I have on Serial1.

I'm going to hook up an external FTDI/USB I/F to Serial1 and see if I can duplicate your results.
While you're here if you don't mind - are there any updates on when this will be available:
[https://www.arduino.cc/pro/hardware/product/portenta-carrier](url) ?

It may take a little while to get back - I need to build a little harness to get a Serial1 terminal going.

Thanks!
John

facchinm, 2020-07-14:
Yup, I'm testing with an FTDI, so the issue could be electrical.
About the carrier we are completing the validation; as soon as it's ok it will probably take another couple of months before general availability (but there could be a beta test phase if you are interested :wink: )

jwestmoreland, 2020-07-14:
Definitely interested in the beta phase...

Will let you know if I see something electrically with my board - sure hope not but I could have done something maybe...

Thanks!
John

jwestmoreland, 2020-07-14:
Martino,

Ah, is Serial1 a designated debug port?  

I'm getting this after hooking up the FTDI I/F - this will definitely goof up the LDM-6432...

Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99866  stat 4
Restart single access loop count 99943  stat 4
Restart single access loop count 99850  stat 4
Scan result: channel=1 signal=-43 ssid= bssid=f2:9f:c2:dd:b6:13
Scan result: channel=1 signal=-69 ssid= bssid=f2:9f:c2:6b:1e:77
Scan result: channel=1 signal=-43 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Scan result: channel=1 signal=-69 ssid= bssid=02:9f:c2:6b:1e:77
Scan result: channel=1 signal=-86 ssid=HP-Print-6f-LaserJet 400 bssid=08:ed:b9:3d:85:6f
Scan result: channel=1 signal=-67 ssid=AJ6BC-ARGUS bssid=12:9f:c2:6b:1e:77
Scan result: channel=1 signal=-42 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Scan result: channel=1 signal=-68 ssid=AJ6BC-ARGUS bssid=12:9f:c2:6b:1e:77
Scan result: channel=1 signal=-54 ssid= bssid=10:33:bf:77:15:dd
Scan result: channel=6 signal=-53 ssid= bssid=10:33:bf:77:15:e1
Restart single access loop count 99849  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99849  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Scan result: channel=6 signal=-81 ssid=AJ6BC-ARGUS bssid=22:9f:c2:83:03:ec
Scan result: channel=1 signal=-57 ssid= bssid=22:9f:c2:6b:1e:77
Scan result: channel=1 signal=-43 ssid= bssid=02:9f:c2:dd:b6:13
Scan result: channel=1 signal=-57 ssid= bssid=10:33:bf:77:15:df
Restart single access loop count 99849  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99849  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99899  stat 4
Restart single access loop count 99850  stat 4
Scan result: channel=6 signal=-40 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Scan result: channel=1 signal=-38 ssid= bssid=f0:9f:c2:dd:b6:13
Scan result: channel=1 signal=-58 ssid= bssid=f0:9f:c2:6b:1e:77
Scan result: channel=1 signal=-77 ssid= bssid=02:9f:c2:83:03:ec
Scan result: channel=1 signal=-40 ssid= bssid=f2:9f:c2:dd:b6:13
Scan result: channel=1 signal=-39 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Scan result: channel=1 signal=-67 ssid=AJ6BC-ARGUS bssid=12:9f:c2:6b:1e:77
Scan result: channel=1 signal=-68 ssid= bssid=f2:9f:c2:6b:1e:77
Scan result: channel=1 signal=-40 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Restart single access loop count 99849  stat 4
Restart single access loop count 99849  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Restart single access loop count 99850  stat 4
Scan result: channel=1 signal=-39 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Scan result: channel=1 signal=-68 ssid= bssid=02:9f:c2:6b:1e:77
Scan result: channel=1 signal=-81 ssid=HP-Print-6f-LaserJet 400 bssid=08:ed:b9:3d:85:6f
Scan result: channel=1 signal=-38 ssid=AJ6BC-ARGUS bssid=12:9f:c2:dd:b6:13
Scan result: channel=1 signal=-80 ssid=AJ6BC-ARGUS bssid=22:9f:c2:83:03:ec
Scan result: channel=1 signal=-69 ssid=AJ6BC-ARGUS bssid=12:9f:c2:6b:1e:77
Scan result: channel=1 signal=-57 ssid= bssid=10:33:bf:77:15:dd
Scan result: channel=6 signal=-57 ssid= bssid=10:33:bf:77:15:e1

Thanks,
John

facchinm, 2020-07-14:
It is, totally my fault not to think that you were using WiFi too.
I'm preparing a PR to remove prints from WiFi driver and will tag you as soon as it's ready!
Sorry for the misunderstanding, now everything is much more clear :slightly_smiling_face:

jwestmoreland, 2020-07-14:
No problem Martino.

Yes - this was at the top:
()WLAN FW download size: 420690 bytes
WLAN FW download time: 127 ms
WLAN MAC Address : C4:AC:59:4A:29:56
WLAN Firmware    : wl0: Sep  5 2019 23:24:33 version 7.45.98.92 (r722362 CY) FWID 01-f7128517
WLAN CLM         : API: 12.2 Data: 9.10.39 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2019-09-05 23:10:00
WHD VERSION      : v1.40.0 : v1.40.0 : GCC 10.1 : 2019-10-02 13:25:35 +0800

Ah, can it be something we can turn on/off with a #define or maybe a compile switch?

It's kindof handy looking at the AP scan and all.

Are we going to be able to make custom builds for the Murata WiFi like the ESP32 BTW???

Thanks!
John

facchinm, 2020-07-14:
Unfortunately the WHD driver for Murata module (that comes with a precompiled and closed source firmware :disappointed: ) is part of mbed buildsystem, so the log can't be enabled and disabled from the sketch side (although we could patch it to expose such functionality).
Anyway, the mbed patch is here https://github.com/arduino/mbed-os/commit/00edf9760d91851f2e49ba68b3577a56123b182e while the compiled version is here https://github.com/arduino/ArduinoCore-mbed/pull/18 .
You should just need the second one; let me know if it fixes the issue with your display

facchinm, 2020-07-14:
I added a way to retrieve the logs and print them on demand; the explanation is here https://github.com/arduino/ArduinoCore-mbed/pull/18#issue-448794179 :wink:

jwestmoreland, 2020-07-14:
Martino,

Using the branch for wifi_nolog - Serial1 seems to be fine now.

Does the debugger work yet with the Portenta?  I guess you have to have a .json file for that like for Visual Code debug, right?

Thanks!
John

jwestmoreland, 2020-07-14:
Martino,

Does this work with the nolog branch or the 'base'?

Logs can be printed by calling

whd_print_logbuffer();
To change default port from Serial1 to another transport (Serial for example), simply declare

REDIRECT_STDOUT_TO(Serial);
at the top of the sketch.

With the nolog branch I'm getting:

In file included from C:\Users\John\AppData\Local\Arduino15\packages\arduinobeta\hardware\mbed\1.2.1\variants\PORTENTA_H7_M7/pins_arduino.h:2:0,
                 from C:\Users\John\AppData\Local\Arduino15\packages\arduinobeta\hardware\mbed\1.2.1\cores\arduino/Arduino.h:86,
                 from c:\temp\.ino:14:
c:\temp\.ino: In function 'void setup()':
C:\Users\John\AppData\Local\Arduino15\packages\arduinobeta\hardware\mbed\1.2.1\cores\arduino/macros.h:44:39: error: 'namespace' definition is not allowed here
 #define REDIRECT_STDOUT_TO(stream)    namespace mbed { \
                                       ^
C:\Users\John\AppData\Local\Arduino15\packages\arduinobeta\hardware\mbed\1.2.1\cores\arduino/macros.h:44:39: note: in definition of macro 'REDIRECT_STDOUT_TO'
 #define REDIRECT_STDOUT_TO(stream)    namespace mbed { \
                                       ^~~~~~~~~
Compilation error: Error: 2 UNKNOWN: exit status 1

Thanks,
John

facchinm, 2020-07-14:
The `REDIRECT_STDOUT_TO` macros should be outside any function, like if it was a global variable

jwestmoreland, 2020-07-14:
Martino,

I've been able to start the debugger using the ThreadDebug lib as in the example/sketch - was able to get to the first breakpoint but then I guess it disconnects for some reason; but this is progress.

Thanks,
John

adamgreen, 2020-07-14:
@jwestmoreland If you want, you can open a separate issue on the ThreadDebug issue and I can try to help solve it.

jwestmoreland, 2020-07-14:
@adamgreen ,

OK Adam:  [https://github.com/arduino/ArduinoCore-mbed/issues/19](url) added.

Thanks,
John

closed 2020-07-23 by facchinm