arduino/ArduinoCore-mbed #17: Portenta Tutorial - MBED v. 1.9.5?

jwestmoreland, 2020-07-14:
Hello,

Please see the attached pic - I only see version 1.2.1 - not 1.9.5 - I'm assuming this was a pre-beta release - but wanted to point it out since it can be a source of confusion when going through the tutorials.

Regards,
John W.
[image]

facchinm, 2020-07-14:
@sbhklr can you check this and create new screenshots?
@jwestmoreland you are right, it was a prerelease.

sbhklr, 2020-07-14:
> @sbhklr can you check this and create new screenshots?
> @jwestmoreland you are right, it was a prerelease.
@facchinm for sure. Can you please assign the GH issue to me?

sbhklr, 2020-07-20:
@facchinm Done. You may close the issue.

closed 2020-07-22 by facchinm

jwestmoreland, 2020-07-22:
Thanks Sebastian and Martino!