arduino/ArduinoCore-mbed #30: Improve console messages while flashing M4 / M7

sbhklr, 2020-07-27:
Two suggestions to improve the console messages when flashing the M4 or M7.

1. The action in the IDE for flashing a board is called "Upload". Hence the success message in the editor window is "Done uploading." Meanwhile the output in the console is "Download done. File downloaded successfully". While technically reasonable I'd suggest to make this consistent.
2. It has been reported by @hpssjellis that accidentally uploading to the wrong core can easily happen every now and then. We might improve the situation a bit by including the core in the upload / download message. E.g. "File uploaded successfully to M7".

@facchinm thoughts?

facchinm, 2020-07-28:
@sbhklr `dfu-util` has no knowledge of M7/M4 partitioning; it considers the flash address space as a whole and we only change the specific address when uploading.

A complete solution to this could be:
* provide some M7/M4 partitioning schemes via a menu
* compile the M4 code with the right offset
* upload a combined M7+M4 image

However, this process involves many changes in the IDE/builder that would be specific for the Portenta.
I'd prefer to keep things simple but I'm open to suggestions
@pnndra

sbhklr, 2020-10-05:
I'm closing this as there is no simple way of solving this at the moment.

closed 2020-10-05 by sbhklr