arduino/ArduinoCore-mbed #31: WHD (WiFi Host Driver) - Can We Update To v1.9x ?

jwestmoreland, 2020-07-27:
Just a suggestion - can we update the WHD version?

I will volunteer to test.

Thanks,

facchinm, 2020-07-28:
Sure, we are going to release the core today; after that the plan is to update to mbed-6.2 and update some dependencies too

jwestmoreland, 2020-07-28:
Sounds great Martino.

Better than trying to patch libmed.a with 

4343WA1-mfgtest_bin.o
4343WA1-mfgtest_clm_blob.o
4343WA1_bin.o
4343WA1_clm_blob.o
v1.90.2.

Plus - some bug fixes since WHD v1.40

Thanks!
John

jwestmoreland, 2020-07-29:
19:14:27.729 -> WLAN FW download size: 421098 bytes
19:14:27.764 -> WLAN FW download time: 127 ms
19:14:27.764 -> WLAN MAC Address : C4:AC:59:4A:29:56
19:14:27.764 -> WLAN Firmware    : wl0: Jan 30 2020 21:41:53 version 7.45.98.95 (r724303 CY) FWID 01-5afc8c1e
19:14:27.764 -> WLAN CLM         : API: 12.2 Data: 9.10.39 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2019-09-05 23:10:00

facchinm, 2020-09-04:
Fixed with https://github.com/arduino/mbed-os/commit/a01a0853d504fb48c2f07ea45c66e9b2878cb427

closed 2020-09-04 by facchinm