arduino/ArduinoCore-mbed #32: Bootloader Build Definitions, Please

jwestmoreland, 2020-07-28:
Can definitions/descriptions be provided for the following:

portentah7_bootloader_mbed_fs.bin *** results in DL2 being solidly lit - OLoD - Orange LED of Doom.
portentah7_bootloader_mbed_fs.elf

portentah7_bootloader_mbed_hs.bin
 portentah7_bootloader_mbed_hs.elf

portentah7_bootloader_mbed_hs_v2.bin  **** works with the Portenta H7 that I have received.
portentah7_bootloader_mbed_hs_v2.elf

Thanks,
John W.

facchinm, 2020-07-28:
Sure, I intended to document them sooner or later :slightly_smiling_face: 

`fs.*` -> for BETA boards without the USB HS phy
`hs.*` -> for BETA boards with the USB HS phy mounted
`hs_v2.*` -> for PRODUCTION boards with the USB HS phy mounted

So the orange led of doom is expected in case you are flashing a production board with a beta firmware.
Instead, v2 firmware boots on both beta and production, but the binary included in this repo is only for the production one (which was factory flashed with it)

jwestmoreland, 2020-07-28:
Thanks Martino!

jwestmoreland, 2020-07-28:
Martino answered - see above.
Thanks!

closed 2020-07-28 by jwestmoreland