arduino/ArduinoCore-mbed #34: mbed: v1.2.2 -

jwestmoreland, 2020-07-30:


facchinm, 2020-07-30:
I think you might have a stray WiFi.cpp from previous experimentation (`wiced_filesystem_mount` comes from https://github.com/arduino/ArduinoCore-mbed/pull/26).
Try cleaning up the folder (by uninstalling and reinstalling 1.2.2) and let me know if it works :wink:

jwestmoreland, 2020-07-30:
Martino,

Is the QSPI update in 1.2.2?  When I downloaded it, I wasn't sure - but again, yes, I probably need to redownload.
I didn't see the PortentaWiFiFirmwareUpdater.ino in the mbed 1.2.2 update that downloaded.

Thanks,
John

jwestmoreland, 2020-07-30:
Also - I should have put in OP - I call whd_print_logbuffer() in my app and it errors there also.

jwestmoreland, 2020-07-30:
Martino,

OK - this is interesting - I have all of the whd_print_logbuffer() calls commented out - but I do have this in my code:
REDIRECT_STDOUT_TO(Serial);

And, I just saw this on com port (Serial) output:

01:31:23.642 -> WLAN MAC Address : C4:AC:59:4A:29:56
01:31:23.642 -> WLAN Firmware    : wl0: Jan 30 2020 21:41:53 version 7.45.98.95 (r724303 CY) FWID 01-5afc8c1e
01:31:23.675 -> WLAN CLM         : API: 12.2 Data: 9.10.39 Compiler: 1.29.4 ClmImport: 1.36.3 Creation: 2020-01-30 21:30:05 
01:31:23.675 -> WHD VERSION      : v1.91.2 : v1.91.2 : GCC 10.1 : 2020-06-25 02:15:47 -0500
01:31:29.388 -> Restart single access loop count 99851  stat 4
01:31:29.872 -> Restart single access loop count 99851  stat 4
01:31:30.388 -> Restart single access loop count 99850  stat 4
01:31:30.873 -> Restart single access loop count 99850  stat 4

And, wifi is working.

Hmmm.

Regards,
John

facchinm, 2020-07-30:
WHD VERSION : v1.91.2 means you are on this https://github.com/arduino/ArduinoCore-mbed/pull/26 probably.
Did you try reinstalling completely the core?

jwestmoreland, 2020-07-30:
Martino,

Yes - I know the QSPI device was written.

And, yes, I did, and whd_print_logbuffer() is AOK in 1.2.2 or at least doesn't fail compiling.

So, when will  #26 be included in 'main download'?

Thanks,
John

facchinm, 2020-07-30:
I'm probably going to merge it today but I have to make sure 1.2.2 is fine before doing so :wink:

closed 2020-07-30 by facchinm

jwestmoreland, 2020-07-30:
OK Martino!
Thanks,
John