arduino/ArduinoCore-mbed #35: mBED runs javascript, what are the Portenta plans?

hpssjellis, 2020-08-03:
Can someone perhaps @janjongboom or @facchinm reply about the possible direction Arduino is taking with Javascript implementation. I am doing fairly well at getting things working. See my github at [https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7)

Mbed runs javascript [https://os.mbed.com/blog/entry/Bringing-JavaScript-to-mbed-OS/](https://os.mbed.com/blog/entry/Bringing-JavaScript-to-mbed-OS/) which would make the most sense. Is that the most plausible direction? Does anyone have any starting perhaps javascript blink code they think might work but it just needs a few tweaks. I am willing to work on this, as I have most of what I need from the Portenta working so far.

janjongboom, 2020-09-14:
@hpssjellis I left Arm >1 year ago and the JerryScript runtime for Mbed has not been updated since. My feeling is that https://github.com/ARMmbed/mbed-js-v2 would be the quickest way forward, it adds JerryScript as a library to any Mbed projects. Not sure if there's a Portenta build target in Mbed OS yet though.

hpssjellis, 2020-09-15:
Thanks @janjongboom for the reply, I will have to wait until someone has a blink example working on the Portenta and then I can test other Javascript abilities. I only have a few boards and I don't want to break the bootloader just yet. I have enough to do for a few months getting TensorflowLIte working on the new board.

janjongboom, 2020-09-15:
@hpssjellis Arduino libraries exported from Edge Impulse actually support Portenta already, including full CMSIS-NN support when using neural networks (using TFLite). Perhaps use it as inspiration?

hpssjellis, 2021-06-22:
I am in contact with the programmer for [Visuino](https://www.visuino.com/) a Visual programming interface who will get the Portenta working eventually. Visuino uses the Delphi platform which is Pascal based, so I will talk to him about if pascal code can be written for the Portenta.

Any other Javascript platforms in the works as of Jun 2021?

hpssjellis, 2023-01-04:
I think I will close this issue. I have never seen a good implementation of Javascript on a microcontroller.

closed 2023-01-04 by hpssjellis