arduino/ArduinoCore-mbed #36: Portenta LED Pattern Decoder Ring

jwestmoreland, 2020-08-04:
Hello,

I was wondering if we could get definitions of the LED patterns on the new Portenta M7 - e.g.:

*)  Red LED - 4 Longs, 4 Shorts - what does that mean?  (Something isn't right, but what's the meaning...)

I suppose this is in the code somewhere - but would be nice to have LED blink guide/decode sheet.

Thanks!
John W.

per1234, 2020-08-13:
Hi @jwestmoreland. It is explained here:
https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/90#issuecomment-654062552
>if the board starts emitting the "morse-like" blinking pattern it means that something has gone very wrong (memory corruption or similar). Double clicking the reset button will bring the board in bootloader mode so you can always use that method as a failsafe.
To discover which bug caused the hang you can hook a serial-to-usb converter to pin TX1 (serial configured as 115200@8n1) and the system will report where the crash happened and what was the root cause.

That issue was from a Nano 33 BLE user, but the same information also applies to the Portenta. On the Portenta, the debug information is output on pin D14.

jwestmoreland, 2020-08-13:
Hello @per1234 ,

OK - I've been connected to D14 (aka TX) - and here's what it dumped:
(didn't know it had debug info there...that was tied to LED blinks anyway.)

............
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 00000000
R1   : 0805B260
R2   : 00000001
R3   : 00000000
R4   : 00000000
R5   : 0000000C
R6   : 00000001
R7   : 00000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : FFFFFFFF
SP   : 2400CFC0
LR   : 080427FB
PC   : 00000000
xPSR : 200F0000
PSP  : 2400CFA0
MSP  : 2407FF78
CPUID: 411FC271
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000002
DFSR : 00000000
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x0
Error Value: 0x24004A70
Current Thread: main Id: 0x2400D024 Entry: 0x8047F69 StackSize: 0x8000 StackMem: 0x24005000 SP: 0x2400CFC0
For more info, visit: [https://mbed.com/s/error?error=0x80FF013D&tgt=PORTENTA_H7_M7](https://mbed.com/s/error?error=0x80FF013D&tgt=PORTENTA_H7_M7)
-- MbedOS Error Info --

Would still be nice to have a guide maybe.

Thanks,
John W.

per1234, 2020-08-13:
>Would still be nice to have a guide maybe.

I agree 1000%. I think it would be helpful for this behavior and the recovery procedure to be documented for all Arduino boards that run Mbed OS.

I didn't at all intend to imply that I considered my previous reply to be sufficient to resolve your request. I started to add that to my previous reply but it's not there so I must have forgotten to click the "Update Comment" button after I wrote the edit.

jwestmoreland, 2020-08-13:
Hello @per1234 ,

No problem - we can start one here I suppose - above decodes to HardFault exception - maybe an enhancement is to add some of the error handling discussed in the Mbed docs to make it more human readable and to be able to read the area in RAM that's dedicated to recording errors on the next reboot; could be handy.

Thanks,
John W.

closed 2021-03-18 by facchinm