arduino/ArduinoCore-mbed #37: Portenta Bootloader Source?

astorks, 2020-08-06:
I am looking into adding FOTA (Firmware Over-The-Air) updates to my Portenta. After a bit of research I found [this wiki page](https://os.mbed.com/teams/Bluetooth-Low-Energy/wiki/Firmware-Over-the-Air-FOTA-Updates?fbclid=IwAR00_ZaVobdpGa4bx8cZC2izzoUlmXP0omwQ07J16AGf-vwzCkZNRoZRVv4), However it looks like the bootloader needs to be modified for FOTA updates to work.
Searching for the Portanta bootloader source yielded no results, so I'm not sure if it just uses a generic mbed bootloader or a custom bootloader.

If it's a custom bootloader is it open source or is there any plans to release the source code?

Thanks,
-Ashton

jwestmoreland, 2020-08-07:
Hello @astorks ,

The Portenta H7 doesn't used the same WiFi I/F as the Nano BLE, it's using a part from Murata:  [https://wireless.murata.com/type-1dx.html](url)

It's based on an SoC from Cypress Semiconductor.

But, I would 2nd the motion on the bootloader code.  I took at look quickly at Arm's Mbed Studio to see if anything was there - nothing obvious jumped out.

The Nano BLE is a target in Arm's Mbed Studio however, not that it helps here at all at the moment.

In related news, I see the NINA has an OTA update on the way - I'm sure a lot will be happy about that.

Regards,
John W.

facchinm, 2020-08-12:
Some initial work on OTA for Portenta is here https://github.com/giulcioffi/ArduinoOTAPortenta .
The APIs are WIP but the concepts are there (multiple storage mediums and filesystem support).
About the bootloader, it will be published soon, no worries :wink:

jwestmoreland, 2020-08-12:
Martino,

OK - that's good news - thanks for letting us know.

Regards,
John W.

jwestmoreland, 2020-09-08:
Hello Martino,

So, this was posted +1 month ago - any ETA on the bootloader instructions?  Also, I have requested this - that I suppose is related: #51 .

Thanks!
John W.

jwestmoreland, 2020-10-01:
Hello,

Will this happen soon: ?

> About the bootloader, it will be published soon, no worries 😉

Thanks,
John W.

flnhst, 2021-04-07:
Is there any update on the release of the bootloader source?

DerDan, 2021-10-04:
Any News ? I need the Bootloader Source

olback, 2021-10-10:
> no worries

It's been over a year. If there are no plans to release the source, just say that. 

If it has been decided to not release the source, a document describing what the bootloader actually does would be neat.

olback, 2022-01-21:
Since my last comment I've started to reverse engineer the bootloader. If you know something about the bootloader that isn't documented [here](https://github.com/olback/h7-bootloader-rev/), please open an issue or a PR so that the H7 starts being useful for environments other than the Arduino IDE.

bm16ton, 2022-03-02:
I tried the portenta in libopencm3 with its limited h7 support and could do can uart blink.leds etc but no usb etc :(  Porting it to tinyusb was cake. But ive had issues even using stms cubecrap code generator, no examples would work. And this lack of understanding bout the bootloader leaves me nervous.Id ditch it entirely except i dont know everything it does and if say voltage mgmt is being done and without it serious issues? My understanding has always been bootloaders should leave no trace they where ever installed during transfer to firmware. But i swear i saw code or a comment that mbed figures out if its fs/hs usb by the bootloader, uhm thats scary. If the firmware crashs and it doesnt reboot but the bootloader starts the leds...how did the bootloader start leds wo a restart? Is it sitting and waiting? Why is it if i start the bootloader with say stty -F /dev/ttyACM0 1200 how come its timeout is totally different? Why doesnt it auto start the firmware after upload or reboot when entered this way? Then add in the HUGE breakout board im forced to buy to use more then a fraction of the io just to find out the board with tons of waisted space has LOTS of pins that go nowhere and do nothing.I understand pins for future add-ons but putting rts cts pins on uart ports that dont have rts and cts is that because they plan on running just the rts and cts pins from a addon board and put them next the uart pins on the breakout? Just dumb. Was arduino bought or change owners in last couple years? Cuz now they have become a company that i hafta read all datasheets and schematics before i buy to get thru the ways they mislead, mis represented or straight up lied in advertisements/posts about the boards. And i used to talk them up and promote them. But even tho a upload script that autodetects/auto reboots into bootloader and uploads is ez to make and at least tinyusb via hal works, im not happy having potentially a lot various hardware setup and monitoring happening that i dont know about or feel safe to delete entirely until i know what if anything it does thats needed..Arduino people can u please tell us that much? If i delete the bootloader that is very much all yours and not mine/ours will i have problems? If not great come and get your bootloader you accidently left on my board, i accidently used it a few times hope thats ok.

facchinm, 2022-04-12:
Bootloader for Portenta H7 is finally live https://blog.arduino.cc/2022/04/12/introducing-the-arduino-secure-boot/ :wink:

closed 2022-04-12 by facchinm