arduino/ArduinoCore-mbed #42: Portenta AP Tutorial:  AP Returns Subnet Mask 192.0.0.0

jwestmoreland, 2020-08-19:
To repeat:

Run the [AP tutorial](https://www.arduino.cc/pro/tutorials/portenta-h7/por-ard-ap) - I've attached my files.

When running the demo - it's returning this:

Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   
   IPv4 Address. . . . . . . . . . . : 192.168.3.3
   Subnet Mask . . . . . . . . . . . : 192.0.0.0
   Default Gateway . . . . . . . . . : 192.168.3.1

For purposes of the demo - wouldn't 255.255.255.0 be a better subnet mask?

Is there a way to change the subnet mask?  (I'll take a look at the mbed stuff to see...).

Just wanted to point this out as I think it's causing some grief for some going through the tutorials.

Regards,
John W.

jwestmoreland, 2020-08-19:
Note, with 1.2.2:

Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   
   IPv4 Address. . . . . . . . . . . : 192.168.3.2
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 192.168.3.1

But Portenta crashes when making either the AP connection or when connecting a web browser to 192.168.3.1.

Regards,
John W.

facchinm, 2020-08-20:
@jwestmoreland the crash should now be gone in `master` branch (which contains the fw on qspi PR).
Netmask is fine too. I'd close this as fixed if you can confirm by testing on master, thanks :wink:

jwestmoreland, 2020-08-20:
Hello Martino,

OK - testing this now - back in a little bit.

Regards,
John

jwestmoreland, 2020-08-20:
Martino,

OK - it appears the situation discussed here:  [#43](https://github.com/arduino/ArduinoCore-mbed/issues/43) is still an issue.

But, I took a look at ping - and even when the web-page to 192.168.3.1 times out - ping appears to still work fine:

C:\Users\john>ping 192.168.3.1

Pinging 192.168.3.1 with 32 bytes of data:
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255

Ping statistics for 192.168.3.1:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 2ms, Maximum = 2ms, Average = 2ms

First time I ran the test - did fine; after that - had issues as discussed in #43.

Let me see if there's a way to disable refresh/reconnect requests in Brave - and I'll post the results.

The latest master commit here does fix the subnet mask issue problem - no question there anymore with 1.2.2 at least.

Regards,
John

jwestmoreland, 2020-08-20:
Martino,

I ran through the firmware update sketch again JIC - I'm attaching that to this - but I'm getting the same result.

I guess there's no way to disable auto refresh in Brave - so I  tried in IE and Chrome but get the same result - if the browser is queued up - it'll make the Portenta either reset or hang if it happens at the same time approximately as the AP connection is made.

Curious issue though is even though HTTP fails ping still works.

Regards,
John
[fw_update.txt](https://github.com/arduino/ArduinoCore-mbed/files/5102258/fw_update.txt)

jwestmoreland, 2020-08-20:
Hello Again Martino,

I applied the patches that are in #44 - and here's a log of what I did:

- Applied patches
- Deleted builds in temp folder(s) so previous compiles would not be used.
- (1) 1st run - ran fine; pinging and running web page worked.
- Disconnected and reset portenta - web page and ping left open/running
- (2) portenta reset when AP connection attempted from PC
- (3) 2nd attempt
- PC connected OK - and ping started back up - 
- web page refreshed
- turned red led on/off
- web page disconnected but ping is still running

from log window (default serial port):
client disconnected
new client
GET /Lr HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windo  <= stopped there

- at about ~30s - tried to refresh web-page - but didn't connect and shows site can't be reached:
- http://192.168.3.1/Lr
- after a few minutes
- tried http://192.168.3.1/ - still didn't reconnect
- 4-5 mins - still no web page reconnect
- web-page appears stuck at this point and isn't reconnecting/running.
- ping is still running fine:

Reply from 192.168.3.1: bytes=32 time=2ms TTL=255
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255
Reply from 192.168.3.1: bytes=32 time=2ms TTL=255

So, the behavior is still similar as reported here: #43

I repeated the process and this is the log file:

Access Point Web Server
Creating access point named: PortentaAccessPoint
SSID: PortentaAccessPoint
IP Address: 192.168.3.1
To see this page in action, open a browser to http://192.168.3.1
new client
GET / HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.135 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

client disconnected
new client
GET /favicon.ico HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.135 Safari/537.36
Accept: image/webp,image/apng,image/*,*/*;q=0.8
Referer: http://192.168.3.1/
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

client disconnected
new client
GET /Hr HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.135 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Referer: http://192.168.3.1/
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

client disconnected
new client
GET /favicon.ico HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.135 Safari/537.36
Accept: image/webp,image/apng,image/*,*/*;q=0.8
Referer: http://192.168.3.1/Hr
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

client disconnected
new client
GET /Lr HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.135 Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
Referer: http://192.168.3.1/Hr
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

client disconnected
new client
GET /favicon.ico HTTP/1.1
Host: 192.168.3.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.135 Safari/537.36
Accept: image/webp,image/ap  <== stopped here

Well, hope this helps in debug/verification.

Regards,
John

jwestmoreland, 2020-08-21:
Martino.

Also - after the above and with the Portenta sitting idle for a about 1 hr, (still pinging) the board reset.

When trying to reconnect - it continually reset:

Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint
Access Point Web Server
Creating access point named: PortentaAccessPoint

I haven't seen that before - maybe the latest updates could have something to do with this?
What do you think about the 3V1 rail current increase?  Could that be a possible issue even though
it was supposed to fix a BLE issue?

Regards,
John

jwestmoreland, 2020-08-21:
Martino,

I could be more help if I had a decent debugger - maybe there's a way to run gdb in 'native' mode right now - note it isn't for lack of trying as proven here:  #19 .

Regards,
John