arduino/ArduinoCore-mbed #45: Portenta:  Excessive Resets By H7 Results In USB I/F Not Being Recognized

jwestmoreland, 2020-08-28:
To All:

First, not sure if this is the 'right' place to report this issue - but - it's somewhat related to the discussion in #42 and I just wanted to post in case someone else is seeing it or maybe someone else has a better workaround than what I've been able to do:

When attempting to debug and building debug builds, sometimes the Portenta goes into a 'reset mode' for lack of a better term - it'll reset to the point the PC no longer recognizes the USB port.

The only effective way I've been able to workaround this issue is to simply shut everything down and reboot.

I tried to uninstall the USB drivers and reinstall - but they will hang seemingly indefinitely when this occurs.

Does anyone else have a better solution and/or has anyone else experienced this?

I realize this could be a USB driver/PC issue but wanted to know if anyone has any better ideas.

Thanks In Advance,
John W.

per1234, 2020-08-28:
When it's in the "reset mode" you describe here, is there any special behavior of the onboard RGB LED (e.g, 4X long, 4X short red blink or green pulse)?

When it's in the "reset mode" you describe here, are you able to put it in bootloader mode by following this procedure?:
1. Press and release the reset button on the Portenta
2. Wait for the green light to come on. It does this almost immediately, but you do need to wait for it.
3. While the green LED is still on (it only stays on for one second so you need to be quick), press and release the reset button again.

You should now see the green LED pulsing to indicate it's in bootloader mode. If not, repeat the process.

If you are able to put it in bootloader mode, and then upload a basic **File > New** sort of sketch, does the behavior you report here go away?

jwestmoreland, 2020-08-28:
Hello @per1234 ,

> When it's in the "reset mode" you describe here, is there any special behavior of the onboard RGB LED (e.g, 4X long, 4X short red blink or green pulse)?

No, it simply hits the reset vector evidently.

> When it's in the "reset mode" you describe here, are you able to put it in bootloader mode by following this procedure?:
>
> 1. Press and release the reset button on the Portenta
> 2. Wait for the green [...]

Yes, that's not an issue.

> You should now see the green LED pulsing to indicate it's in bootloader mode. If not, repeat the process.
> 

> If you are able to put it in bootloader mode, and then upload a basic **File > New** sort of sketch, does the behavior you report here go away?

Yes.

The point of the question is not  that the board can't be put back into bootloader mode.

Has anyone else had the issue where this happens to the point the USB driver no longer recognizes the Portenta?

I thought that was rather clean in the OP, maybe not.

Thanks,
John W.

amber-aw, 2020-12-28:
For the record, just wanted to report I am also seeing this issue (and it is frustrating because a full reboot is not a workable solution). During testing, my Portenta gets an MbedOS error. This causes the usual red led blink. The problem occurs afterwards. I try to do the double-press reset to switch the board to bootloader mode, but instead it either goes back to the red led error signal, or I get a slow green 'glow' pulse. In either case, I am not able to upload a new sketch because the Portenta board is not being recognised on the USB interface. I get a 'mismatched command completion event' in the system log. Unplugging the device's usb connector does not fix the issue. It seems the device misbehaves and causes the usb host controller to fail.