arduino/ArduinoCore-mbed #46: New ThreadDebug Commit - g++ not found in %PATH%

jwestmoreland, 2020-08-28:
Hello All (Martino),

Trying your latest commit with the ThreadDebug lib - getting this:

[image]

Compilation error:  Error: 2 UNKNOWN: exec: "g++": executable file not find in %PATH%.

Probably an easy one to fix - just wanted to report it - I'll see if I can do a workaround since I really want to see this run.

Thanks!
John W.

jwestmoreland, 2020-08-28:
Current 'workarounds':

In boards.txt:

added:
envie_m7_thread_debug.debug.tool=gdb  (1st line)
envie_m7_thread_debug.file=C:/tmp/demo-web-server/build/arduinobeta.mbed.envie_m7_thread_debug/demo-web-server.ino.elf (last line)

Manually created arduinobeta.mbed.envie_m7_thread_debug directory and copied build into that.

Getting this now:

[image]

jwestmoreland, 2020-08-28:
--==--
I think this line may have fixed it in boards.txt:
envie_m7_thread_debug.build.crossprefix=arm-none-eabi-

[boards.txt](https://github.com/arduino/ArduinoCore-mbed/files/5140781/boards.txt)

facchinm, 2020-08-28:
Hi John,
the `envie_m7_thread_debug` entry is marked as `hide` to allow using only its fancy name to distinguish between normal and debug port.
It shouldn't be used as a real target and the IDEs shouldn't allow selecting it (the Java one does, should be backported to the Pro IDE).
Said that, exposing it could make sense (it would allow adding a define that includes ThreadDebug library for example). @pnndra what do you think?

jwestmoreland, 2020-08-28:
Hello Martino,

Does debug happen on a different pid?  Just wondering.

When doing the above - it does make a thread_debug directory:

C:\tmp\demo-web-server\build\arduinobeta.mbed.envie_m7_thread_debug

and it uses that file - demo-web-server.ino.elf in this case for the debug session.

So, I vote for making this a 'feature' so debug builds can be differentiated from regular release (non-debug) types
of builds.

Unfortunately - I'm still getting this: (8/28/20 nightly build):

[image]

And two com ports listed under IDE:

[image]

Thanks!
John W.

kittaakos, 2020-09-11:
> Unfortunately - I'm still getting this: (8/28/20 nightly build):

The IDE2 does not do any magic when it comes to running the debugger but executes:
```
/path/to/arduino-cli "debug" "-p" "the_port" "-b" "arduino-beta:mbed:envie_m7" "--interpreter" "mi2" "/path/to/your/Sketch"
```

Executing the `debug`  command 👆will cause the CLI to run the following command:
```
/directories.data/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin//arm-none-eabi-gdb --interpreter=mi2 -ex 'set pagination off' --baud 115200 -ex 'set target-charset ASCII' -ex 'target remote the_port' /path/to/your/Sketch/build/arduino-beta.mbed.envie_m7/Sketch.ino.elf
```

Unlike with the Zero board, nothing happens:
```
=thread-group-added,id="i1"
~"GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git\n"
~"Copyright (C) 2017 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=x86_64-apple-darwin10 --target=arm-none-eabi\".\nType \"show configuration\" for configuration details."
~"\nFor bug reporting instructions, please see:\n"
~"<http://www.gnu.org/software/gdb/bugs/>.\n"
~"Find the GDB manual and other documentation resources online at:\n<http://www.gnu.org/software/gdb/documentation/>.\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\"...\n"
~"Reading symbols from /Users/akos.kitta/Documents/Arduino/Blink/build/arduino-beta.mbed.envie_m7/Blink.ino.elf..."
~"done.\n"
=cmd-param-changed,param="pagination",value="off"
=cmd-param-changed,param="target-charset",value="ASCII"
```

Can someone please help how to run gdb properly from a terminal using the CLI only, then I can fix the behavior in the IDE2. Thank you!

----

From [here](https://github.com/arduino/ArduinoCore-mbed/issues/19#issuecomment-658491335):
> > Do you have the Portenta working with ThreadLib and the standalone GDB?
> 
> Yes, I have used it that way before. This is the command line I use on macOS and might need to be modified a bit to work in your Windows environment:
> 
> ```
> arm-none-eabi-gdb --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote /dev/tty.usbserial-FTDXR25Z" BinaryFilename.elf
> ```
> 
> I use the **remotelogfile** parameter to allow diagnosing any issues that GDB has in communicating with ThreadDebug.

This does not work for me either.

jwestmoreland, 2020-09-15:
Akos,

In the 0.1.2 IDE - I'm getting this:

GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
.
Find the GDB manual and other documentation resources online at:
.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from c:/tmp2/Display/build/arduinobeta.mbed.envie_m7_thread_debug/Display.ino.elf...
done.
GDB unhandled notify: cmd-param-changed: {"param":"pagination","value":"off"}
GDB unhandled notify: cmd-param-changed: {"param":"target-charset","value":"ASCII"}
Remote debugging using COM8
setup () at c:\tmp2\Display/Display.ino:239
239	{
Will reset on next continue.
Note: automatically using hardware breakpoints for read-only addresses.
Remote connection closed

I'm not sure when you say **IDE2** - what that means.

Thanks,
John

kittaakos, 2020-09-15:
> In the 0.1.2 IDE - I'm getting this:

> I'm not sure when you say **IDE2** - what that means.

Sorry for the confusion, @jwestmoreland; they are the same.

jwestmoreland, 2020-09-15:
OK Akos.

From the command line - it runs:

C:\tmp2\Display\build\arduinobeta.mbed.envie_m7_thread_debug>C:\Users\John\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\bin\arm-none-eabi-gdb -ex "set pagination off" --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote COM8" Display.ino.elf
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from Display.ino.elf...done.
Remote debugging using COM8
setup () at c:\tmp2\Display/Display.ino:239
239     {
(gdb) finish
Run till exit from #0  setup () at c:\tmp2\Display/Display.ino:239
Note: automatically using hardware breakpoints for read-only addresses.
[New Thread 604079360]
main () at C:\Users\john\AppData\Local\Arduino15\packages\arduinobeta\hardware\mbed\1.2.2\cores\arduino\main.cpp:47
47                      loop();
(gdb) continue
Continuing.

Thread 5 "" received signal SIGINT, Interrupt.
[Switching to Thread 604006360]
0x080615ae in osKernelResume (sleep_ticks=56) at .\mbed-os\rtos\source\TARGET_CORTEX\rtx5\RTX\Source\rtx_kernel.c:492
492     .\mbed-os\rtos\source\TARGET_CORTEX\rtx5\RTX\Source\rtx_kernel.c: No such file or directory.
(gdb) Quit (expect signal SIGINT when the program is resumed)
(gdb) quit
A debugging session is active.

        Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) y

C:\tmp2\Display\build\arduinobeta.mbed.envie_m7_thread_debug>

kittaakos, 2020-09-15:
> From the command line - it runs:

Thank you for writing this down. It does not work for me. I am going to add the steps I have used, but let me add some remarks first:
 - Unfortunately, I still do not know where you get the steps from, there is something else in the `platform.txt`:
https://github.com/arduino/ArduinoCore-mbed/blob/43105c7454d8185e10804d3bcc30f5baca96d3ee/platform.txt#L174
 - Your path to the build is different than on my machine:
   - Yours is `build\arduinobeta.mbed.envie_m7_thread_debug`,
   - Mine is `build/arduino-beta.mbed.envie_m7`. (Please note the name of the  platform)

I did the followings:
1. Make sure the sketch exists:
```
cat ~/Documents/Arduino/Blink/Blink.ino                                      
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

2. Check the CLI version:
```
./arduino-cli version                                                        
arduino-cli Version: 0.13.0 Commit: 693a045
```

3. Make sure the Portenta board is recognized:
```
./arduino-cli board list                                                                              
Port                            Type              Board Name                    FQBN                       Core             
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown                                                                   
/dev/cu.MALS                    Serial Port       Unknown                                                                   
/dev/cu.SOC                     Serial Port       Unknown                                                                   
/dev/cu.usbmodem14401           Serial Port (USB) Arduino Portenta H7 (M7 core) arduino-beta:mbed:envie_m7 arduino-beta:mbed
```

4. Compile:
```
./arduino-cli compile -b arduino-beta:mbed:envie_m7 ~/Documents/Arduino/Blink
Sketch uses 122792 bytes (15%) of program storage space. Maximum is 786432 bytes.
Global variables use 57072 bytes (10%) of dynamic memory, leaving 466552 bytes for local variables. Maximum is 523624 bytes.
```

5. Upload:
```
./arduino-cli upload -p /dev/cu.usbmodem14401 -b arduino-beta:mbed:envie_m7 ~/Documents/Arduino/Blink
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2341:035b
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08040000, size = 124632
Download	[=========================] 100%       124632 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

6. Try to run the debugger manually, based on your comment, although something else is defined in the `platform.txt`:
```
~/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gdb -ex "set pagination off" --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote /dev/cu.usbmodem14401" ~/Documents/Arduino/Blink/build/arduino-beta.mbed.envie_m7/Blink.ino.elf 
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/akos.kitta/Documents/Arduino/Blink/build/arduino-beta.mbed.envie_m7/Blink.ino.elf...done.
Remote debugging using /dev/cu.usbmodem14401
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
quit
^CRemote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb) Quit
(gdb) ^CQuit
(gdb) ^CQuit
(gdb) quit
```

jwestmoreland, 2020-09-15:
Hello Akos,

First - are you including this at the top of your file:

#include <ThreadDebug.h>
//
// UartDebugCommInterface debugComm(SERIAL1_TX, SERIAL1_RX, 230400);
// ThreadDebug            threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);

UsbDebugCommInterface  debugComm(&SerialUSB);
// ThreadDebug            threadDebug(&debugComm, DEBUG_NO_BREAK_IN_SETUP);
ThreadDebug            threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);

And, I've attached the files I'm using - note, this deals with the issue in the OP and we're talking about something related now.

I'm compiling in the IDE(2) - and then running from the command line.  The debugger disconnects when trying to run in the IDE(2).

Regards,
John
[boards.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224160/boards.txt)
[platform.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224161/platform.txt)
[programmers.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224162/programmers.txt)

jwestmoreland, 2020-09-15:
Akos,

Also - this is just something I type in from the command line:

C:\Users\John\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\bin\arm-none-eabi-gdb -ex "set pagination off" --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote COM8" Display.ino.elf

That's not in any of the .txt files.

Regards,
John

kittaakos, 2020-09-15:
Thank you so much for helping me with this issue, John.

I will try to run the debugger the way you did; once it works from the terminal, I can make the IDE adjustments. I'll keep you posted.

jwestmoreland, 2020-09-15:
Akos,

No problem.

@adamgreen thought a monitor reset command was being issued that was causing the disconnect in the IDE but I didn't find that - could be in the source someplace maybe.   Note there was some discussion here as well:  #19 .

Thanks,
John

jwestmoreland, 2020-09-15:
Akos,

Also - please try this:

USBSerial   SerialUSB1(false, "DebugPort");
UsbDebugCommInterface  debugComm(&SerialUSB1);
// ThreadDebug            threadDebug(&debugComm, DEBUG_NO_BREAK_IN_SETUP);
ThreadDebug            threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);

Thanks,
John

kittaakos, 2020-09-15:
> Regards,
> John
> [boards.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224160/boards.txt)
> [platform.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224161/platform.txt)
> [programmers.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224162/programmers.txt)
> When you use these files and make a debug build - that build will show up under ./build/...debug.. directory.

I can confirm, it works from the terminal, once I have overridden the files under `~/Library/Arduino15/packages/arduino-beta/hardware/mbed/1.2.2/` with yours.

My sketch looks like this:
```cpp
#include <ThreadDebug.h>
//
// UartDebugCommInterface debugComm(SERIAL1_TX, SERIAL1_RX, 230400);
// ThreadDebug threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);

UsbDebugCommInterface debugComm(&SerialUSB);
// ThreadDebug threadDebug(&debugComm, DEBUG_NO_BREAK_IN_SETUP);
ThreadDebug threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);

// the setup function runs once when you press reset or power the board
void setup()
{
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop()
{
  digitalWrite(LED_BUILTIN, HIGH); // turn the LED on (HIGH is the voltage level)
  delay(1000);                     // wait for a second
  digitalWrite(LED_BUILTIN, LOW);  // turn the LED off by making the voltage LOW
  delay(1000);                     // wait for a second
}
```

And here is the command I have executed:
```
~/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gdb -ex "set pagination off" --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote /dev/cu.usbmodem14401" ~/Documents/Arduino/Blink/build/arduino-beta.mbed.envie_m7/Blink.ino.elf
```
----

I will try to get this run from the IDE.

kittaakos, 2020-09-15:
> USBSerial SerialUSB1(false, "DebugPort");
> UsbDebugCommInterface debugComm(&SerialUSB1);
> // ThreadDebug threadDebug(&debugComm, DEBUG_NO_BREAK_IN_SETUP);
> ThreadDebug threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);

This does not compile on my side.

kittaakos, 2020-09-15:
> [boards.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224160/boards.txt)
> [platform.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224161/platform.txt) [...]

Here are the diffs (excluding the line endings `--strip-trailing-cr`) between the original files and the modified ones:

 1. `programmers.txt` diff:
```diff
13c13,21
< stlink.program.extra_params=
\ No newline at end of file
---
> stlink.program.extra_params=
> 
> jlink.name=SEGGER JLINK
> jlink.communication=USB
> jlink.protocol=
> jlink.program.protocol=swd
> jlink.program.tool=openocd
> jlink.program.extra_params=
> 
```

 2. `boards.txt` diff:
```diff
7a8,13
> # envie_m7_thread_debug.name=Arduino Portenta H7 (M7 core)
> envie_m7_thread_debug.build.core=arduino
> envie_m7_thread_debug.build.crossprefix=arm-none-eabi-
> 
> envie_m7_thread_debug.build.compiler_path={runtime.tools.arm-none-eabi-gcc.path}/bin/
> 
42a49,61
> ### WIP!  Note, this may not be able to coexist - proceed accordingly! --- jcw
> 
> ## for jlink - REM:  Must use WINUSB w/jlink (Zadig good utility for this under win10.)
> 
> envie_m7.debug.tool=gdb
> envie_m7.bootloader.tool=openocd
> envie_m7.bootloader.config=-f target/stm32h7x_dual_bank.cfg
> envie_m7.bootloader.programmer=-f interface/jlink.cfg
> envie_m7.bootloader.extra_action.preflash=stm32h7x option_write 0 0x01c 0xb86aaf0
> envie_m7.bootloader.file=PORTENTA_H7/portentah7_bootloader_mbed_hs_v2.elf
> 
> ## for stlink
> 
48c67,107
< envie_m7.bootloader.file=PORTENTA_H7/portentah7_bootloader_mbed_hs.elf
---
> envie_m7.bootloader.file=PORTENTA_H7/portentah7_bootloader_mbed_hs_v2.elf
> 
> envie_m7_thread_debug.build.variant=PORTENTA_H7_M7
> envie_m7_thread_debug.build.mcu=cortex-m7
> envie_m7_thread_debug.build.fpu=fpv5-d16
> envie_m7_thread_debug.build.float-abi=softfp
> envie_m7_thread_debug.build.extra_flags=
> envie_m7_thread_debug.build.architecture=cortex-m7
> envie_m7_thread_debug.build.ldscript=linker_script.ld
> envie_m7_thread_debug.compiler.mbed.arch.define=
> envie_m7_thread_debug.compiler.mbed.defines={build.variant.path}/defines.txt
> envie_m7_thread_debug.compiler.mbed.ldflags={build.variant.path}/ldflags.txt
> envie_m7_thread_debug.compiler.mbed.cflags={build.variant.path}/cflags.txt
> envie_m7_thread_debug.compiler.mbed.cxxflags={build.variant.path}/cxxflags.txt
> envie_m7_thread_debug.compiler.mbed.includes={build.variant.path}/includes.txt
> envie_m7_thread_debug.compiler.mbed.extra_ldflags=-lstdc++ -lsupc++ -lm -lc -lgcc -lnosys
> envie_m7_thread_debug.compiler.mbed="{build.variant.path}/libs/libmbed.a"
> 
> envie_m7_thread_debug.debug.tool=gdb
> envie_m7_thread_debug.name=Arduino Portenta H7 (ThreadDebug)
> envie_m7_thread_debug.vid.0=0x2341
> envie_m7_thread_debug.pid.0=0x025b
> envie_m7_thread_debug.descriptor.0=DebugPort
> envie_m7_thread_debug.hide=true
> envie_m7_thread_debug.build.board=PORTENTA_H7_M7
> envie_m7_thread_debug.file=C:/tmp/demo-web-server/build/arduinobeta.mbed.envie_m7_thread_debug/demo-web-server.ino.elf
> 
> envie_m7_thread_debug.upload.tool=dfu-util
> envie_m7_thread_debug.upload.protocol=
> envie_m7_thread_debug.upload.vid=0x2341
> # envie_m7_thread_debug.upload.pid=0x035b
> envie_m7_thread_debug.upload.pid=0x025b
> # envie_m7_thread_debug.upload.pid=0x045b
> envie_m7_thread_debug.upload.address=0x08040000
> envie_m7_thread_debug.upload.interface=0
> envie_m7_thread_debug.upload.use_1200bps_touch=true
> envie_m7_thread_debug.upload.wait_for_upload_port=true
> envie_m7_thread_debug.upload.native_usb=true
> envie_m7_thread_debug.upload.maximum_size=786432
> envie_m7_thread_debug.upload.maximum_data_size=523624
> 
```

 3. `platform.txt` diff:
```diff
127c127
< tools.openocd.bootloader.pattern="{path}/{cmd}" {bootloader.verbose} -s "{path}/share/openocd/scripts/" {bootloader.programmer} {bootloader.config} -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; {bootloader.extra_action.preflash}; program {{runtime.platform.path}/bootloaders/{bootloader.file}}; reset run; shutdown"
---
> tools.openocd.bootloader.pattern="{path}/{cmd}" {bootloader.verbose} -s "{path}/share/openocd/scripts/" {bootloader.programmer} {bootloader.config} -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; transport select swd; {bootloader.extra_action.preflash}; program {{runtime.platform.path}/bootloaders/{bootloader.file}}; reset run; shutdown"
```

jwestmoreland, 2020-09-15:
> > USBSerial SerialUSB1(false, "DebugPort");
> > UsbDebugCommInterface debugComm(&SerialUSB1);
> > // ThreadDebug threadDebug(&debugComm, DEBUG_NO_BREAK_IN_SETUP);
> > ThreadDebug threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);
> 
> This does not compile on my side.

Hmmm, it was added [here:](uhttps://github.com/arduino/ArduinoCore-mbed/commit/197b1b8a194ce9435cf7faed53f447cfa0e6458d)

jwestmoreland, 2020-09-16:
> > [boards.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224160/boards.txt)
> > [platform.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224161/platform.txt)
> > [programmers.txt](https://github.com/arduino/ArduinoCore-mbed/files/5224162/programmers.txt)
> > When you use these files and make a debug build - that build will show up under ./build/...debug.. directory.
> 
> Here are the diffs (excluding the line endings `--strip-trailing-cr`) between the original files and the modified ones:
> 
> 1. `programmers.txt` diff:
> 
> ```diff
> 13c13,21
> < stlink.program.extra_params=
> \ No newline at end of file
> ---
> > stlink.program.extra_params=
> > 
> > jlink.name=SEGGER JLINK
> > jlink.communication=USB
> > jlink.protocol=
> > jlink.program.protocol=swd
> > jlink.program.tool=openocd
> > jlink.program.extra_params=
> > 
> ```
> 
> 1. `boards.txt` diff:
> 
> ```diff
> 7a8,13
> > # envie_m7_thread_debug.name=Arduino Portenta H7 (M7 core)
> > envie_m7_thread_debug.build.core=arduino
> > envie_m7_thread_debug.build.crossprefix=arm-none-eabi-
> > 
> > envie_m7_thread_debug.build.compiler_path={runtime.tools.arm-none-eabi-gcc.path}/bin/
> > 
> 42a49,61
> > ### WIP!  Note, this may not be able to coexist - proceed accordingly! --- jcw
> > 
> > ## for jlink - REM:  Must use WINUSB w/jlink (Zadig good utility for this under win10.)
> > 
> > envie_m7.debug.tool=gdb
> > envie_m7.bootloader.tool=openocd
> > envie_m7.bootloader.config=-f target/stm32h7x_dual_bank.cfg
> > envie_m7.bootloader.programmer=-f interface/jlink.cfg
> > envie_m7.bootloader.extra_action.preflash=stm32h7x option_write 0 0x01c 0xb86aaf0
> > envie_m7.bootloader.file=PORTENTA_H7/portentah7_bootloader_mbed_hs_v2.elf
> > 
> > ## for stlink
> > 
> 48c67,107
> < envie_m7.bootloader.file=PORTENTA_H7/portentah7_bootloader_mbed_hs.elf
> ---
> > envie_m7.bootloader.file=PORTENTA_H7/portentah7_bootloader_mbed_hs_v2.elf
> > 
> > envie_m7_thread_debug.build.variant=PORTENTA_H7_M7
> > envie_m7_thread_debug.build.mcu=cortex-m7
> > envie_m7_thread_debug.build.fpu=fpv5-d16
> > envie_m7_thread_debug.build.float-abi=softfp
> > envie_m7_thread_debug.build.extra_flags=
> > envie_m7_thread_debug.build.architecture=cortex-m7
> > envie_m7_thread_debug.build.ldscript=linker_script.ld
> > envie_m7_thread_debug.compiler.mbed.arch.define=
> > envie_m7_thread_debug.compiler.mbed.defines={build.variant.path}/defines.txt
> > envie_m7_thread_debug.compiler.mbed.ldflags={build.variant.path}/ldflags.txt
> > envie_m7_thread_debug.compiler.mbed.cflags={build.variant.path}/cflags.txt
> > envie_m7_thread_debug.compiler.mbed.cxxflags={build.variant.path}/cxxflags.txt
> > envie_m7_thread_debug.compiler.mbed.includes={build.variant.path}/includes.txt
> > envie_m7_thread_debug.compiler.mbed.extra_ldflags=-lstdc++ -lsupc++ -lm -lc -lgcc -lnosys
> > envie_m7_thread_debug.compiler.mbed="{build.variant.path}/libs/libmbed.a"
> > 
> > envie_m7_thread_debug.debug.tool=gdb
> > envie_m7_thread_debug.name=Arduino Portenta H7 (ThreadDebug)
> > envie_m7_thread_debug.vid.0=0x2341
> > envie_m7_thread_debug.pid.0=0x025b
> > envie_m7_thread_debug.descriptor.0=DebugPort
> > envie_m7_thread_debug.hide=true
> > envie_m7_thread_debug.build.board=PORTENTA_H7_M7
> > envie_m7_thread_debug.file=C:/tmp/demo-web-server/build/arduinobeta.mbed.envie_m7_thread_debug/demo-web-server.ino.elf

You should probably just comment out the above.  I did that to try and force the debug file.

> > 
> > envie_m7_thread_debug.upload.tool=dfu-util
> > envie_m7_thread_debug.upload.protocol=
> > envie_m7_thread_debug.upload.vid=0x2341
> > # envie_m7_thread_debug.upload.pid=0x035b
> > envie_m7_thread_debug.upload.pid=0x025b
> > # envie_m7_thread_debug.upload.pid=0x045b
> > envie_m7_thread_debug.upload.address=0x08040000
> > envie_m7_thread_debug.upload.interface=0
> > envie_m7_thread_debug.upload.use_1200bps_touch=true
> > envie_m7_thread_debug.upload.wait_for_upload_port=true
> > envie_m7_thread_debug.upload.native_usb=true
> > envie_m7_thread_debug.upload.maximum_size=786432
> > envie_m7_thread_debug.upload.maximum_data_size=523624
> > 
> ```
> 
> 1. `platform.txt` diff:
> 
> ```diff
> 127c127
> < tools.openocd.bootloader.pattern="{path}/{cmd}" {bootloader.verbose} -s "{path}/share/openocd/scripts/" {bootloader.programmer} {bootloader.config} -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; {bootloader.extra_action.preflash}; program {{runtime.platform.path}/bootloaders/{bootloader.file}}; reset run; shutdown"
> ---
> > tools.openocd.bootloader.pattern="{path}/{cmd}" {bootloader.verbose} -s "{path}/share/openocd/scripts/" {bootloader.programmer} {bootloader.config} -c "telnet_port disabled; init; reset init; halt; adapter speed 10000; transport select swd; {bootloader.extra_action.preflash}; program {{runtime.platform.path}/bootloaders/{bootloader.file}}; reset run; shutdown"
> ```

jwestmoreland, 2020-09-16:
Akos,

Also - for the explanation of the different bootloader files - please look here:  #32 .

Regards,
John