arduino/ArduinoCore-mbed #48: "undefined reference" when compiling sketches that use pulseIn() for Portenta H7 boards

per1234, 2020-09-03:
`pulseIn()` has not yet been implemented for the Portenta.

```
undefined reference to `pulseIn'
```

hpssjellis, 2021-06-14:
@per1234 can we have an update on this? I am trying to get an UltraSonic Rangefinder working and it uses pulseIn(). I feel that this function could be done from scratch. If it is not yet implemented I think a work around would not be very hard to make.

per1234, 2021-06-14:
It seems it was fixed by https://github.com/arduino/ArduinoCore-mbed/commit/f5ddfa85431fa7ae391118cbb48c986687dba497, but this issue and https://github.com/arduino/ArduinoCore-mbed/issues/187 were just never closed.

Try updating to the latest version of the Arduino Mbed OS Portenta Boards platform. The fix was released in 2.1.0

closed 2021-06-14 by per1234

hpssjellis, 2021-06-15:
@per1234 

I am not getting any sensible values for either pulseIn() or pulseInLong() on the Portenta. The typical Arduino example does nothing

```
int pin = 7;
unsigned long duration;

void setup() {
  Serial.begin(9600);
  pinMode(pin, INPUT);
}

void loop() {
  duration = pulseIn(pin, HIGH);
  Serial.println(duration);
}
```

and a more advanced example loses the serial port

```
// pulseIn Example

const byte INPIN = 7;

void setup() {
   Serial.begin(9600);
   pinMode(INPIN, INPUT);
}

void loop() {
unsigned long res;

  Serial.print("Time: ");
  Serial.println(millis());
  Serial.println("Measuring...");
  Serial.flush(); // Needed since serial requires interrupts to operate.

  noInterrupts();
  res = pulseIn(INPIN, HIGH, 3000000UL);
  interrupts();

  if (res) {
     Serial.print("Time high (us): ");
     Serial.println(res);
  } else
     Serial.println("No signal present");

}

```

and pulseInLong() seems to also not give any useful values with a rangefinder

```
// pulseIn Example

const byte INPIN = 5;  // echo

void setup() {
   Serial.begin(9600);
   pinMode(INPIN, INPUT);
   pinMode(6, OUTPUT);     // trigger
}

void loop() {
unsigned long res;

  Serial.print("Time: ");
  Serial.println(millis());
  Serial.println("Measuring...");

  digitalWrite(D6, HIGH);
  delayMicroseconds(10); 
  digitalWrite(D6, LOW);   
  res = pulseInLong(INPIN, HIGH, 3000000UL);

  if (res) {
     Serial.print("Time high (us): ");
     Serial.println(res);
  } else
     Serial.println("No signal present");

}

```

Can anyone give an example of pulseIn() on the Portenta that gives any data?

facchinm, 2021-06-15:
:wave: 
A sensible sketch could be like 
```c++
int pin = 7;
unsigned long duration;

void setup() {
  Serial.begin(9600);
  pinMode(pin, INPUT_PULLDOWN);
  analogWrite(6, 12);
}

void loop() {
  duration = pulseIn(pin, HIGH);
  Serial.println(duration);
  delay(10);
}
```
then short pins 6 and 7 and you should get 95us pulses correctly printed.
A couple of notes: `pulseIn` is effectively a blocking function, while `Serial.print` runs in another thread; if you don't insert any delay the print thread will slow down till it won't be able to execute anymore. Just inserting a `delay()` makes the scheduler give some space for the print to be executed.

This is only needed if you want to print the values; in case you just want to use them to trigger another function there's no need for the delay and the function can run at full speed.

About the rangefinder, I don't have an ultrasonic sensor at hand right now but I can get tested by a colleague and report back :wink:

hpssjellis, 2021-06-15:
@facchinm Thank you so much for helping with this little issue. Your code works perfectly. I should be able to solve my RangeFinder issue now, since I have a starting point.

Here is the type of rangefinder I have an HC-SR04 which works on other boards.

[image]

hpssjellis, 2021-06-16:
I put this RnageFinder question on the Arduino forum at

[https://forum.arduino.cc/t/portenta-ultrasonic-rangefinder-hc-sr04/875414](https://forum.arduino.cc/t/portenta-ultrasonic-rangefinder-hc-sr04/875414)