arduino/ArduinoCore-mbed #51: Request:  Build instructions - libmbed.a

jwestmoreland, 2020-09-07:
Hello All,

I was wondering if we could get build instructions for libmbed.a - (rather than just patching it...) - 

[https://github.com/arduino/ArduinoCore-mbed/tree/master/variants/PORTENTA_H7_M7/libs](https://github.com/arduino/ArduinoCore-mbed/tree/master/variants/PORTENTA_H7_M7/libs)

I have mbed studio downloaded already - I see the Nano BLE is already a target there.

Thanks in advance,
John W.

aentinger, 2020-09-08:
It's halfway documented already in this repositories [README](https://github.com/arduino/ArduinoCore-mbed/blob/master/README.md). The following instructions should do the trick. Please let me know if they work, I'll augment the documentation (if you could create a PR even better).
```bash
git clone https://github.com/ARMmbed/mbed-os
cd ArduinoCore-mbed
./mbed-os-to-arduino -r $PATH_TO_CLONED_MBED_OS ENVIE_M7:ENVIE_M7
```

jwestmoreland, 2020-09-08:
Hello Alexander,

Quick question - I'll try that - but are any linker scripts required that we may not have available - or does the above make them  - i.e. - .link_options.txt and .link_script.sct ?

Thanks,
John W.

jwestmoreland, 2020-09-09:
Hello Alexander,

OK - gave that a try - I had to comment this out in the script:

#	if [ -n "$LOCAL_REPO" ]; then
#		echo -n " Ling local repo $LOCAL_REPO..."
#		# ... or link your local repo
#		if [ -d mbed-os ]; then
#			if [ ! -L mbed-os ]; then
#				rm -rf mbed-os
#				ln -s "$LOCAL_REPO" mbed-os
#			fi
#		fi
#	fi

And this is what I'm getting now:

:\Users\john\Arduino\Arduino_mbed-LATEST\ArduinoCore-mbed>sh .\mbed-os-to-arduino -r .\mbed-os ENVIE_M7:ENVIE_M7

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=0
LOCAL_REPO=.\mbed-os
REMOTE_BRANCH=
MBED_CORE_LOCATION=/c/Users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed

VARIANT=ENVIE_M7 BOARD=ENVIE_M7
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "C:\AppData\Local\Temp\mbed-os-program" (program)
[mbed] ENVIE_M7 now set as default target in program "mbed-os-program"
[mbed] Working path "C:\AppData\Local\Temp\mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
================================================
Please, consider creating a 'conf/mbed_app.json'
to avoid mbed-cli always recompile from scratch.
================================================
 done.
Compiling Mbed Application...argument -m/--mcu: ENVIE_M7 is not a supported MCU. Supported MCUs are:
ADV_WISE_1510,         ADV_WISE_1570,         ARCH_MAX,
ARCH_PRO,              ARDUINO_NANO33BLE,     ARM_CM3DS_MPS2,
ARM_MPS2_M0,           ARM_MPS2_M0P,          ARM_MPS2_M3,
ARM_MPS2_M4,           ARM_MPS2_M7,           ARM_MUSCA_B1,
ARM_MUSCA_B1_NS,       CY8CKIT_062S2_43012,   CY8CKIT_062_BLE,
CY8CKIT_062_WIFI_BT,   CY8CKIT_064B0S2_4343W, CY8CPROTO_062S3_4343W,
CY8CPROTO_062_4343W,   CYSBSYSKIT_01,         CYTFM_064B0S2_4343W,
CYW9P62S1_43012EVB_01, CYW9P62S1_43438EVB_01, DISCO_F413ZH,
DISCO_F429ZI,          DISCO_F469NI,          DISCO_F746NG,
DISCO_F769NI,          DISCO_H747I,           DISCO_H747I_CM4,
DISCO_H747I_CM7,       DISCO_L072CZ_LRWAN1,   DISCO_L475VG_IOT01A,
DISCO_L476VG,          DISCO_L496AG,          DISCO_L4R9I,
DISCO_L562QE,          EFM32GG11_STK3701,     EFM32GG_STK3700,
EP_AGORA,              EP_ATLAS,              EV_COG_AD3029LZ,
EV_COG_AD4050LZ,       FF1705_L151CC,         FF_LPC546XX,
FVP_MPS2_M0,           FVP_MPS2_M0P,          FVP_MPS2_M3,
FVP_MPS2_M4,           FVP_MPS2_M7,           GD32_F307VG,
GD32_F450ZI,           GR_LYCHEE,             GR_MANGO,
HEXIWEAR,              K22F,                  K64F,
K66F,                  K82F,                  KL25Z,
KL43Z,                 KL46Z,                 KW41Z,
LPC1114,               LPC1768,               LPC54114,
LPC546XX,              MAX32600MBED,          MAX32620FTHR,
MAX32625MBED,          MAX32625PICO,          MAX32630FTHR,
MIMXRT1050_EVK,        MOTE_L152RC,           MTS_DRAGONFLY_F411RE,
MTS_DRAGONFLY_F413RH,  MTS_DRAGONFLY_L471QG,  MTS_MDOT_F411RE,
NRF52840_DK,           NRF52_DK,              NUCLEO_F070RB,
NUCLEO_F072RB,         NUCLEO_F091RC,         NUCLEO_F103RB,
NUCLEO_F207ZG,         NUCLEO_F303K8,         NUCLEO_F303RE,
NUCLEO_F303ZE,         NUCLEO_F401RE,         NUCLEO_F411RE,
NUCLEO_F412ZG,         NUCLEO_F413ZH,         NUCLEO_F429ZI,
NUCLEO_F439ZI,         NUCLEO_F446RE,         NUCLEO_F446ZE,
NUCLEO_F746ZG,         NUCLEO_F756ZG,         NUCLEO_F767ZI,
NUCLEO_G031K8,         NUCLEO_G071RB,         NUCLEO_G474RE,
NUCLEO_H743ZI2,        NUCLEO_L073RZ,         NUCLEO_L152RE,
NUCLEO_L432KC,         NUCLEO_L433RC_P,       NUCLEO_L452RE_P,
NUCLEO_L476RG,         NUCLEO_L486RG,         NUCLEO_L496ZG,
NUCLEO_L496ZG_P,       NUCLEO_L4R5ZI,         NUCLEO_L4R5ZI_P,
NUCLEO_L552ZE_Q,       NUCLEO_WB55RG,         NUMAKER_IOT_M252,
NUMAKER_IOT_M263A,     NUMAKER_IOT_M487,      NUMAKER_PFM_M453,
NUMAKER_PFM_M487,      NUMAKER_PFM_NANO130,   NUMAKER_PFM_NUC472,
RHOMBIO_L476DMW1K,     RZ_A1H,                S1SBP6A,
S5JS100,               SDP_K1,                SDT32620B,
SDT32625B,             SDT52832B,             SDT64B,
TB_SENSE_12,           TMPM46B,               TMPM4G9,
UHURU_RAVEN,           WIO_3G,                WIO_BG96,
WIO_EMW3166,           XDOT_L151CC
[mbed] ERROR: "C:\Users\john\AppData\Local\Mbed Studio\mbed-studio-tools\python\python.exe" returned error.
       Code: 2
       Path: "C:\AppData\Local\Temp\mbed-os-program"
       Command: "C:\Users\john\AppData\Local\Mbed Studio\mbed-studio-tools\python\python.exe -u C:\AppData\Local\Temp\mbed-os-program\mbed-os\tools\make.py -t GCC_ARM -m ENVIE_M7 --source . --build .\BUILD\ENVIE_M7\GCC_ARM -v"
---

Thanks!
John

jwestmoreland, 2020-09-09:
Hello @aentinger .

After pointing to the arduino/mbed-os repo - getting this:
c:\Users\john\Arduino\Arduino_mbed-LATEST\ArduinoCore-mbed>sh .\mbed-os-to-arduino -r c:\AppData\Local\Temp\mbed-os-program\mbed-os PORTENTA_H7_M7:PORTENTA_H7_M7

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=0
LOCAL_REPO=c:\AppData\Local\Temp\mbed-os-program\mbed-os
REMOTE_BRANCH=
MBED_CORE_LOCATION=/c/Users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed

VARIANT=PORTENTA_H7_M7 BOARD=PORTENTA_H7_M7
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "C:\AppData\Local\Temp\mbed-os-program" (program)
[mbed] PORTENTA_H7_M7 now set as default target in program "mbed-os-program"
[mbed] Working path "C:\AppData\Local\Temp\mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Compiling Mbed Application...[mbed] ERROR: "C:\Users\john\AppData\Local\Mbed Studio\mbed-studio-tools\python\python.exe" returned error.
       Code: 1
       Path: "C:\AppData\Local\Temp\mbed-os-program"
       Command: "C:\Users\john\AppData\Local\Mbed Studio\mbed-studio-tools\python\python.exe -u C:\AppData\Local\Temp\mbed-os-program\mbed-os\tools\make.py -t GCC_ARM -m PORTENTA_H7_M7
--source . --build .\BUILD\PORTENTA_H7_M7\GCC_ARM -v"
---

Thanks,
John

jwestmoreland, 2020-09-09:
ok - fixed repo copied over to tmp - getting this now:

Generating defines...c:\Users\john\Arduino\Arduino_mbed-LATEST\ArduinoCore-mbed\mbed-os-to-arduino: line 142: /tmp/mbed-os-program/variants/PORTENTA_H7_M7/defines.txt: No such file or dGenerating defines...c:\Users\john\Arduino\Arduino_mbed-LATEST\ArduinoCore-mbed\mbed-os-to-arduGenerating defines...c:\Users\john\Arduino\Arduino_mbed-LATEST\ArduinoCore-mbed\mbed-os-to-arduino: line 142: /tmp/mbed-os-program/variants/PORTENTA_H7_M7/defines.txt: No such file or directory

aentinger, 2020-09-09:
Good morning :wave: :coffee: 
I'm pretty sure this script was never before run on Windows - please try Linux and use bash (not any fancy shell, e.g. fish shell).

jwestmoreland, 2020-09-09:
Hello @aentinger ,

> It's halfway documented already in this repositories [README](https://github.com/arduino/ArduinoCore-mbed/blob/master/README.md). The following instructions should do the trick. Please let me know if they work, I'll augment the documentation (if you could create a PR even better).
> 
> ```shell
> git clone https://github.com/ARMmbed/mbed-os
> cd ArduinoCore-mbed
> ./mbed-os-to-arduino -r $PATH_TO_CLONED_MBED_OS ENVIE_M7:ENVIE_M7
> ```

Can you make a build doing the above?

Clearly, I haven't been able to - and, I'm running in the mbed terminal - so, that should be fine.

Not having a symbolic link doesn't kill the build.

Are your instructions pointing to the correct repository?

Regards,
John

jwestmoreland, 2020-09-09:
Hello Alexander,

Getting this now - which is progress I suppose:

'''
$ c:/Users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed/mbed-os-to-arduino -r c:/AppData/Local/Temp/mbed-os-program/mbed-os PORTENTA_H7_M7:PORTENTA_H7_M7

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=0
LOCAL_REPO=c:/AppData/Local/Temp/mbed-os-program/mbed-os
REMOTE_BRANCH=
MBED_CORE_LOCATION=/c/users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed

VARIANT=PORTENTA_H7_M7 BOARD=PORTENTA_H7_M7
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... done.
Setting up Mbed Application...[mbed] Working path "C:\AppData\Local\Temp\mbed-os-program" (program)
[mbed] PORTENTA_H7_M7 now set as default target in program "mbed-os-program"
[mbed] Working path "C:\AppData\Local\Temp\mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Compiling Mbed Application...Compile [100.0%]: main.cpp
 done.
Generating defines... done.
Generating includes... copying to destination... /tmp/mbed-os-program
 done.
Generating flags...Patching '-fno-exceptions' flag for /c/users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed/variants/PORTENTA_H7_M7/cflags.txt
Patching '-fno-exceptions' flag for /c/users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed/variants/PORTENTA_H7_M7/cxxflags.txt
Patching '-fno-exceptions' flag for /c/users/john/Arduino/Arduino_mbed-LATEST/ArduinoCore-mbed/variants/PORTENTA_H7_M7/ldflags.txt
 done.
Generating libs... done.
Copying generic MbedOS headers to core... ssh: Could not resolve hostname c: Name or service not known
rsync: connection unexpectedly closed (0 bytes received so far) [sender]
rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.2.2]
'''

Thanks,
John W.

jwestmoreland, 2020-09-09:
Hello Again Alexander,

So, I got a 'blinky' app and libmbed.a to build.  The blinky app worked.  I'll try the new libmbed.a lib in a few sketches before declaring any sort of success there.

Since this 'flow' evidently isn't completely documented under M$ Win10 - would you like a PR request outlining that somewhat or how would you like to approach this?

Are folks at Arduino running this in the mbed terminal window?  I imagine that can work under mbed studio for linux and you can run your shell of choice in that as well.

I must admit the mbed deploy command was a little interesting...

Here's the blinky app - JIC:
`#include "mbed.h"
// Blinking rate in milliseconds
#define BLINKING_RATE     500ms
int main()
{
    // Initialise the digital pin LED1 as an output
    DigitalOut led(LED1);

    while (true) {
        led = !led;
        ThisThread::sleep_for(BLINKING_RATE);
    }
}
`

And, to add that to the script:
```
create_mbed_program () {
	echo -n "Setting up Mbed Application..."
	rm -rf .mbedignore

	mbed target "$BOARDNAME"
	mbed toolchain GCC_ARM

	cat > main.cpp << MAIN_C
#include "mbed.h"
// Blinking rate in milliseconds
#define BLINKING_RATE     500ms
int main()
{
    // Initialise the digital pin LED1 as an output
    DigitalOut led(LED1);

    while (true) {
        led = !led;
        ThisThread::sleep_for(BLINKING_RATE);
    }
}
MAIN_C
```

Thanks,
John W.

aentinger, 2020-09-09:
>Since this 'flow' evidently isn't completely documented under M$ Win10 - would you like a PR request outlining that somewhat or how would you like to approach this?

Thank you but I don't think so. It's not like the typical user will be cooking up their own mbed core every other day. Instead a new mbed core will be provided by Arduino in measured intervals when doing so is required (and I imagine that less and less as the core gets more "settled").

>Are folks at Arduino running this in the mbed terminal window? I imagine that can work under mbed studio for linux and you can run your shell of choice in that as well.

Not that I know of. Usually it's bash and that does the trick.

jwestmoreland, 2020-09-10:
Hello @aentinger ,

Thanks for the feedback.

I think a new write-up is in order since things like target name, variant, etc, plus some of the instructions are out of date.

Regards,
John

jwestmoreland, 2020-09-11:
Hello @aentinger ,

Sure - when I get to a good point on #28 - I'll do that.

Regards,
John