arduino/ArduinoCore-mbed #55: RPC on the M4 core needs -fexceptions in cflags.txt and cxxflags.txt files

hpssjellis, 2020-09-13:
As far as I know for the M4 core to run RPC commands it needs [cflags.tx](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M4/cflags.txt) and [cxxflags.txt](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M4/cxxflags.txt) to have ```-fexceptions``` inserted at the bottom of the file. 

Is this something that should be inserted by default for everyone? Would you like a PR for it?

sbhklr, 2020-09-16:
@hpssjellis How do you compile your sketches? I didn't have to change any flags.

hpssjellis, 2020-09-16:
Things may have changed. I did it a while ago. Nothing worked on the M4 core, made the changes and things worked. I will remove them and test it. It was mentioned on the arduino forum.

hpssjellis, 2020-09-17:
@sbhklr  I first made sure [this](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/my02f_easy_m4_rpc_print.ino) program compiled to the M4 core, I then deleted -fexceptions from the 2 files (cflags.txt and cxxflags.txt) and re compiled without even rebooting the Arduino IDE and got this error. Replaced -fexceptions and no error. I hope that is helpful.

```
arduino-beta\hardware\mbed\1.2.2\libraries\rpclib\src/rpc/msgpack/v1/detail/cpp11_zone.hpp:75:38: error: exception handling disabled, use -fexceptions to enable

                 throw std::bad_alloc();

```

sbhklr, 2020-09-20:
@hpssjellis it is, thank you! We'll look into it.

sbhklr, 2020-09-30:
@hpssjellis It will be solved with the the core update.

hpssjellis, 2020-09-30:
@sbhklr Thank you. Looking forward to the next update as I have a lot of examples to re-test. Any idea if we are talking days or month until the next release?

sbhklr, 2020-09-30:
@hpssjellis We're talking days :-)

hpssjellis, 2020-10-04:
@sbhklr Things are working great with the new mbed Version 1.2.3

One minor thing. RPC1.begin() would be more fault tolerant if it could have a serial speed that it then ignores. Presently I am converting Serial.begin(115200); using

#define Serial RPC1  // So the M4 regular serial prints to RPC

but then the 115200 halts the compile.

Probably not needed for this version of mbed, but something to think about for the future.

sbhklr, 2020-10-05:
@hpssjellis Thanks for the input! We'll consider this for the "virtual serial" implementation for M4.
/cc @facchinm

closed 2020-10-05 by sbhklr