arduino/ArduinoCore-mbed #60: Using any Serial1 config other than the default SERIAL_8N1 crashes Mbed OS

jareiko, 2020-09-24:
Quite simple to reproduce on Arduino Nano 33 BLE Sense:

```
void setup() {
  Serial1.begin(100000, SERIAL_8E2);
}

void loop() {
}
```

Expected: Clean startup.

Actual result: Mbed crashes, going into 4x long, 4x short LED flashing loop.

facchinm, 2020-09-24:
Hi @jareiko ,
the issue is about Nordic hardware (see these lines https://github.com/arduino/mbed-os/blob/cf72a296b0d388a76f797059dae2944559ca1d88/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L1090-L1095) .
The reason why the old core doesn't crash is that it doesn't apply the serial format but always fallbacks to 8N1 (https://github.com/arduino/ArduinoCore-nRF528x-mbedos/blob/6216632cc70271619ad43547c804dabb4afa4a00/cores/arduino/Serial.cpp#L33 , config is ignored).

I'm closing the issue but maybe we should add some documentation about this nRF specific limitation @per1234

closed 2020-09-24 by facchinm

per1234, 2020-09-24:
Great idea @facchinm! I have submitted a PR to add this information to the `Serial.begin()` reference page:
https://github.com/arduino/reference-en/pull/781

I welcome any feedback on that proposal @jareiko

jareiko, 2020-09-25:
I don't think that link is accurate. I was able to get 8E2 mode working by writing directly to the registers per the datasheet:

https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.1.pdf

facchinm, 2020-09-25:
Indeed, you are right; I'm opening an issue on mbed-os to apply the configuration from the datasheet

For reference, the section is
```
6.33.10.24 CONFIG
Address offset: 0x56C
```

per1234, 2020-09-25:
Should I close https://github.com/arduino/reference-en/pull/781?

facchinm, 2020-09-25:
I think we can keep it until ARMmbed/mbed-os#13665 is not fixed

kallaspriit, 2022-07-09:
Custom UART config and baudrate can indeed be set using registers.

For example to use `1000000` baud and `SERIAL_8E2` config you could do something like this (tested with the `sbus` library to communicate with a SBUS RC receiver).

```c++
#if defined(ARDUINO_ARCH_NRF52840) // Arduino Nano 33 BLE

#define UARTE0_BASE_ADDR 0x40002000
#define UARTE1_BASE_ADDR 0x40028000
#define UART_BAUDRATE_REG_OFFSET 0x524
#define UART_CONFIG_REG_OFFSET 0x56C
#define UART_CONFIG_8E2 0x000000F
#define UART_BAUD_100000 0x19114A7
#define UART0_BAUDRATE_REGISTER (((unsigned int *)(UARTE0_BASE_ADDR + UART_BAUDRATE_REG_OFFSET)))
#define UART1_BAUDRATE_REGISTER (((unsigned int *)(UARTE1_BASE_ADDR + UART_BAUDRATE_REG_OFFSET)))
#define UART0_CONFIG_REGISTER (((unsigned int *)(UARTE0_BASE_ADDR + UART_CONFIG_REG_OFFSET)))
#define UART1_CONFIG_REGISTER (((unsigned int *)(UARTE1_BASE_ADDR + UART_CONFIG_REG_OFFSET)))

#endif

...
// call begin with parameters that work
uart->begin(9600);

// set the registers for either UART0 or UART1
if (uart == 0)
{
  *UART0_BAUDRATE_REGISTER = UART_BAUD_100000;
  *UART0_CONFIG_REGISTER = UART_CONFIG_8E2;
}
else if (uart == 1)
{
  *UART1_BAUDRATE_REGISTER = UART_BAUD_100000;
  *UART1_CONFIG_REGISTER = UART_CONFIG_8E2;
}
```

Would be nice if this was handled on the platform side but it works as a work-around :)