arduino/ArduinoCore-mbed #72: CI uses the latest mbed-beta core which is a placeholder release

sbhklr, 2020-10-15:
To "smoothly" phase out the users from using the beta track of the mbed core we chose to mark it deprecated. To make the users get a notification about this we bumped the version to 1.3.1. That version is a dummy release though and the CI wants to use it. Let's change the CI to use the release version of the mbed core.

@per1234 
@facchinm

sbhklr, 2020-10-15:
Solved via https://github.com/arduino/ArduinoCore-mbed/commit/896611135750e85adf7bca1b9654ad6a8eeaab2f

closed 2020-10-15 by sbhklr

per1234, 2020-10-15:
@sbhklr would you like me to take charge of updating the other CI workflows that use the `arduino-beta:mbed` core?

sbhklr, 2020-10-15:
@per1234 That would be great as it seems like they will also fail :-)