arduino/ArduinoCore-mbed #74: High Density Connector Pin Names

hpssjellis, 2020-10-22:
@facchinm @sbhklr 

With other ~30 pin Arduinos I understand pin names relate to ports, for example::

PA14, PB3, PC1, PD10

, but on the PortentaH7 we have potentially 160 Pins. That many pins must be confusing for the engineers but is basically ridiculous for any users. Any chance we could define all 160 pins by the logical names:

J1-1
J1-2
J1-3

J2-1
J2-2   etc, etc

So that users could actually write code that makes sense, like this for D6:

digitalWrite( J2-59 , LOW);

I don't think #defines take up any programming space in memory so I can't think of a reason not to simplify the naming on the Portenta pins.

facchinm, 2020-10-26:
Hi @hpssjellis .
the problem here, IMO, is the "single source of truth" principle.
In fact, since the Portenta has no pin number printed on the silkscreen, also "MKR" pin numbers are arbitrary.
If we take the schematics as source of truth your proposal makes a lot of sense but the HD page also reports pin names (Px_y) right now, so when you search the schematics to find where is `J1-34` you would also find its "real name".
[image]

Still not sure which would be the best way to tackle this. Any proposal is appreciated btw :slightly_smiling_face:

hpssjellis, 2020-10-26:
Thanks for the reply @facchinm it does seem a bit confusing and then when we include OpenMV and the Vision Shield using Python it gets even more confusing. I made an Excel chart and CSV output here to try to understand it but I think I need more information, perhaps the "True Names"  

[https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/PINS/portenta-pins05.csv](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/PINS/portenta-pins05.csv)

[image]

hpssjellis, 2020-10-27:
@facchinm Quick, possibly silly question. The high density connector pins are hardwired aren't they? There isn't a software method to change the pins configuration is there?

facchinm, 2020-10-28:
@hpssjellis yes, they are hardwired to the corresponding ball of the processor (Pxy) or the the output pins of some ICs (in case of Ethernet, USB HS and DSI)

hpssjellis, 2021-03-13:
I would send a PR of the J1_0 to J2_80 #defines needed if I could get some clear data. This page looks useful for defining the PA_1 to PK_7 pins [patches](https://github.com/arduino/ArduinoCore-mbed/blob/2cd3540ce2cd2e13450f322747280b7cb018d8af/patches/0026-PortentaH7-fix-mbed-6.1-build.patch?fbclid=IwAR1sUlL60k8-vSQ-EKojG52TXusfwSh0ciRzAaEUrq7YhpJsvGIS5t7O1-Y#L659-L914).  

I presently have the Portenta Breakout board and it is possible a few of the pins are not accurate defined, such as  GPIO 0 - 6 and PWM 3, 7, 8, 9,  a spreadsheet of checked pins would be very useful. I can make it, I just need better information. The hardware guys must have something.

hpssjellis, 2021-03-14:
This PR seems relevant https://github.com/arduino/ArduinoCore-mbed/pull/163

hpssjellis, 2021-03-14:
I sent in this PR as a starting point for updating the Digital Pin defines to match how the Nano 33 BLE does Digital pins. This is   for both cores of the PortentaH7  [Match Portenta to Nano 33 BLE Digital Pins format 165](https://github.com/arduino/ArduinoCore-mbed/pull/165) 

I am working on defines for the J1 and J2 connectors.

hpssjellis, 2021-06-22:
Still curious if there is any plans to include pin names like PWM_1, or J1_02 as they are on the Portenta pin out diagram at the Arduino Store [here](https://store.arduino.cc/usa/portenta-h7) --> documents

[image]

facchinm, 2021-07-09:
@hpssjellis the PWM attached to HRTIm now work (in `master`) due to https://github.com/arduino/ArduinoCore-mbed/commit/2e8c0ecc2445b56133d05288aaa673f92c54ae39.
We also published the library for the Portenta Breakout Carrier that can be used "standalone" to reference the pins by their "logical" names https://github.com/arduino-libraries/Arduino_PortentaBreakout

wolfmanus, 2022-02-28:
Without clear pin numbering a lot of arduino library doesn't work, they cannot use it the pins "logical" names, only the uint8_t format.

hpssjellis, 2022-03-03:
@wolfmanus 

> Without clear pin numbering a lot of arduino library doesn't work, they cannot use it the pins "logical" names, only the uint8_t format.

I tried to make a conversion list of logical pin names to uint8_t pin names and also HD connector J1-0 to J2-80 pin names but it was too confusing. Does anyone have a list of all three formats, since that would be very useful?

facchinm, 2022-03-03:
Hi @wolfmanus and @hpssjellis ,
there's a way to obtain the "uint8_t" pin number from the "logical" pin name but it's a bit hidden inside the core (for a purpose, I'll explain it later).

The function is `PinNameToIndex()` and you can expose it by including `#include "pinDefinitions.h"`

The reason why it's hidden is that it brings in the whole mbed environment, making the compilation much slower (unbearably slow on Windows)

wolfmanus, 2022-03-03:
Hi @hpssjellis,
Thank you and respect for your work.
Hi @facchinm,
Thanks for the new info, i will try it soon.

I tried to use from @hpssjellis 's table two GPIO pin arduino number, but it worked uncertainly (the code works well with MKR ZERO).
The slow translation it appeared me too on Windows, but I have a strong machine.

closed 2022-03-03 by sebromero

facchinm, 2022-03-03:
@hpssjellis there is :wink: `digitalPinToPinName()` in fact does what you need, but of course the return is not "stringified" -> if you call
`digitalPinToPinName(0)` on a Portenta you'll get 127, which is `PH_15` -> `( 16 * ( 'H' - 'A' ) + 15 )`

hpssjellis, 2022-03-03:
So Awesome @facchinm 

```
#include "pinDefinitions.h"

String myPinNameFunction(int thePin){
  return "P" + String(char('A' + (digitalPinToPinName(thePin) / 16))) +"_" + String(digitalPinToPinName(thePin) % 16);

  
  }
  
void setup() {
  Serial.begin(115200);
  while (!Serial) {}

   // This tests each pinName converts to pin index
   // Serial.println("Pin PH_15: " + String(PinNameToIndex(PH_15)) );

   for (int myLoop=0; myLoop<=195; myLoop++){
      Serial.println("Special Pin " + String(myLoop) + ": " + myPinNameFunction(myLoop));
   }
   Serial.println("-----------------------------");
   
}

void loop() {

}

```

outputs:

```
Special Pin 0: PH_15
Special Pin 1: PK_1
Special Pin 2: PJ_11
Special Pin 3: PG_7
Special Pin 4: PC_7
Special Pin 5: PC_6
Special Pin 6: PA_8
Special Pin 7: PI_0
Special Pin 8: PC_3
Special Pin 9: PI_1
Special Pin 10: PC_2
Special Pin 11: PH_8
Special Pin 12: PH_7
Special Pin 13: PA_10
Special Pin 14: PA_9
Special Pin 15: P1_0
Special Pin 16: P1_1
Special Pin 17: P3_2
Special Pin 18: P3_3
Special Pin 19: PS_2
Special Pin 20: PS_3
Special Pin 21: PA_4
Special Pin 22: PA_6
Special Pin 23: PK_5
Special Pin 24: PK_6
Special Pin 25: PK_7
Special Pin 26: PA_0
Special Pin 27: PA_1
Special Pin 28: PA_2
Special Pin 29: PA_3
Special Pin 30: PA_4
Special Pin 31: PA_5
Special Pin 32: PA_6
Special Pin 33: PA_7
Special Pin 34: PA_8
Special Pin 35: PA_9
Special Pin 36: PA_10
Special Pin 37: PA_11
Special Pin 38: PA_12
Special Pin 39: PA_13
Special Pin 40: PA_14
Special Pin 41: PA_15
Special Pin 42: PB_0
Special Pin 43: PB_1
Special Pin 44: PB_2
Special Pin 45: PB_3
Special Pin 46: PB_4
Special Pin 47: PB_5
Special Pin 48: PB_6
Special Pin 49: PB_7
Special Pin 50: PB_8
Special Pin 51: PB_9
Special Pin 52: PB_10
Special Pin 53: PB_11
Special Pin 54: PB_12
Special Pin 55: PB_13
Special Pin 56: PB_14
Special Pin 57: PB_15
Special Pin 58: PC_0
Special Pin 59: PC_1
Special Pin 60: PC_2
Special Pin 61: PC_3
Special Pin 62: PC_4
Special Pin 63: PC_5
Special Pin 64: PC_6
Special Pin 65: PC_7
Special Pin 66: PC_8
Special Pin 67: PC_9
Special Pin 68: PC_10
Special Pin 69: PC_11
Special Pin 70: PC_12
Special Pin 71: PC_13
Special Pin 72: PC_14
Special Pin 73: PC_15
Special Pin 74: PD_0
Special Pin 75: PD_1
Special Pin 76: PD_2
Special Pin 77: PD_3
Special Pin 78: PD_4
Special Pin 79: PD_5
Special Pin 80: PD_6
Special Pin 81: PD_7
Special Pin 82: PD_8
Special Pin 83: PD_9
Special Pin 84: PD_10
Special Pin 85: PD_11
Special Pin 86: PD_12
Special Pin 87: PD_13
Special Pin 88: PD_14
Special Pin 89: PD_15
Special Pin 90: PE_0
Special Pin 91: PE_1
Special Pin 92: PE_2
Special Pin 93: PE_3
Special Pin 94: PE_4
Special Pin 95: PE_5
Special Pin 96: PE_6
Special Pin 97: PE_7
Special Pin 98: PE_8
Special Pin 99: PE_9
Special Pin 100: PE_10
Special Pin 101: PE_11
Special Pin 102: PE_12
Special Pin 103: PE_13
Special Pin 104: PE_14
Special Pin 105: PE_15
Special Pin 106: PF_0
Special Pin 107: PF_1
Special Pin 108: PF_2
Special Pin 109: PF_3
Special Pin 110: PF_4
Special Pin 111: PF_5
Special Pin 112: PF_6
Special Pin 113: PF_7
Special Pin 114: PF_8
Special Pin 115: PF_9
Special Pin 116: PF_10
Special Pin 117: PF_11
Special Pin 118: PF_12
Special Pin 119: PF_13
Special Pin 120: PF_14
Special Pin 121: PF_15
Special Pin 122: PG_0
Special Pin 123: PG_1
Special Pin 124: PG_2
Special Pin 125: PG_3
Special Pin 126: PG_4
Special Pin 127: PG_5
Special Pin 128: PG_6
Special Pin 129: PG_7
Special Pin 130: PG_8
Special Pin 131: PG_9
Special Pin 132: PG_10
Special Pin 133: PG_11
Special Pin 134: PG_12
Special Pin 135: PG_13
Special Pin 136: PG_14
Special Pin 137: PG_15
Special Pin 138: PH_0
Special Pin 139: PH_1
Special Pin 140: PH_2
Special Pin 141: PH_3
Special Pin 142: PH_4
Special Pin 143: PH_5
Special Pin 144: PH_6
Special Pin 145: PH_7
Special Pin 146: PH_8
Special Pin 147: PH_9
Special Pin 148: PH_10
Special Pin 149: PH_11
Special Pin 150: PH_12
Special Pin 151: PH_13
Special Pin 152: PH_14
Special Pin 153: PH_15
Special Pin 154: PI_0
Special Pin 155: PI_1
Special Pin 156: PI_2
Special Pin 157: PI_3
Special Pin 158: PI_4
Special Pin 159: PI_5
Special Pin 160: PI_6
Special Pin 161: PI_7
Special Pin 162: PI_8
Special Pin 163: PI_9
Special Pin 164: PI_10
Special Pin 165: PI_11
Special Pin 166: PI_12
Special Pin 167: PI_13
Special Pin 168: PI_14
Special Pin 169: PI_15
Special Pin 170: PJ_0
Special Pin 171: PJ_1
Special Pin 172: PJ_2
Special Pin 173: PJ_3
Special Pin 174: PJ_4
Special Pin 175: PJ_5
Special Pin 176: PJ_6
Special Pin 177: PJ_7
Special Pin 178: PJ_8
Special Pin 179: PJ_9
Special Pin 180: PJ_10
Special Pin 181: PJ_11
Special Pin 182: PJ_12
Special Pin 183: PJ_13
Special Pin 184: PJ_14
Special Pin 185: PJ_15
Special Pin 186: PK_0
Special Pin 187: PK_1
Special Pin 188: PK_2
Special Pin 189: PK_3
Special Pin 190: PK_4
Special Pin 191: PK_5
Special Pin 192: PK_6
Special Pin 193: PK_7
Special Pin 194: PA_-1
Special Pin 195: PA_-1
-----------------------------

```

hpssjellis, 2022-03-03:
@wolfmanus did you see the above list. Those pin names do not line up with any of the pin names I have collected from the data sheets. Do you understand any of it? I think the list above is the most important, my list below is probably totally wrong. I will make another one in the same folder.

Here is my old list of what I thought was the pin names  https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/PINS/Portenta-pins-working-copy.csv

Most of my stuff is taken from the datasheets  https://content.arduino.cc/assets/Pinout-PortentaH7_latest.pdf

facchinm, 2022-03-04:
Hi Jeremy,
the pins that don't make sense are either alternative pinmuxing or the special purpose `_C` pins
To get a better list, a sample code would be like
```
String myPinNameFunction(int thePin){
  int realPin = digitalPinToPinName(thePin) % ALT0;
  if (realPin == NC) {
    return "Not a pin";
  }
  String extra = "";
  switch (digitalPinToPinName(thePin) & DUAL_PAD) {
    case 0:
      break;
    case ALT0:
      extra = "ALT0";
      break;
    case ALT1:
      extra = "ALT1";
      break;
    case ALT2:
      extra = "ALT2";
      break;
    case DUAL_PAD:
      extra = "_C";
      break;
  }
  return "P" + String(char('A' + (realPin / 16))) +"_" + String(realPin % 16) + extra;
}
```

hpssjellis, 2022-03-05:
Thanks @facchinm That prints a bit better. Is ALT1 and ALT2 for different boards. For the Portenta only C and ALT0 show up?

Pin Index 0: PH_15, Back to pin index: 0    
Pin Index 1: PK_1, Back to pin index: 1    
Pin Index 2: PJ_11, Back to pin index: 2    
Pin Index 3: PG_7, Back to pin index: 3    
Pin Index 4: PC_7, Back to pin index: 4    
Pin Index 5: PC_6, Back to pin index: 5    
Pin Index 6: PA_8, Back to pin index: 6    
Pin Index 7: PI_0, Back to pin index: 7    
Pin Index 8: PC_3, Back to pin index: 8    
Pin Index 9: PI_1, Back to pin index: 9    
Pin Index 10: PC_2, Back to pin index: 10    
Pin Index 11: PH_8, Back to pin index: 11    
Pin Index 12: PH_7, Back to pin index: 12    
Pin Index 13: PA_10, Back to pin index: 13    
Pin Index 14: PA_9, Back to pin index: 14    
Pin Index 15: PA_0_C, Back to pin index: 15    
Pin Index 16: PA_1_C, Back to pin index: 16    
Pin Index 17: PC_2_C, Back to pin index: 17    
Pin Index 18: PC_3_C, Back to pin index: 18    
Pin Index 19: PC_2ALT0, Back to pin index: 19    
Pin Index 20: PC_3ALT0, Back to pin index: 20    
Pin Index 21: PA_4, Back to pin index: 21    
Pin Index 22: PA_6, Back to pin index: 22    
Pin Index 23: PK_5, Back to pin index: 23    
Pin Index 24: PK_6, Back to pin index: 24    
Pin Index 25: PK_7, Back to pin index: 25    
Pin Index 26: PA_0, Back to pin index: 26    
Pin Index 27: PA_1, Back to pin index: 27    
Pin Index 28: PA_2, Back to pin index: 28    
Pin Index 29: PA_3, Back to pin index: 29    
Pin Index 30: PA_4, Back to pin index: 21    
Pin Index 31: PA_5, Back to pin index: 31    
Pin Index 32: PA_6, Back to pin index: 22    
Pin Index 33: PA_7, Back to pin index: 33    
Pin Index 34: PA_8, Back to pin index: 6    
Pin Index 35: PA_9, Back to pin index: 14    
Pin Index 36: PA_10, Back to pin index: 13    
Pin Index 37: PA_11, Back to pin index: 37    
Pin Index 38: PA_12, Back to pin index: 38    
Pin Index 39: PA_13, Back to pin index: 39    
Pin Index 40: PA_14, Back to pin index: 40    
Pin Index 41: PA_15, Back to pin index: 41    
Pin Index 42: PB_0, Back to pin index: 42    
Pin Index 43: PB_1, Back to pin index: 43    
Pin Index 44: PB_2, Back to pin index: 44    
Pin Index 45: PB_3, Back to pin index: 45    
Pin Index 46: PB_4, Back to pin index: 46    
Pin Index 47: PB_5, Back to pin index: 47    
Pin Index 48: PB_6, Back to pin index: 48    
Pin Index 49: PB_7, Back to pin index: 49    
Pin Index 50: PB_8, Back to pin index: 50    
Pin Index 51: PB_9, Back to pin index: 51    
Pin Index 52: PB_10, Back to pin index: 52    
Pin Index 53: PB_11, Back to pin index: 53    
Pin Index 54: PB_12, Back to pin index: 54    
Pin Index 55: PB_13, Back to pin index: 55    
Pin Index 56: PB_14, Back to pin index: 56    
Pin Index 57: PB_15, Back to pin index: 57    
Pin Index 58: PC_0, Back to pin index: 58    
Pin Index 59: PC_1, Back to pin index: 59    
Pin Index 60: PC_2, Back to pin index: 10    
Pin Index 61: PC_3, Back to pin index: 8    
Pin Index 62: PC_4, Back to pin index: 62    
Pin Index 63: PC_5, Back to pin index: 63    
Pin Index 64: PC_6, Back to pin index: 5    
Pin Index 65: PC_7, Back to pin index: 4    
Pin Index 66: PC_8, Back to pin index: 66    
Pin Index 67: PC_9, Back to pin index: 67    
[... 103 lines cut ...]
Pin Index 171: PJ_1, Back to pin index: 171    
Pin Index 172: PJ_2, Back to pin index: 172    
Pin Index 173: PJ_3, Back to pin index: 173    
Pin Index 174: PJ_4, Back to pin index: 174    
Pin Index 175: PJ_5, Back to pin index: 175    
Pin Index 176: PJ_6, Back to pin index: 176    
Pin Index 177: PJ_7, Back to pin index: 177    
Pin Index 178: PJ_8, Back to pin index: 178    
Pin Index 179: PJ_9, Back to pin index: 179    
Pin Index 180: PJ_10, Back to pin index: 180    
Pin Index 181: PJ_11, Back to pin index: 2    
Pin Index 182: PJ_12, Back to pin index: 182    
Pin Index 183: PJ_13, Back to pin index: 183    
Pin Index 184: PJ_14, Back to pin index: 184    
Pin Index 185: PJ_15, Back to pin index: 185    
Pin Index 186: PK_0, Back to pin index: 186    
Pin Index 187: PK_1, Back to pin index: 1    
Pin Index 188: PK_2, Back to pin index: 188    
Pin Index 189: PK_3, Back to pin index: 189    
Pin Index 190: PK_4, Back to pin index: 190    
Pin Index 191: PK_5, Back to pin index: 23    
Pin Index 192: PK_6, Back to pin index: 24    
Pin Index 193: PK_7, Back to pin index: 25    
Pin Index 194: Not a pin, Back to pin index: 255    
Pin Index 195: Not a pin, Back to pin index: 255

facchinm, 2022-03-07:
The `ALT*` pins mean that the default selected mux is different from GPIO (you can find an explanation here https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/TARGET_STM32H747xI/TARGET_PORTENTA_H7/PeripheralPins.c#L24-L27 ). On Portenta we only have some analog pins configured as ALT0 (https://github.com/bcmi-labs/ArduinoCore-mbed/blob/master/variants/PORTENTA_H7_M7/variant.cpp#L42-L43) .

hpssjellis, 2022-05-06:
@sebromero can we open this until we have a conversion list or set of #defines that converts HD pin names to regular pin names or numbers. 

So I made a PCB to physically organize the HD connector pin names like they are presented on the Portenta. You can see it works since a Machine Learning program is showing on the OLED using the reverse side vision shield.  The PCB traces are very long as they go from the HD connectors to the headers and then on the other side back to the other HD connector. I still want to be able to code just using the HD pin names. Does anyone have an HD pin name to regular pin name conversion list so I can make a list of #defines  

@facchinm 
Strange bug, when the portenta is connected to my pcb, serial communication with the Arduino IDE dies. Both regular and bootloader communication. Any reason for this or suggestions for a fix?

[image]

[image]

[image]

facchinm, 2022-05-09:
Hi Jeremy,
if the board keeps running the sketch and just the communication is not functional it's probably due to the long stubs on high density USB0 pins (J1 25-27, which are shared with the ones in the USB-C connector).
Cutting the traces close to the HD connector looks like a no go in your PCB but maybe it's worth trying :upside_down_face:

hpssjellis, 2022-05-09:
Thanks @facchinm . Great idea, and easy to test. I build PCB's like I 3D Print (It takes me 8 trys to get anything correct). Easy next PCB just disconnect J1-25, 26, 27

hpssjellis, 2022-05-09:
@facchinm Thank you so much.    At least one of those pins J1-(25->27) caused the issue. I now have bootloader and regular serial port Arduino IDE connection. 

I have more PCB's I can test. I will have to check if the Vision Shield needs any of those USB0 pins and possibly check if only one of the 3 is the important one to disconnect.

Have a great day Martino!

[image]