arduino/ArduinoCore-mbed #75: Set preferred connection parameters

Dorsel89, 2020-10-23:
Hey everyone,

with the change from _ArduinoCore-nRF528x-mbedos_ to _ArduinoCore-mbed_ we have some problems with the _setPreferredConnectionParams()_ function.

If I understand this correctly the function moves from GAP (mbed 5) to GattServer (mbed >= 6.2.1). Unfortunately I dont see a way to change the connection parameter on mbed 6.0.0 which is currently used by ArduinoCore-mbed.

So my questions are:
- Is there any way to set preferred connection paramter in mbed 6.0.0?
- Is there a rough schedule when you want to switch to mbed 6.2.1 ?

Best regards

facchinm, 2020-10-26:
Hi @Dorsel89 ,
during this week we are trying to merge in mainline mbed-os the portenta related changes (https://github.com/arduino/mbed-os/pulls).
When this happens a full rebuild over 6.4.0 will be published and the core updated.

Since you are using the Nano33BLE which is already mainlined you can "simply" (if you are on Linux) rebuild `libmbed` using `mbed-os-to-arduino` script in the root of this repo (running `./mbed-os-to-arduino ARDUINO_NANO33BLE:ARDUINO_NANO33BLE` will just work)

Dorsel89, 2020-10-26:
Hey @facchinm,

this sounds good! Im working on the phyphox-arduino library (https://github.com/Staacks/phyphox-arduino) which is made for educational projects. So I am quite happy if teachers use microcontrollers at all - using linux or a console in general is probably to much for some of them.

Best regards

Dorsel89, 2021-01-06:
Hey everyone,

Im still waiting for the mbed-os rebase. Can you give any timeline how long it will take to go to mbed-os 6.2.1 or higher?

Best regards

facchinm, 2021-01-07:
PR tracking the rebuild https://github.com/arduino/ArduinoCore-mbed/pull/88 . Portenta changes were merged last Tuesday (https://github.com/ARMmbed/mbed-os/pull/13826) so we are ready to merge and release soon

closed 2021-04-27 by Dorsel89