arduino/ArduinoCore-mbed #76: WiFiClient client = server.available(); blocking

qedo, 2020-10-23:
is this the preferred behavior?
if it is not:
add in WiFiServer.cpp above line 38:
sock->set_blocking(false);

Best Regards
Andreas D.

facchinm, 2020-10-26:
Hi Andreas,
you are right, `available()` should not be blocking. @sbhklr can you check if adding the non blocking flag has any side effect on our tutorials?

sbhklr, 2020-10-26:
@facchinm Shouldn't be a problem as in the example code the `loop` function already implements a polling mechanism.
`begin()` and `write()` both are expected to be blocking from what I can tell looking at the examples. That means that the socket should be set back to blocking before `available()` ends. Makes sense?

qedo, 2020-10-27:
Hi all

additional info 

add a simple blink at the end of loop in the portenta wifi example. 
While server.available() is blocking, it never blinks.

Same with client.available. 

„By default, recv blocks until some data is received. If socket is set to non-blocking or times out, NSAPI_ERROR_WOULD_BLOCK can be returned to indicate no data.“

The wifi example works because of the break in „if (currentLine.length() == 0) { ...„
But in case of a broken http request without an empty line it waits forever. 
using while (client.available()) {... to get the whole request doesn‘t work. After the last byte it waits for data and is blocking. 

sock->set_timeout(SOCKET_TIMEOUT);
above 
int ret = sock->recv(data, rxBuffer.availableForStore());
helps but delays. 
Better use non-blocking and NSAPI_ERROR_WOULD_BLOCK ?

Best Regards
Andreas D.

sbhklr, 2020-10-27:
Thanks @qedo for pointing this out! Of course we want the APIs to be rock solid. I gave it a shot. Would you like to help testing it? It's here: https://github.com/arduino/ArduinoCore-mbed/pull/81

qedo, 2020-10-29:
@sebromero - 
after some more testing on a rather lame Network it turns out that _socket->set_blocking(false); in arduino::WiFiClient::receiveData() is sometimes clipping the data. I tried _socket->set_timeout(20); and now it works fine for me.

You can take a look at this example and play with the timeout:
https://github.com/qedo/arduinoPortentaTestFiles/tree/master/portentaSimpleWebserverPost

The example also shows a second issue. If you comment this line it doesn’t work. Portenta is blinking SOS:
https://github.com/qedo/arduinoPortentaTestFiles/blob/8386c7380d40b3eeaa0a284a24e5fbb4d9c94f37/portentaSimpleWebserverPost/portentaSimpleWebserverPost.ino#L44

Best Regards
Andreas D.

sebromero, 2020-10-29:
@qedo Hmm, interesting, I'll have a look. Can you elaborate on what you mean by clipping the data? The TCP protocol should take care of ensuring complete submission of the data.

qedo, 2020-10-29:
@sebromero after two newline characters follows the post part. Here the "ledC2=On"

POST / HTTP/1.1
Host: 192.168.1.31
Origin: http://192.168.1.31
Content-Type: application/x-www-form-urlencoded
Accept-Encoding: gzip, deflate
Connection: keep-alive
Upgrade-Insecure-Requests: 1
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0 Safari/605.1.15
Referer: http://192.168.1.31/
Content-Length: 8
Accept-Language: de-de

ledC2=On

The example shows the post part from the previous request:
[image]

With _socket->set_blocking(false); this post part sporadically is missing. With _socket->set_timeout(20); it is always there.

facchinm, 2021-07-09:
Should be fixed by https://github.com/arduino/ArduinoCore-mbed/pull/281

facchinm, 2021-07-19:
Closing as fixed since #281 was merged

closed 2021-07-19 by facchinm