arduino/ArduinoCore-mbed #83: Port to NRF52840 with crystal 32Mhz and crystal 32Khz

sabas1080, 2020-10-30:
Hi

I am trying to port a new chip with nrf52840 for this core, the module has two crystal 32Mhz and 32Khz, in other arduino core work fine
https://github.com/RAKWireless/RAK-nRF52-Arduino/tree/master/variants/WisCore_RAK4631_Board

and circuitpython working fine, but the two framework working with softdevice

My problem the bootloader work fine, but the blink not work for me, I have ported other chips and work fine, any possible problem with the glass or do you think it is something else?

Thanks!

facchinm, 2020-10-30:
Hi @sabas1080 ,
"the bootloader works fine" means you can upload a compiled sketch using bossac?
On Nano33BLE we have some board specific initialization here https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/ARDUINO_NANO33BLE/variant.cpp#L138 , some of these instruction may be harmful to your board maybe?
Second thought: if you attach a debugger can you see where it stops executing?

sabas1080, 2020-10-30:
@facchinm 

Yes, I can upload sketch using bossac, even with Adafruit bootloader, both bootloader work ok

I have done tests deleting the initialization and I have the error

If you can give me an example of how to use the debugger, I can do it, I have a jlink.

something that happens is that the USB port disappears when upload the sketch, I must re-enter bootloader mode

hpssjellis, 2020-10-31:
@sabas1080 I have moved away from the nrf52840 but here is a link to what I was doing about a year ago. I put the Arduino bootloader onto the Nordic nrf52840 and Particle Argon and messed around with it. Blink worked fine but I did have to find the new pins.

[https://community.platformio.org/t/same-bootloader-same-softdevice-different-board-different-pins/11411/8](https://community.platformio.org/t/same-bootloader-same-softdevice-different-board-different-pins/11411/8)

and also this blog

[https://community.platformio.org/t/like-if-you-have-an-nrf52840-nordic-usb-dongle-arduino-nano-33-ble-adafruit-sparkfun-april-makerdairy-fanstel-particle-or-any-other/11112](https://community.platformio.org/t/like-if-you-have-an-nrf52840-nordic-usb-dongle-arduino-nano-33-ble-adafruit-sparkfun-april-makerdairy-fanstel-particle-or-any-other/11112)

sabas1080, 2020-11-01:
@hpssjellis thanks, but I have already defined the pins on other cards and it works ok, I think this may be a clock problem or some other internal thing

https://github.com/ElectronicCats/ArduinoCore-mbed/tree/bastfrontier/variants/BAST_FRONTIER

facchinm, 2020-11-02:
@sabas1080 
Arduino and Adafruit bootloader differ significantly (since ours is not based on softdevices) and they require the user sketch to be relocated at different addressed.
But if you use the ARduino bootloader with the Arduino-mbed core everything is self consistent.
To debug, wire SWDIO and SWCLK from the bottom pads (https://content.arduino.cc/assets/Pinout-NANOsense_latest.pdf) to the corresponding pins in Jlink connector, attach the `elf` file and you'll see where the program stops

sabas1080, 2020-11-04:
@facchinm 
I have the next error via GDB

```
Program received signal SIGTRAP, Trace/breakpoint trap.
SystemInit ()
    at ./mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/device/system_nrf52840.c:194
```

 https://github.com/arduino/mbed-os/blob/portenta-mbed-6.2.0/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/device/system_nrf52840.c#L194

apparently if it is a problem with the clock

sabas1080, 2020-12-10:
The problem  was close

closed 2020-12-10 by sabas1080