arduino/ArduinoCore-mbed #85: Crashing when writing more than 64 bytes to Serial

tjpetz, 2020-11-06:
Writing more than 64 bytes to Serial on the Nano 33 BLE Sense board causes the board to crash and to start blinking the LED in the SOS pattern.

The following code reproduces the problem.  The execution of the line that writes 65 bytes (buff_65) will crash.

```
/**
 * Demonstrate the bug on Nano BLE board when attempting to write more than 64 bytes
 * in one shot to the serial port.
 */
 
void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  delay(3000);
  
  const char *buff_64 = "abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ01234567890a";
  Serial.println(strlen(buff_64));
  Serial.println(buff_64);

  const char *buff_65 = "abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ01234567890ab";
  Serial.println(strlen(buff_65));
  
  // This statement will cause the board to go into SOS mode.
  Serial.println(buff_65);

  Serial.println("If this line prints then the limit of 64 in a single shot is fixxed!");
  
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

sebromero, 2020-11-16:
I can confirm this with the BLE Sense. The Portenta doesn't that this issue.

facchinm, 2020-11-16:
Confirmed here too, working on a fix

andbe-gh, 2020-11-16:
I've encountered the same problem. (see [https://forum.arduino.cc/index.php?topic=713771.msg4796293#msg4796293](https://forum.arduino.cc/index.php?topic=713771.msg4796293#msg4796293))
The problem seems to happen only with the new 1.3 boardfiles. The old ones (1.1.6) are working.

facchinm, 2020-11-16:
It is indeed a problem with mbed 6.x ; this is the proposed fix https://github.com/facchinm/ArduinoCore-mbed/commit/a885c589f48619f8ba8a5e86758ae1096b446754 , it would be nice if you could test it

sebromero, 2020-11-16:
@facchinm LGTM. 👌

andbe-gh, 2020-11-16:
@facchinm works for me, too.

PaulZC, 2020-11-20:
It works for me too - thank you @facchinm

tjpetz, 2020-11-20:
It works for me too.  Thanks @facchinm

closed 2020-12-03 by facchinm