arduino/ArduinoCore-mbed #89: I2C frequency and waveform on Portenta H7

rchard-84, 2020-11-16:
Frequency:
Wire.begin() should default to 100kHz clock for the I2C bus.  In the default appears to give 116kHz.  Wire.setClock(400kHz) gives 333kHz. 

Waveform:
The waveform for the SCL line is not normal.  It stays low for <2 mu seconds and then goes high.  If frequency was the only problem you'd expect 4.3 mus (half of the period).  This may be evidence it's a code issue and not a bungled circuit.

Attached are 2 pics showing scope traces of the SCL (yellow trace) and SDA( red trace).  The same code running on UNO (first pic) and another of it running on the Portenta (second pic).  UNO is right on frequency and has that nice clock signal shape.

On The UNO
[image]

On The Portenta
[image]

Some I2C devices did successfully receive data from the Portenta on the I2C Bus however I couldn't get several sensors to send data back.  The same sensors have no issue on the UNO.  My sensor connectivity challenge could be related to the waveform issue, I don't know.

facchinm, 2020-11-16:
Hi @rchard-84 ,
you are very right. I2C "waveform" calculation on STM32 mbed implementation is very rough (https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32H7/i2c_device.h#L38) and very wrong if APB1CLK is not 54Mhz . We are preparing a patch for fixing it and make I2C compliant again.
Will let you know as soon as it lands on something stable so you can test it.

facchinm, 2020-12-22:
Will be fixed by https://github.com/ARMmbed/mbed-os/pull/14030

closed 2021-07-09 by facchinm