arduino/ArduinoCore-mbed #90: Bootloader for custom board

torenrl, 2020-11-17:
Hi

I am designing a custom board based on the nRF52XX mcu, where I am planning to use the Arduino IDE to upload sketches. During the development process, versions having different versions of the mcu family will be used, f.ex. the nRF52840 and nRF52820. Since the bootloader is written for the Nano 33 BLE featuring the nRF52840 mcu, would it be possible to get access to the bootloader source in order to modify this for other members of the nRF52 mcu family?

Also, is the name of the board hard-coded in the bootloader or possible to modify elsewhere?

-Tore

sabas1080, 2020-11-17:
Hi @torenrl 

the source code is not available yet, we have adapted the core to be compatible with the UF2 bootloader
https://github.com/ElectronicCats/ArduinoCore-mbed

hpssjellis, 2020-11-17:
Curious what your working on @torenrl as I have several nrf52840 boards (particle, nordic, makerdiary, fanstel). I have put the nano33BLE bootloader on several boards with moderate success. 3/4 of the pins work to some extent, just never managed to easily figure out how to re-assign pins that were never assigned by the Nano33BLE.

[discussion here](https://community.platformio.org/t/like-if-you-have-an-nrf52840-nordic-usb-dongle-arduino-nano-33-ble-adafruit-sparkfun-april-makerdairy-fanstel-particle-or-any-other/11112)

torenrl, 2020-11-17:
@sabas1080 Thanks. Does this mean, that if I modify the Adafruit bootloader, this will work with the Arduino IDE while using mbed libraries? (https://github.com/ElectronicCats/Adafruit_nRF52_Bootloader)

@hpssjellis I am working on a custom nRF52 board, where visible light communication for IoT purposes, controlled using a smartphone (user) and raspberry pi (infrastructure). BLE is needed for easy connection with a smartphone. Thank you for the link, I will read the discussion, it seems like you have managed to collect a lot of information about the different available bootloaders and made a fine comparison.

sabas1080, 2020-11-17:
@torenrl I modify the arduino variant that each board