arduino/ArduinoCore-mbed #92: vsnprintf with length of 80 chars for format hardfaults the board

janjongboom, 2020-11-23:
This sketch hardfaults the board on v1.3.0 of this Arduino Core on the Arduino Nano 33 BLE Sense:

```
void printf_wrapper(const char *format, ...) {
    static char print_buf[2048] = { 0 };

    va_list args;
    va_start(args, format);
    int r = vsnprintf(print_buf, 2048, format, args);
    va_end(args);

    if (r > 0) {
        Serial.write(print_buf);
    }
}

void setup() {
  // put your setup code here, to run once:
  delay(5000);
}

void loop() {
  // put your main code here, to run repeatedly:

  printf_wrapper("hello world\n");
  printf_wrapper("hello this is a long text with 40 chars\n");
  printf_wrapper("hello this is a long text with 80 chars hello this is a long text with 80 chars\n");

  delay(1000);

}
```

This works fine on the 1.1.6 core.

facchinm, 2020-11-23:
Hi Jan,
it looks like https://github.com/arduino/ArduinoCore-mbed/pull/88 already fixes the issue so it may be an mbed problem.
Would you mind testing it?

janjongboom, 2020-11-23:
@facchinm OK, good to know - we're patching our sketches for now. When do you expect a new core version to come out for this?

facchinm, 2020-11-23:
Just waiting for this PR to be merged https://github.com/ARMmbed/mbed-os/pull/13826 :wink:

facchinm, 2020-12-22:
Fixed by https://github.com/arduino/ArduinoCore-mbed/commit/6c36b4252e7d56c03c112ea5549a39eb90ed5de5

closed 2020-12-22 by facchinm