arduino/ArduinoCore-mbed #93: Arduino Nano 33 BLE mbed os crashes when PWM on more than 3 digital pins

vtpgit, 2020-11-29:
Arduino Nano 33 BLE mbed os crashes when PWM on more than 3 digital pins.

We are using ARDUINO NANO 33 BLE, Model NINA-8306

We are evaluating Nano 33 BLE for a project. The project requires 8 digital PWM outputs.

According to the documentation, pins D2 to D13 are PWM capable. We chose pins D2 to D9.

First we discovered that PWM on more than 4 digital pins crashed the mbed os (4 long, 4 short yellow blinks). Going over the nordic documentation we discovered that more than 4 pwm at a time seems like a HW limitation. The documentation wasn't very clear. Fine. I went over the github cpp code, RE'd it and found wiring_analog.cpp. Inside we found someone already discovered the problem and made a change to it. Being rather lazy I made a similar change directly in my wiring_analog.cpp and recompiled. All digital pins can now be used one at a time provided we destroy them after use. Next we tried to run 4 pins simultaneously and this is where we hit a wall. We could not run more than 3 at a time, adding the fourth again crashes the mbed os.

I was advised by a moderator over at the arduino forum to post it here as an mbed os software issue.

We are using the latest IDE and libraries as of time of this post.

Below is the piece of code. Not complicated.

`  
  // D2
  pinMode(D2, OUTPUT);
  analogWrite(D2, 255);

  // D4
  pinMode(D4, OUTPUT);
  analogWrite(D4, 255);

  // D6
  pinMode(D6, OUTPUT);
  analogWrite(D6, 255);

  // D8
  // this crashes mbed os
  pinMode(D8, OUTPUT);
  analogWrite(D8, 255);`

facchinm, 2020-12-02:
Hi @vtpgit ,
thanks for reporting! The reason for the bug is that a structure is not cleared after `free()` on the mbed side so the pwm you are instantiating on D3 and D5 would share the timer.
I pushed https://github.com/arduino/mbed-os/commit/2f3a1ea8b1a5b25af029b4805f66b3b0be60a0de on mbed-os core to fix this; as soon as https://github.com/ARMmbed/mbed-os/pull/13826 gets merged I'll rebuild the core and make a release :wink:

hex705, 2021-04-29:
is this close to a solution -- I am experiencing problems with a multi-motor project that runs fine on Arduino nano iot but crashes on nano BLE boards.  If it is not fixed an update to docs indicating this limitation would be appreciated.

closed 2021-07-09 by facchinm