arduino/ArduinoCore-mbed #94: Board version appears to be lagging previous deprecated nRF528x-mbedos in IDE

rmlearney-digicatapult, 2020-12-04:
Looking at the Boards Manager in Arduino IDE 1.8.13 I notice the following versioning discrepancy:

Arduino nRF528x Boards (Mbed OS) - latest version is 1.3.0

[DEPRECATED] Arduino embed-enabled Boards - latest version is 1.3.1

Are there any elements that didn't get carried over when the previous library was deprecated or is this just an honest mistake?

per1234, 2020-12-04:
Hi @rmlearney-digicatapult. The sole purpose of the 1.3.1 release was to force the deprecation of the beta platform. It wasn't actually a code release.

You can see the package index entry here:
https://downloads.arduino.cc/packages/package_index.json
```json
        {
          "name": "[DEPRECATED] Arduino mbed-enabled Boards",
          "architecture": "mbed",
          "version": "1.3.1",
          "category": "Contributed",
          "url": "http://downloads.arduino.cc/cores/ArduinoCore-mbed-nonexistant.tar.bz2",

```
...

Note the "nonexistant" part of the URL. This release is actually impossible to install, and will uninstall the beta platform if you attempt to install it via the Arduino IDE.

rmlearney-digicatapult, 2020-12-04:
Thanks very much @per1234 !

closed 2020-12-04 by rmlearney-digicatapult