arduino/ArduinoCore-mbed #96: Thread example RPC_m4

JSCLadon, 2020-12-09:
Hi there,
since I am learning, I really rely on the examples.  The following example seems to be not working correctly: 
https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/RPC/examples/RPC_m4/RPC_m4.ino

It produces the following output: 

17:30:10.131 -> Calling subtract -33
17:30:10.877 -> Calling subtract -33
17:30:11.456 -> Calling add Calling subtract 57
17:30:12.507 -> -33
17:30:13.188 -> Calling subtract -33
17:30:13.937 -> Calling subtract -33
17:30:14.475 -> Calling add Calling subtract 57
17:30:15.553 -> -33
17:30:16.228 -> Calling subtract -33
17:30:16.978 -> Calling subtract -33
17:30:17.555 -> Calling add Calling subtract 57
17:30:18.605 -> -33
17:30:19.282 -> Calling subtract -33
17:30:20.023 -> Calling subtract -33
17:30:20.596 -> Calling add Calling subtract 57
17:30:21.646 -> -33
17:30:22.352 -> Calling subtract -33
17:30:23.101 -> Calling subtract -33
17:30:23.641 -> Calling add Calling subtract 57
17:30:24.693 -> -33
17:30:25.373 -> Calling subtract -33
17:30:26.148 -> Calling subtract -33
17:30:26.690 -> Calling add Calling subtract 57
17:30:27.738 -> -33
17:30:28.447 -> Calling subtract -33
17:30:29.190 -> Calling subtract -33
17:30:29.731 -> Calling add Calling subtract 57
17:30:30.784 -> -33

I guess, that is not, what it is supposed to do. M7 is never really calling the functions. Is that correct?

facchinm, 2020-12-11:
Hi @JSCLadon ,
the example was never meant to be final, sorry.
Two things here:
* I'm attaching the same example but better commented :slightly_smiling_face: 
* The `printf` called by CM7 in the original code was printing to the internal UART, not on the USB, so it was "invisible" unless you connected a standalone USB-to-serial adapter

Here's the new code
```c++
#include "Arduino.h"
#include "RPC_internal.h"

String currentCPU() {
  if (HAL_GetCurrentCPUID() == CM7_CPUID) {
    return "CM7: ";
  } else {
    return "CM4: ";
  }
}

int add(int a, int b) {
  Serial.println(currentCPU() + "executing add()");
  delay(1000);
  return a + b;
}
int subtract(int a, int b) {
  Serial.println(currentCPU() + "executing subtract()");
  return a - b;
}

rtos::Thread t;

void call_substract() {
  while (1) {
    delay(700);
    int a = random(100);
    int b = random(100);
    RPC1.println(currentCPU() + "calling subtract on " + String(a) + " and " + String(b));
    auto res = RPC1.call("sub", a, b).as<int>();
    RPC1.println(currentCPU() + "result= " + String(res));
  }
}

void setup() {

  // Initialize RPC library; this also boots the M4 core
  RPC1.begin();
  Serial.begin(115200);
  //while (!Serial) {}

  // Both CPUs will execute this instruction, just at different times
  randomSeed(analogRead(A0));

  pinMode(LED_BUILTIN, OUTPUT);
  if (HAL_GetCurrentCPUID() == CM7_CPUID) {
    // M7 cpu becomes the server, so it exposes (binds) two functions
    RPC1.bind("add", add);
    RPC1.bind("sub", subtract);
  }
  else {
    // M4 cpu becomes the client, so spawns a thread that will call subtract() every 700ms
    t.start(call_substract);
  }
}

void loop() {

  if (HAL_GetCurrentCPUID() == CM4_CPUID) {

    // On M4 let's blink an LED; during the blink, call_substract() thread is running, so it will execute roughly 3 times (2000 / 700 ms)
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);

    // Let's invoke add() and wait for a result.
    // This will be delayed by the forced delay() in add() function
    // Exercise: if you are not interested in the result of the operation, what operation would you invoke?
    int a = random(100);
    int b = random(100);
    // PRC.print works like a Serial port, but it needs some receiver [1 - see below] to actually print the strings
    RPC1.println(currentCPU() + "calling add on " + String(a) + " and " + String(b));
    auto res = RPC1.call("add", a, b).as<int>();
    RPC1.println(currentCPU() + "result= " + String(res));
  }
  else
  {
    // On M7, let's print everything that is received over RPC1 Stream interface [1]
    // Buffer it otherwise all characters will be interleaved by other prints
    String to_print = "";
    while (RPC1.available()) {
      to_print += (char)RPC1.read();
    }
    if (to_print.length() > 0) {
      Serial.print(to_print);
    }
  }
}
``` 

Let me know if it is more understandable.
@sebromero should we improve the example replacing it with this one (or something even better commented / simpler) ?

JSCLadon, 2020-12-15:
Thanks a lot! Really appreciate your work!

sebromero, 2020-12-15:
@facchinm Might be a good call to update the RPC example. What's the benefit of using `HAL_GetCurrentCPUID ` during runtime vs. using the precompiler directives? We're wasting (a bit of) memory with this aren't we?

closed 2020-12-22 by facchinm