arduino/ArduinoCore-mbed #100: MBED symbols no longer exported since 1.1.4

gvamos, 2020-12-11:
mbed symbols that used to work from Arduino Nano 33 BLE sensor are no longer visible to user code. 

Symbols rtos, Thread, etc. are not recognized by the compiler and get error messages, 

Using board manager -> Arduino mbed-enabled Boards version 1.3.0  

Is this supposed to be done in some different way now?  If so, how?  Could you publish the correct way to access MBED functionality from user space?

Test case below  is from https://dumblebots.com/2020/04/06/programming-with-mbed-on-arduino/ Code in example cannot be copied from screen, so here is my manually entered copy below.
================================================================
#include <mbed.h>

using namespace rtos;
using namespace mbed;

Thread t1;
Thread t2;

void func1(){
  while(1) {
    Serial.println("A");
    wait(3);
  }
}

void func2(){
  while(1) {
    Serial.println("B");
    wait(2);
  }
}

void setup()
{
  Serial.begin(9600);
  t1.start(func1);
  t2.start(func2);
}

void loop()
{
}

facchinm, 2020-12-14:
Hi @gvamos ,
`wait(float seconds)` has been deprecated by mbed 6.0 and removed entirely from mbed 6.2 (which the lates core is based on) https://os.mbed.com/docs/mbed-os/v6.5/apis/wait.html.
This modified sketch works just as intended
```c++
#include <mbed.h>

using namespace rtos;
using namespace mbed;
using namespace std::chrono;
using namespace std::chrono_literals;

Thread t1;
Thread t2;

void func1(){
  while(1) {
    Serial.println("A");
    ThisThread::sleep_for(3s);
  }
}

void func2(){
  while(1) {
    Serial.println("B");
    ThisThread::sleep_for(2s);
  }
}

void setup()
{
  Serial.begin(9600);
  t1.start(func1);
  t2.start(func2);
}

void loop()
{
}
```

Using `chrono_literals` makes it trivial to understand the `sleep_for` duration (without checking the documentation to understand if it means seconds ot milliseconds or whatever).

closed 2020-12-14 by facchinm

gvamos, 2020-12-14:
Thanks for the prompt reply.  It works.  Yay!