arduino/ArduinoCore-mbed #105: Feature Request debugBreak(); not set breakpoint for regular Arduino build

hpssjellis, 2020-12-17:
Can [debugBreak()](https://github.com/arduino/ArduinoCore-mbed/blob/20a74c56a711f0acc13abda38228a28b210e76a4/libraries/ThreadDebug/src/ThreadDebug.h#L145-L147); only be a hard break when running the debugger? 

Yes I can set breakpoints in TRACE32, but much easier to set them in your code at specific places.

Presently I have to check that my code works using the serial monitor, then recompile for TRACE32 using debugBreak(); There must be a way that the 2 serial ports used by TRACE32 can be analyzed in your code so that debugBreak(); only works when both serial com ports are in use. 

[https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/ThreadDebug](https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/ThreadDebug)

P.S. if anyone is having issues getting TRACE32 working for the Portenta I made a video of the hacky way I do it [here](https://www.youtube.com/watch?v=pboPKm5ZDOQ&list=PL57Dnr1H_egtm0pi-okmG0iE_X5dROaLw&index=35).

adamgreen, 2020-12-17:
It seems like setting the breakpoints from the TRACE32 UI is the best way to only have a breakpoint enabled when the debugger is attached and to run normally otherwise.

Usually hard coded breakpoints are only used to issue a hard stop into the debugger if you know that your code has gone down a really bad code path that you want to look at under the debugger (ie. an assert).

hpssjellis, 2020-12-17:
@adamgreen: 

[Lauterbach.com](https://www.lauterbach.com/) has contacted me and we are working on a more efficient system-settings.cmm file that makes it easy to load the arduino .elf file making my request less important. 

Thanks for replying. I will make a new video when things are working. 

Twitter @rocksetta

closed 2020-12-17 by hpssjellis