arduino/ArduinoCore-mbed #107: Onboard IMU on Nano 33 BLE Sense Fails to Initialize When Using USBMouse Library

s-agar, 2020-12-20:
When using the USBMouse, USBKeyboard, and USBMIDI libraries in an Arduino sketch, the LSM9DS1 IMU on the Arduino Nano 33 BLE Sense does not initialize. I have posted on the forum [here](https://forum.arduino.cc/index.php?topic=718218) but I have not received any replies. As noted on the forum, the problem has also been reported [here](https://forum.arduino.cc/index.php?topic=666596.msg4495831#msg4495831).

facchinm, 2020-12-21:
Hi @s-agar ,
for Nano 33 BLE and Portenta the existing USB libraries haven't been ported yet, but you can use [these](https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/USBHID) as a replacement.
MIDI is not supported yet but it will be added in the near future.
Hope it helps, let me know if you need more info!

closed 2020-12-21 by facchinm

s-agar, 2020-12-21:
Hi @facchinm,

Thanks for the reply.

To clarify, in my sketch, would I include the libraries like this?:
`#include "PluggableUSBHID.h"` Haven't tried this one.
Or like this?:
`#include "USBMouse.h"` Previously I had been using this one. This one works; for example, I can run the code found [here](https://os.mbed.com/docs/mbed-os/v5.15/apis/usbmouse.html#usbmouse-example) and I get the expected output.

facchinm, 2020-12-22:
If you only include `"USBMouse.h"` the default mbed libray will be used, so you'll lose the CDC serial port.
By including `"PluggableUSBHID.h"` you'll also keep the serial and can compose as many devices as possibile (until there are endpoints available).
As you correctly stated, the APIs are mbed ones :slightly_smiling_face:

s-agar, 2020-12-23:
I'm not sure exactly how to use the PluggableUSBHID library... would appreciate it if you could guide me in the right direction. Also, would the IMU work with the PluggableUSBHID library? Does the IMU not work with the USBMouse and USBKeyboard libraries because they haven't been ported yet?

facchinm, 2020-12-23:
The IMU works just fine in combination with PluggableUSB libraries.
To get started, you can use something like this:
```c++
#include "PluggableUSBHID.h"
#include "USBMouse.h"
#include <Arduino_LSM9DS1.h>

USBMouse mouse;

void setup() {
  IMU.begin();
}

void loop() {
  float x, y, z;

  if (IMU.accelerationAvailable()) {
    IMU.readAcceleration(x, y, z);
    mouse.move(x, y);
  }
}
```
Hope it helps!

facchinm, 2020-12-23:
Or, in case you want to use mouse + keyboard, it is slightly different from standard Arduino HID 
```c++
#include "PluggableUSBHID.h"
#include "USBMouseKeyboard.h"
#include <Arduino_LSM9DS1.h>

USBMouseKeyboard key_mouse;

void setup() {
  IMU.begin();
}

void loop() {
  float x, y, z;

  if (IMU.accelerationAvailable()) {
    IMU.readAcceleration(x, y, z);
    key_mouse.move(x, y);
    if (z > 2.0) {
        key_mouse.printf("q\n");
    }
  }
}
```
Note that we must use `USBMouseKeyboard` library.
@sebromero could it make sense to have a tutorial? Or would it be better to just port to Arduino APIs and keep the existing documentation?

s-agar, 2020-12-23:
I tried both examples and the cursor doesn't move. I tried [the example from the Mbed documentation](https://github.com/ARMmbed/mbed-os-snippet-USBMouse_relative_pos/blob/v6.6/main.cpp) and got the expected results. I edited the example by including `"PluggableUSBHID.h` and uploaded it to the board, and nothing happened. Could this be a problem with just my PC/Nano board? I'm using version 1.3.0 of the Mbed core.

facchinm, 2020-12-24:
If you are on Windows you might need to help the device manager "forgetting" to board and registering again the driver.
To do this see [this](https://github.com/arduino/ArduinoCore-samd/issues/423#issuecomment-615093627) comment.
Rationale: windows caches the USB VID/PID/serial_number triplet and registers it with a driver (normally, just the standard Serial one). When we change descriptors we should also change at least one of the values to retrigger the discovery (or increase BCD version number as suggested [here](https://github.com/arduino-libraries/Keyboard/issues/41#issuecomment-730700142)). I'm going to take a look if this can be implemented easily.

facchinm, 2020-12-24:
@s-agar before uninstalling the drivers, would you mind testing #110 ? It should solve your issue automatically

s-agar, 2020-12-24:
@facchinm yes, I am on Windows. Should I just copy and paste the additions into their respective files on my PC or somehow download the pluggableUSB_bcd branch and replace the core on my PC with the core from the branch?

BTW Happy Holidays!

facchinm, 2020-12-28:
@s-agar you can do both :slightly_smiling_face:  Since the differences are small, maybe manually changing the files is the way to go for testing it.

s-agar, 2021-01-05:
Hi @facchinm,

I tried both ways:

- Manually changing the files:
     - Both sketches compiled, but neither produced any cursor/text output
     - In both sketches, the onboard orange LED repeatedly flashes 4 times fast and then 4 times slow
- Using Git to checkout pluggableUSB_bcd branch
     - Neither sketch compiles
     - Compiler error: In file "Wire.h": File "api/HardwareI2C.h" not found
     - I tried copying the api folder from the original core into the branch, and this gave another compiler error

 Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano 33 BLE"

In file included from C:\Users\shour\OneDrive\Documents\Arduino\libraries\Arduino_LSM9DS1\src/LSM9DS1.h:21:0,

                 from C:\Users\shour\OneDrive\Documents\Arduino\libraries\Arduino_LSM9DS1\src/Arduino_LSM9DS1.h:23,

                 from C:\Users\shour\AppData\Local\Temp\arduino_modified_sketch_747781\sketch_jan05a.ino:3:

C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\libraries\Wire/Wire.h:46:20: error: conflicting return type specified for 'virtual size_t arduino::MbedI2C::requestFrom(uint8_t, size_t, bool)'

     virtual size_t requestFrom(uint8_t address, size_t len, bool stopBit);

                    ^~~~~~~~~~~

In file included from C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/api/ArduinoAPI.h:30:0,

                 from C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/Arduino.h:51,

                 from sketch\sketch_jan05a.ino.cpp:1:

C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/api/HardwareI2C.h:39:21: error:   overriding 'virtual uint8_t arduino::HardwareI2C::requestFrom(uint8_t, size_t, bool)'

     virtual uint8_t requestFrom(uint8_t address, size_t len, bool stopBit) = 0;

                     ^~~~~~~~~~~

In file included from C:\Users\shour\OneDrive\Documents\Arduino\libraries\Arduino_LSM9DS1\src/LSM9DS1.h:21:0,

                 from C:\Users\shour\OneDrive\Documents\Arduino\libraries\Arduino_LSM9DS1\src/Arduino_LSM9DS1.h:23,

                 from C:\Users\shour\AppData\Local\Temp\arduino_modified_sketch_747781\sketch_jan05a.ino:3:

C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\libraries\Wire/Wire.h:47:20: error: conflicting return type specified for 'virtual size_t arduino::MbedI2C::requestFrom(uint8_t, size_t)'

     virtual size_t requestFrom(uint8_t address, size_t len);

                    ^~~~~~~~~~~

In file included from C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/api/ArduinoAPI.h:30:0,

                 from C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/Arduino.h:51,

                 from sketch\sketch_jan05a.ino.cpp:1:

C:\Users\shour\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/api/HardwareI2C.h:40:21: error:   overriding 'virtual uint8_t arduino::HardwareI2C::requestFrom(uint8_t, size_t)'

     virtual uint8_t requestFrom(uint8_t address, size_t len) = 0;

                     ^~~~~~~~~~~

exit status 1

Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

facchinm, 2021-01-07:
Hi @s-agar ,
the git clone method is perfect, but [this](https://github.com/arduino/ArduinoCore-API/tree/master/api) `api` folder should be used.
With that, you'll get rid of the compilation error and can test safely :slightly_smiling_face: 
Thanks!

s-agar, 2021-01-07:
Thanks, this worked perfectly!