arduino/ArduinoCore-mbed #112: Portenta Vision Shield Camera demo not running

hpssjellis, 2020-12-27:
Does anyone have the [demo example](https://github.com/arduino/ArduinoCore-mbed/blob/14e0f40fd85f2bf103278a5f430e1fe10c3e4ed6/libraries/Portenta_Camera/examples/Envie_camera/Envie_camera.ino#L12) running for the Portenta Vision Shield? 

I had to send an integer to start for the demo to even compile.

```
cam.start(3000);
```

My issues are very similar to the issues with the ov7670 Camera and [Java Processing program](https://processing.org/download/) except those issues were solved by switching from Windows to Linux. For the Portenta both Windows and Linux seem to have the zigzag issue for me using the Portenta Vision Shield.

[https://github.com/arduino-libraries/Arduino_OV767X/issues/5](https://github.com/arduino-libraries/Arduino_OV767X/issues/5)

It feels like the buffer is not getting properly reset, or it is just the Processing program that can't read the serial information fast enough. Does anyone know of a different Processing program than  [https://processing.org/download/](https://processing.org/download/)

facchinm, 2020-12-28:
The example not compiling is due to this https://github.com/arduino/ArduinoCore-mbed/commit/8911e84b92de6f4555f45b0fbd1dcf4be62f3e36 (I'll fix the example by adding a sensible default timeout).
About the zigzag issue, are you sure you set the resolution right (Processing side)? The four extra pixels on both axis must be added unless you're cropping.
A screenshot after uncommenting `cam.testPattern(true) ` would help too.

hpssjellis, 2020-12-28:
By the way in my download of Core 1.3.1 the extras folder did not come with the Camera demo like it did with 1.3.0. I will play with the idea of extra pixels for processing although the [included Processing code](https://github.com/arduino/ArduinoCore-mbed/blob/14e0f40fd85f2bf103278a5f430e1fe10c3e4ed6/libraries/Portenta_Camera/extras/CameraRawBytesVisualizer/CameraRawBytesVisualizer.pde#L16-L17) see below, does not mention it.
```
// must match resolution used in the sketch
final int cameraWidth = 324;
final int cameraHeight = 244;
final int cameraBytesPerPixel = 1;

```
 
I will also load my ```cam.testpattern(true)``` although I remember it not being very interesting, skinny black and white lines.

hpssjellis, 2020-12-28:
@facchinm 

I think the returned FRAME_BUFFER ``` (uint8_t *)LCD_FRAME_BUFFER; ``` from the [cam.grab() ](https://github.com/arduino/ArduinoCore-mbed/blob/14e0f40fd85f2bf103278a5f430e1fe10c3e4ed6/libraries/Portenta_Camera/camera.cpp#L568)  is not being zeroed before being written to, my reasoning is that the cam.snapshot() works fine (but only allows one picture, I can't seem to reset the camera in my main loop) and the test patterns seem fine, knowing that the Processes program works much better on Ubuntu than windows.

Ubuntu TestPattern
[image]

Windows Test Pattern (Even with the ov7670 camera I never got a great image using windows)
[image]

P.S. I tried different: 
```
cameraWidth = 328;    // 324
cameraHeight = 248;   // 244
```

Small changes to cameraWidth shifts the image diagonally
Small changes to cameraHeight seems to have minimal effects.

hpssjellis, 2021-01-07:
@facchinm @sebromero or anyone else. Does the Portenta Vision Shield using the [camera library](https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries/Portenta_Camera) and [Processes](https://processing.org/download/) work for you? It is very close to working for me just wondering if I am getting the same results or if I have made a minor mistake.

I get the proper black and white image, just has a few layer issues. It would be really good for me to know that the basic example does work for someone else.

pnndra, 2021-01-08:
@hpssjellis the issue is not really with capture but rather with the way image is being transferred. very likely some character is being added or removed for some reason.
you're right in saying we don't zero the buffer before acquisition because that would really be a waste of CPU bandwidth. 
the issue you're experiencing seems to be related to the windows counterpart that likely is handling the stream as an ASCII stream rather than binary and likely that is causing things like CR to be translated to CRLF, which would justify the stuff you're seeing that basically seems "random" insertion of bogus characters.
of course you should keep the resolution coherent with the one programmed on the camera otherwise image will be skewed horizontally.

hpssjellis, 2021-01-08:
Thanks @pnndra the windows CR issue probably explains why [Processe](https://processing.org/download/)s works much better on Ubuntu than Windows. I might just start working with the images as they are as I am building machine learning apps using [Edge Impulse](https://www.edgeimpulse.com/) so eventually the image will never actually be seen using a serial port as the ML program extracts information internally from each image.

pnndra, 2021-01-11:
@hpssjellis you will soon get something out of the box for your latter use case :). not asking you to wait but be aware we're on it.

facchinm, 2021-03-18:
Should be fixed by #122 and friends, feel free to reopen if the problem persists

closed 2021-03-18 by facchinm

sebromero, 2021-03-18:
@hpssjellis We also updated the demo slightly. See here: https://github.com/arduino/ArduinoCore-mbed/pull/143

gargoyle777, 2022-04-20:
hi, sorry to open this again.
I've been working on a windows pc, portenta H7 with a ethernet vision shield.
As them, i have the same zigzag error, i have tried both arduino Ide 1.8.x and 2, both updated. Exemples updated too.
I can see the proper image when i use openMV, but for a begginer like me it's getting hard to turn the portenta into an wifiAP and using multithreading with the modified python used in the openMV ide.
Any help would be appreciated.

hpssjellis, 2022-04-20:
@gargoyle777   I am working on these portenta vision shield examples, which uses a grayscale OLED (very fast) and make debugging much easier than the default examples. I hope it helps in some way.

https://github.com/hpssjellis/portenta-pro-community-solutions/tree/main/examples/dot3-portenta-vision-shields/dot35-camera-and-oled