arduino/ArduinoCore-mbed #113: Add SPI clock divisor for compatibility with legacy SPI library for AVR

Rotzbua, 2020-12-28:
## Problem

Compatibility problems with libraries who use hard coded old SPI constants.

## Example 

`MFRC522.h:22:26: error: 'SPI_CLOCK_DIV4' was not declared in this scope`
https://github.com/miguelbalboa/rfid/issues/544

## Possible solution

Add constants as done in samd implementation:
https://github.com/arduino/ArduinoCore-samd/commit/98341bfb821d25821e4fc60511b50405cb874c2b

facchinm, 2020-12-30:
Hi @Rotzbua ,
since `SPI_CLOCK_DIV4` is directly used [here](https://github.com/miguelbalboa/rfid/blob/16c419b6579d0a8ff25ee4ef5ec9690dd3738326/src/MFRC522.cpp#L49) as frequency parameter and not as `.setClockDivider()` (https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/SPI/src/SPI.h#L302) I think the library should be patched to conform to SPISettings specifications (https://www.arduino.cc/en/Reference/SPISettings) more than patching this core to accept bogus values :slightly_smiling_face: 
Also, samd will probably try to configure the SPI at 12Hz, not sure what happens, but it's definitely a bug in the library.

closed 2020-12-30 by facchinm

Rotzbua, 2020-12-30:
@facchinm Thanks for the hint. So basically the following code is BS?

> SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0));	

Because `SPISettings` needs as argument the clock speed as `Hz` but `SPI_CLOCK_DIV4` does not contain the clock speed?