arduino/ArduinoCore-mbed #116: Computer no longer recognizes the Portenta H7 when connected by USB.

kevb1540, 2021-01-10:
the first day I had the Portenta H7 everything worked fine. On the second day, I uploaded an Arduino sketch that would flash the onboard RGB LED in different colours. After that, the computer would no longer recognize the Portenta H7 when connected. I have tried different ports and I have tried restarting the computer. Nothing has worked. Please help if you can and thank you for taking the time to read this. Have a good day.

per1234, 2021-01-10:
Hi @kevb1540. Please try this:

1. Press and release the reset button on the Portenta
2. Wait for the green light to come on. It does this almost immediately, but you do need to wait for it.
3. While the green LED is still on (it only stays on for one second so you need to be quick), press and release the reset button again. You should now see the green LED pulsing. If not, you may have gotten the timing wrong so repeat the process.

The upload should now finish successfully. After this, you should be able to go back to doing normal uploads without needing to press the reset button. If you still need to do the reset trick to do uploads after this, the problem may be caused by your code. You can verify this by uploading a simple sketch like **File > Examples > 01.Basics > BareMinimum**.

---
If this solves your problem, you can go ahead and close this issue.

In the future, it's better to post this sort of support request to [the Arduino Forum](https://forum.arduino.cc/). This issue tracker is intended to be used for reporting bugs and making feature requests.

hpssjellis, 2021-01-10:
Reminder @kevb1540 and @per1234 that the bootloader has an update, because a few situations lead to problems rebooting. Did @kevb1540 do the File --> Examples -->Portenta_System --> PortentaH7_update_bootloader? If this step is important it should be mentioned when we get our Portenta's.

By the way I also once got into a bricked looking situation, I spent a while reconnecting and pushing the button using several different speeds and it eventually connected. Also the Portenta Vision Shield with a JATG or SWD debugger should be able to reset the bootloader but I have not anywhere seen the steps for how to do that.

pennam, 2022-03-29:
@diplodocuslongus have you tried to update the bootloader using the Tools-> Burn bootloader menu of the IDE or using the update_bootloader Sketch?
Which core version are you using?

diplodocuslongus, 2022-03-29:
@pennam I was using the update_bootloader sketch from the IDE (which by the way is now located in STM32H747_System instead of the portenta_System from the examples). My original bootloader version was 15. Now all is fine, I have been able to update it.

JIHANET, 2025-06-03:
I’m having the same issue with the orange LED on my Portenta H7 (using the official breakout board). It worked perfectly until today, when I uploaded a sketch and, after that, the board is no longer recognized by my computer at all. I’ve tried every bootloader method—double-reset, BOOT0 jumper, USB-only, VIN+USB—with the Portenta by itself and also attached to the breakout (making sure BT\_SELECT is set correctly), but no COM port ever appears. Has anyone seen this before or know how to recover the board?