arduino/ArduinoCore-mbed #117: SPI.h doesn't work with Nano BLE 33

marcoumlaut, 2021-01-13:
I've used an Arduino Nano with SPI.h to make a SPI Slave. Now I have the Nano 33 BLE and want to use the same code, but there are Errors when I build the file. Does anyone have an idea what's the issue?

The Code:

#include <SPI.h>

volatile byte c;
volatile byte command = 0;

void setup (void)
  {
  // have to send on master in, *slave out*
  pinMode(MISO, OUTPUT);
  pinMode(MOSI, INPUT);

  // turn on SPI in slave mode
  SPCR |= _BV(SPE);

  // turn on interrupts
  SPI.attachInterrupt();
  
  Serial.begin(115200);
 }  // end of setup

// SPI interrupt routine
ISR (SPI_STC_vect)
  {
  c = SPDR;
  Serial.println(SPDR);

  //SPDR = 8;

  }  // end of interrupt service routine (ISR) SPI_STC_vect

void loop (void)
  {
  if (digitalRead (SS) == HIGH)
    command = 0;
  
  }  // end of loop

The Error:

Arduino: 1.8.13 (Windows 10), Board: "Arduino Nano 33 BLE"

sketch_jan12a:22:5: error: expected constructor, destructor, or type conversion before '(' token

 ISR (SPI_STC_vect)

     ^

C:\Users\...\Documents\Arduino\sketch_jan12a\sketch_jan12a.ino: In function 'void setup()':

sketch_jan12a:13:3: error: 'SPCR' was not declared in this scope

   SPCR |= _BV(SPE);

   ^~~~

C:\Users\...\Documents\Arduino\sketch_jan12a\sketch_jan12a.ino:13:3: note: suggested alternative: 'SPI'

   SPCR |= _BV(SPE);

   ^~~~

   SPI

sketch_jan12a:13:15: error: 'SPE' was not declared in this scope

   SPCR |= _BV(SPE);

               ^~~

C:\Users\...\Documents\Arduino\sketch_jan12a\sketch_jan12a.ino:13:15: note: suggested alternative: 'SPI'

   SPCR |= _BV(SPE);

               ^~~

               SPI

sketch_jan12a:13:11: error: '_BV' was not declared in this scope

   SPCR |= _BV(SPE);

           ^~~

C:\Users\...\Documents\Arduino\sketch_jan12a\sketch_jan12a.ino:13:11: note: suggested alternative: '_B'

   SPCR |= _BV(SPE);

           ^~~

           _B

C:\Users\...\Documents\Arduino\sketch_jan12a\sketch_jan12a.ino: At global scope:

sketch_jan12a:22:5: error: expected constructor, destructor, or type conversion before '(' token

 ISR (SPI_STC_vect)

     ^

exit status 1

expected constructor, destructor, or type conversion before '(' token

facchinm, 2021-01-13:
Hi @marcoumlaut ,
all arduino libraries are portable as soon as you don't use any architecture specific code in  your sketch.
In this case, `SPDR` and friends are registers that only exists in AVR microcontrollers. 
If you need to implement an SPI slave there isn't a wrapper yet for Nano33BLE but these APIs https://os.mbed.com/docs/mbed-os/v6.6/apis/spislave.html can be used

closed 2021-01-13 by facchinm

marcoumlaut, 2021-01-13:
But when I compile it I get the error description:

src\main.cpp:4:1: error: 'SPISlave' does not name a type; did you mean 'SPIName'?
 SPISlave device(D12, D11, D13, D10); // mosi, miso, sclk, ssel
 ^~~~~~~~
 SPIName
src\main.cpp: In function 'int main()':
src\main.cpp:8:5: error: 'device' was not declared in this scope
     device.reply(0x00);              // Prime SPI with first reply
     ^~~~~~
src\main.cpp:8:5: note: suggested alternative: 'dev_t'
     device.reply(0x00);              // Prime SPI with first reply
     ^~~~~~
     dev_t
*** [.pio\build\nano33ble\src\main.cpp.o] Error 1

marcoumlaut, 2021-01-14:
I've found out there are inactive regions in the SPISlave.h library. Why is the whole code inactive there?

manchoz, 2021-02-16:
@marcoumlaut because SPISlave is not supported on nRF52840 on MbedOS.