arduino/ArduinoCore-mbed #119: mbedOS crashes trying to run examples from Tensorflow 2.1.0-alpha

mbanzi, 2021-01-13:
Trying to run examples from the latest version of the TensorFlow library crashed mbedOS

mbanzi, 2021-01-26:
update. I tried the "hello_world" on Portenta M7 Core and the code compiles ok but the code crashes when run. TF Lite 2.4.0 Alpha , mbedOS core 1.3.1

facchinm, 2021-01-26:
@mbanzi the crash on Portenta is due to the example trying to `analogWrite()` on `LED_BUILTIN` which is not PWM capable. mbed core crashes in this case instead than falling back to `digitalWrite()` due to an assert in mbed itself.
I'm wondering which is the best way to proceed, if "masking" the assert or changing the example for Portenta

mbanzi, 2021-01-26:
Thanks for pointing that out. :) In the other Arduino cores trying to PWM a non PWM pin we just fall back to digitalWrite 
is this something we can/want to do with Portenta as well? We can always modify the example adding some ifdef for Portenta

closed 2021-01-26 by silvanocerza