arduino/ArduinoCore-mbed #124: pins_arduino.h macros unavailable during library discovery phase

tttapa, 2021-01-28:
## Problem

The `pins_arduino.h` header is not included by `Arduino.h` whenever `ARDUINO_LIBRARY_DISCOVERY_PHASE == 1`. As a result, sketches and libraries that rely on the macros defined in `pins_arduino.h` fail to build during the library discovery phase.

## Reproduction

```cpp
#ifndef NUM_DIGITAL_PINS
#error "NUM_DIGITAL_PINS not defined"
#endif 

void setup() {}
void loop() {}
```
```
Using board 'nano33ble' from platform in folder: /home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1
Using core 'arduino' from platform in folder: /home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1
Detecting libraries used...
/home/pieter/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE/defines.txt @/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/cores/arduino -I/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE -I/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/cores/arduino/api/deprecated -iprefix/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/cores/arduino @/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_584601/sketch/NUM_DIGITAL_PINS_test.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Error while detecting libraries included by /tmp/arduino_build_584601/sketch/NUM_DIGITAL_PINS_test.ino.cpp
Generating function prototypes...
/home/pieter/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g -Os -nostdlib @/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE/defines.txt @/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE/cxxflags.txt -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/cores/arduino -I/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE -I/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/cores/arduino/api/deprecated -iprefix/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/cores/arduino @/home/pieter/.arduino15/packages/arduino/hardware/mbed/1.3.1/variants/ARDUINO_NANO33BLE/includes.txt /tmp/arduino_build_584601/sketch/NUM_DIGITAL_PINS_test.ino.cpp -o /tmp/arduino_build_584601/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
NUM_DIGITAL_PINS_test:2:2: error: #error "NUM_DIGITAL_PINS not defined"
 #error "NUM_DIGITAL_PINS not defined"
  ^~~~~
```
This code does compile correctly on other Arduino Cores, such as AVR, SAMD, Teensy, ESP32, etc. It also works in version 1.8.9 of the IDE because it doesn't define `ARDUINO_LIBRARY_DISCOVERY_PHASE`.

## Use cases
My specific use case is having a library conditionally include code based on the number of pins a specific board has, using the `NUM_DIGITAL_PINS` macro.  
The `pins_arduino.h` header contains mostly macros, so in my opinion it should be possible to use these macros in `#ifdef`s, which is not the case right now.

## Proposed solution
Always include `pins_arduino.` in `Arduino.h`, regardless of the value of `ARDUINO_LIBRARY_DISCOVERY_PHASE`.

facchinm, 2021-01-28:
Hi @tttapa, thanks for spotting this! Rather than modifying the core I'd prefer to fix the builder. The issue is here https://github.com/arduino/arduino-cli/blob/master/legacy/builder/gcc_preproc_runner.go#L60 , since `prepareGCCPreprocRecipeProperties` is used when discovering libraries and when we add the prototypes.
Moreover, https://github.com/arduino/ArduinoCore-mbed/pull/106 would make the LIB_DISCOVERY macro obsolete, so fixing both sides could be ideal.
@cmaglie what do you think about this?

tttapa, 2021-06-24:
Any updates on this? I still get the same error using Arduino Mbed OS Nano Boards v2.2.0 using version 1.8.15 of the IDE and version 0.18.3 of the arduino-cli.