arduino/ArduinoCore-mbed #145: Fastest way to share data between cores (SRAM3/4 usage)?

MatInGit, 2021-02-23:
Posted already on the [portenta forum](https://forum.arduino.cc/index.php?topic=728683.0) but also posting here:

I have been trying to create a shared structure between the M7 and M4 cores. I have found this post describing a couple of implementations of a tx/rx buffer. I have tried a number of the implementations outlined in the post, however, they did not work, or worked for only one loop. 

Here is a number of things I have tried:

* Create a uint8_t flag variable for when the buffer is being written to (same as in the linked post). This didn't cause a crash but ceased execution after a single loop. (i.e it passed the variable back and forth once)
* Use `PlatformMutex' instead of the uint8_t and use .lock and .unlock() - This caused a crash on start up.
* Tried loop instead of thread - did not alter anything.
* Allocated at different SRAM addresses (SRAM3/4) same result as within structs.
* Tried creating the flag variables separate from the struct that is shared between the cores.

I would use the RPC, but it is simply too slow for what I want to do. 

M7 code:

```cpp
#include "mbed.h"
#include "Arduino.h"

using namespace mbed;
using namespace rtos;
Thread txrx;
struct shared_data
{
  PlatformMutex flag0;
  PlatformMutex flag1;
  int M4toM7;
  int M7toM4;
};

int localm7m4 = 0;
int localm4m7 = 0;

static struct shared_data * const xfr_ptr = (struct shared_data *)0x30040000;

void setup() {
  
  delay(3000);
  LL_RCC_ForceCM4Boot();
  Serial.begin(2000000);
  txrx.start(callback(txrxThread));
}
  
void txrxThread() {
      //flag1.lock(); or xfr_ptr -> flag1.lock(); ?
      Serial.println("Writing M7 to M4");
      digitalWrite(LEDR, LOW);
      localm7m4 = rand() % 99;
      xfr_ptr -> M7toM4 = localm7m4;
      //flag1.unlock(); or xfr_ptr -> flag1.unlock(); ?
      
      //flag0.lock(); or xfr_ptr -> flag0.lock();
      Serial.println("Reading M4 to M7");
      digitalWrite(LEDB, HIGH);
      localm4m7 = xfr_ptr -> M4toM7;
      //flag0.unlock(); or xfr_ptr -> flag0.unlock(); ?

      Serial.println("Loop!");
  }

  void loop(){
    delay(100);
}
```

M4 code:

```cpp
// inter-core buffers
#include "mbed.h"
#include "Arduino.h"

using namespace mbed;
using namespace rtos;

struct shared_data
{
  PlatformMutex flag0;
  PlatformMutex flag1;
  int M4toM7;
  int M7toM4;
};

int localm7m4 = 0;
int localm4m7 = 0;

// pointer to shared_data struct (inter-core buffers and status)
static struct shared_data * const xfr_ptr = (struct shared_data *)0x30040000;
Thread txrx;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  
  txrx.start(callback(txrxThread));
}

void txrxThread() {
      //xfr_ptr -> flag0.lock();
      digitalWrite(LEDB, LOW);
      localm4m7 = rand() % 99;
      xfr_ptr -> M4toM7 = localm4m7;
      //xfr_ptr -> flag0.unlock();
      
      //xfr_ptr -> flag1.lock();
      digitalWrite(LEDR, HIGH);
      localm7m4 = xfr_ptr -> M7toM4;
      //xfr_ptr -> flag1.unlock();
  }

void loop()
{

}
```

Any pointers to what else I can try would be greatly appreciated

facchinm, 2021-02-23:
Hi @MatInGit ,
if you don't have strict low latency needs you could use RPC to signal the other core that the buffer is ready (like [this](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/RPC/SerialRPC.h#L48) but `on_write` only unlocks a semaphore the other core is waiting on.
A tricky thing is that you won't be able to specify the absolute address of the buffers; you can, however, assign a big buffer (to be partitioned) in a core (based on the position if the structures created by RPC / openamp) and share its address to the other core, always by RPC.
Let me know if you need any other hint or if it's enough :wink:

MatInGit, 2021-02-24:
Hi @facchinm, 
Thank you for your answer, I think it has put me on the right path to getting to where I want to get but I am still having a few issues. I can pass the address and read the data directly from the other core. but as soon as I start using any bound RPC functions the M4 core ceases to do anything (it does not throw an error). In general use of any bound functions seems to cause the caller to stop. I am not sure why that is? 

`RPC1.print()` seems to work without any issues hence I was able to use it to pass the address to the M7 core. I was quite surprised to find out that the M7 can read the data from the M4 address. I haven't tried yet but I think if the RPC bindings prove too difficult to get to work I can try "hardcoding" the address of the M4 array in the M7 code and use the `RPC.print()` to signal when the array is ready for the readout. 

Another note about the code below is that after being read out once the values don't update, even though I made the `jjj[3]` element update every loop. 

I will update the post if I get any of the versions I have outlined above to work. Meanwhile, if you see any obvious mistakes I am making I would appreciate the feedback and some more pointers :).

M7 code:
```c++
#include "RPC_internal.h"

int rcv = 0;
int jjj;
int rx_p = (int) &jjj;
int rx_buffer[1600];

int getRcv() {
  return rcv;
}

void setup() {
  RPC1.begin();
  RPC1.bind("getRcv",getRcv);
  LL_RCC_ForceCM4Boot();
  Serial.begin(115200);
}

void loop() {
  char buf[16];
  int i = 0;
  while (RPC1.available() ) {
    buf[i] =  RPC1.read();
    i++;
    rcv = 1;
  }
  if (rcv == 1) {
    rcv = 0;
    rx_p = String(buf).toInt();
    int *p = (int*) rx_p;
    
    for (int j = 0;j<1600;j++){
      rx_buffer[j] = *(p + j);
      }
      
    Serial.println(rx_buffer[0]);
    Serial.println(rx_buffer[1]);
    Serial.println(rx_buffer[2]);
    Serial.println(rx_buffer[3]);

  }

  delay(50);
}
```

M4 code:
```c++
#include "RPC_internal.h"

int jjj[1600];
int i = 0;
int p = (int) &jjj[0];

void setup() {
  
  RPC1.begin();
  for (int j = 0; j < 1600; j++) {
    jjj[j] = rand() % 99;
  }
  // constant markers to veirfy I am reading the correct area
  jjj[0] = 999;
  jjj[1] = 111;

}

void loop() {
  //auto res = RPC1.call("getRcv").as<int>(); // when this is uncommented this core stops
  
  delay(100);
  
  digitalWrite(LEDB, LOW);

  
  delay(50);
  if (true){ // want to have if (res == 0)
    jjj[3] = 444 + rand()%3; // Check if data is being updated
    RPC1.print(p, DEC);
    digitalWrite(LEDB, HIGH);
  }
}
```

MatInGit, 2021-03-02:
Hi, I managed to solve the issue by using the RPC mechanism to signal when the data is readable. Thank you for the pointers, I will upload the code when I clean it up a bit and add a link here for future reference for anyone else looking at the issue.

closed 2021-03-02 by MatInGit

facchinm, 2021-03-02:
Thank you @MatInGit , happy the hints were useful :wink: 
I'm ccing @sebromero because the example you are writing could really become a full fledged tutorial if you wish!

sebromero, 2021-03-02:
👀 Thanks for the ping. @MatInGit Are you happy with the performance in the end? If you would like to collaborate on a tutorial that could be hosted on our website https://www.arduino.cc/pro/tutorials/portenta-h7 let me know 👍

MatInGit, 2021-03-03:
@sebromero The performance was a little bit out of spec for what I wanted to do, but that is only because I am trying to use it for real-time applications, I think it should be fine for most use cases. I have found a faster way, using the STM hardware semaphores (which I will add to the linked git). 

I would be interested in contributing to the tutorials, I have been using them since I started using the Portenta!

Here is the [link](https://github.com/MatInGit/PortentaRPCMemory.git), let me know if there are any obvious changes that need to be made. 😄

sebromero, 2021-03-06:
@MatInGit Awesome 👌  Do you mind if I fetch your code and make some suggestions?
In the mean time you can have a look at our repository that hosts the tutorials:
https://github.com/arduino/arduino-pro-content

If you want to start with a draft, here's the template:
https://github.com/arduino/arduino-pro-content/blob/main/content/tutorials/portenta-h7/template/content.md

MatInGit, 2021-03-09:
Sounds good! One obvious change I had realized after uploading the code was that the passed variable should be a structure pointer.

Stenapan, 2023-11-29:
@MatInGit Hi I'm trying to resolve this exact issue and I am having trouble getting any reasonable speed with RPC and trying to do this without the HSEM seems to onlcy crash the board. At least I think that's the reason it crashes. You mentioned above that you have had success with the hardware sempahore and would add that code to the git repo you linked. I can't, however, find any references to HSEM in your code. Is it possible you could upload a simple example or give any hints at where one can find good info about the HSEM? It seems the portenta is still (2 years later) largely undocumented if you want to do anything more advanced.