arduino/ArduinoCore-mbed #147: Odd measurements with micros on Portenta H7 M7

sebromero, 2021-02-25:
On Portenta H7 `micros()` supposedly has a resolution of 1us running on the M7. When measuring this with a sketch it produces odd results. Something seems to be broken either with `delayMicroseconds` or `micros`.

```cpp
unsigned long start, end;

void setup() {
  Serial.begin(115200);
  while(!Serial);
  
  for(int i = 1; i <= 64; ++i){
    start = micros();
    delayMicroseconds(i);
    end = micros();
    Serial.print("Delay: " + String(i) + " measured: ");
    Serial.println(end - start); 
  }
}

void loop() {}
```
Produces the following output:
```
Delay: 1 measured: 0
Delay: 2 measured: 0
Delay: 3 measured: 0
Delay: 4 measured: 0
Delay: 5 measured: 32
Delay: 6 measured: 0
Delay: 7 measured: 0
Delay: 8 measured: 0
Delay: 9 measured: 0
Delay: 10 measured: 0
Delay: 11 measured: 31
Delay: 12 measured: 31
Delay: 13 measured: 31
Delay: 14 measured: 31
Delay: 15 measured: 31
Delay: 16 measured: 31
Delay: 17 measured: 31
Delay: 18 measured: 31
Delay: 19 measured: 31
Delay: 20 measured: 31
Delay: 21 measured: 31
Delay: 22 measured: 31
Delay: 23 measured: 31
Delay: 24 measured: 31
Delay: 25 measured: 31
Delay: 26 measured: 31
Delay: 27 measured: 31
Delay: 28 measured: 32
Delay: 29 measured: 32
Delay: 30 measured: 31
Delay: 31 measured: 31
Delay: 32 measured: 31
Delay: 33 measured: 31
Delay: 34 measured: 31
Delay: 35 measured: 32
Delay: 36 measured: 31
Delay: 37 measured: 31
Delay: 38 measured: 31
Delay: 39 measured: 63
Delay: 40 measured: 32
Delay: 41 measured: 32
Delay: 42 measured: 31
Delay: 43 measured: 62
Delay: 44 measured: 63
Delay: 45 measured: 63
Delay: 46 measured: 62
Delay: 47 measured: 62
Delay: 48 measured: 32
Delay: 49 measured: 62
Delay: 50 measured: 62
Delay: 51 measured: 32
Delay: 52 measured: 62
Delay: 53 measured: 62
Delay: 54 measured: 63
Delay: 55 measured: 62
Delay: 56 measured: 62
Delay: 57 measured: 63
Delay: 58 measured: 62
Delay: 59 measured: 63
Delay: 60 measured: 62
Delay: 61 measured: 62
Delay: 62 measured: 63
Delay: 63 measured: 62
Delay: 64 measured: 63

```

hpssjellis, 2021-02-25:
@sebromero isn't a microsecond delay much too short for any kind of Serial.println? I would do a count down test from some much bigger number and see when Serial.println starts messing up. Or save the values into an array and print several at one time.

sebromero, 2021-02-25:
@hpssjellis The measurement is relative to the start time and starts over every time in the loop so any Serial printing shouldn't affect the measurement.

sebromero, 2021-02-26:
@facchinm When you have time, can you shed some light on this? 🙏

facchinm, 2021-03-09:
@sebromero I investigated the issue a bit and the real cause are these lines https://github.com/arduino/ArduinoCore-mbed/blob/master/cores/arduino/wiring.cpp#L29-L33 .
So, by using the LPTIMER for both `micros()` and `millis()` (the advantage is that it doesn't stop during sleep) we end up here https://github.com/arduino/mbed-os/blob/portenta-mbed-6.2.0/targets/TARGET_STM/lp_ticker.c#L136 . So the resolution is roughly 1 / 32768 which correcponds to 30.5 us as per your test.
Do you advise using an high resolution timer for `micros()` to keep consistency with other platforms?

facchinm, 2021-03-18:
Fixed by #157

closed 2021-03-18 by facchinm