arduino/ArduinoCore-mbed #158: Question: enable crash report for Arduino Nano 33 ble

tylerhartwig, 2021-03-10:
Hi,

From what I understand, mbed should generate a crash report and send it over stdio when a crash occurs:
[Documentation](https://os.mbed.com/docs/mbed-os/v5.14/tutorials/analyzing-mbed-os-crash-dump.html)

Is there a way to enable this from the Arduino build? I've over-ridden the stdout to be over USB, however I get no crash report when the mbed OS crashes.

Mostly looking for a good awy to debug mbed os crashes on this board.

Thanks,
Tyler Hartwig

sebromero, 2021-03-12:
Hi @tylerhartwig, you should be able to read the stack trace by connecting another board to the hardware Serial1 RX TX pins on the board.

facchinm, 2021-03-12:
@tylerhartwig the redirection of stdout to USB only works when things go well :slightly_smiling_face: 
In case of crash, the system relies on very low level prints that can only be performed on standard UART. As @sebromero said, in Nano33BLE the default UART is routed to pin TX1, while on Portenta it's on pin14 (TX for the MKR format)

closed 2021-03-12 by facchinm

tylerhartwig, 2021-03-12:
Connecting an external UART to USB did it for me, thanks!