arduino/ArduinoCore-mbed #161: analogReference() not implemented for Portenta H7

sebromero, 2021-03-12:
Using `analogReference()` in a sketch throws a linker error. Looking at the code, there seems to be only an implementation for the Nano33 BLE variant but not for the H7.

sebromero, 2021-03-12:
@facchinm ☝️

malmr, 2022-04-22:
Unfortunately still not working

facchinm, 2022-04-22:
To set the analog reference in STM32H7 boards there's no need to call any function, just connect the external reference to AREF pin and it will be used. I'm going to add an empty function with a warning at compile time to make the linker happy.

romainreignier, 2022-04-25:
It would be nice to be able to use the `INTERNAL` reference voltage of the stm32 to be able to do some analog readings without needing an external reference. The built-in pull-up on the AREF pin provides analog values that are not stable as stated several times in the forum.

sebromero, 2022-06-01:
@romainreignier As far as I know, an external voltage reference is required for stable readings. When not connecting it, it automatically uses the internal voltage reference via weak pull-ups. Correct me if I'm wrong @facchinm

closed 2022-06-01 by sebromero