arduino/ArduinoCore-mbed #169: State of merging mbed-os patches upstream

maxgerhardt, 2021-03-23:
It is my understand that this Arduino here takes mbed-os version

https://github.com/arduino/ArduinoCore-mbed/blob/a2c06d768f5ebb6821ae6505b2032ee58f4ef70d/cores/arduino/mbed/platform/mbed_version.h#L34-L48

(for the latest 1.3.2 release, in development right now it's 6.8.0), and applies certain patches stored in the [patches](https://github.com/arduino/ArduinoCore-mbed/tree/1.3.2/patches) folder. 

These patches seem quite substantial, with a few saying e.g. "Enabling WiFi", "BLE", etc. When I pick e.g. the patch [`0066-STM32-USB-allow-dynamic-USB-speed-configuration-base.patch`](https://github.com/arduino/ArduinoCore-mbed/blob/master/patches/0066-STM32-USB-allow-dynamic-USB-speed-configuration-base.patch) which adds a function `getUSBSpeedFromBlootloader()` in `targets/TARGET_STM/USBPhy_STM32.cpp`, that is nowhere to be found in [the upstream file](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/USBPhy_STM32.cpp). 

What is the state of these patches regarding upstream merging? Can a pure mbed-os (e.g. 6.2.0 or 6.8.0) be used with the e.g. Portenta H7 or Nano 33 BLE boards, or will it absolutely not work without them?

facchinm, 2021-03-23:
Hi @maxgerhardt ,
the patches come from this branch https://github.com/arduino/mbed-os/commits/rp2040 , we are going to mainline the relevant parts soon (after a cleanup)

pennam, 2021-03-29:
Hi @maxgerhardt regarding your last question most of the patches for PORTENTA_H7 were already submitted here the complete list https://github.com/ARMmbed/mbed-os/pulls?q=facchinm. As you can see 4 of them still need to be merged. Despite this  if you want to use a pure mbed-os 6.8.0 (without BLE and WiFi) should work on your board.

For the NANO_33_BLE if i didn't miss something everything but this https://github.com/ARMmbed/mbed-os/pull/14172 has been mainlined starting from mbed-os 6.7.0

closed 2021-07-09 by facchinm