arduino/ArduinoCore-mbed #174: Recent changes to USB Serial ports in Arduino.h breaks code using mbed USBSerial redirection

tjpetz, 2021-04-02:
This is not critical as there is a simple, and I think now better, approach.  However, it might be worth mentioning in a release note for the next release.

In mbedOS the printf() function goes to the default console/serial port.  In release 1.3.2 one could use the following technique to redirect to the USB serial port and thus see printf() messages when using the IDE Serial Monitor.

```
#include <mbed.h>

USBSerial mySerial(false);

// Redirect the console to the USB serial port.
mbed::FileHandle *mbed::mbed_override_console(int fd) {
  return &mySerial;
}
```

In the current commit <tbd> this code generates a compilation error.

I see you've added the mbed::FileHandle operator to the Arduino Serial code.  I think this is a great addition.  As the override to get printf() working is now just:

```
#include <mbed.h>

// Redirect the console to the USB serial port.
mbed::FileHandle *mbed::mbed_override_console(int fd) {
  return Serial;
}
```

While this is a great change, it breaks a few examples on the web.  Please include at least a mention in the release notes for the next release about this change.

facchinm, 2021-04-06:
Hi @tjpetz ,
the commit fixing `REDIRECT_STDOUT_TO` macro is https://github.com/arduino/ArduinoCore-mbed/commit/5dd679e9f8db0e0087a2f664c1556d003df12cc1 . I think most of the examples on the web use the macro and not `mbed_override_console`. 
Said that, it is indeed a breaking change and should be documented accordingly.
@per1234 any suggestion?

per1234, 2021-04-06:
I suggest clearly documenting it in the GitHub release description:
https://github.com/arduino/ArduinoCore-mbed/releases
You might add a dedicated section for breaking changes, similar to how the Arduino CLI changelog is formatted:
https://github.com/arduino/arduino-cli/releases/tag/0.18.0

In the case of this sort of API which is not part of the standard Arduino API, I think clearly documenting it in the repository is sufficient.

I'm a proponent of following semver strictly in order to communicate breaking changes to the users, but I'm also aware that a lot of extra significance tends to be put on the major version, which can make this difficult.

tjpetz, 2021-04-06:
@facchinm thanks, I hadn't noticed the macro before.  For context, I've been working on better understanding mbedOS and in particular using it's BLE capabilities to unlock some of the more advanced BLE capabilities on the Nano 33 BLE boards.  So I've been coming to this mostly from the mbedOS documentation which is were they discuss mbed_override_console.

I agree the macro is a better approach for Arduino native development to better hide mbedOS.

@per1234 I agree covering in the release notes as suggested should cover it.

One additional question.  It seems that this would prevent the use of the mbedOS native USBSerial class.  Is that a design intent, or perhaps an unintended consequence of running Arduino on top of mbedOS?  (Note, this is really critical, just more of a nice to know.)

facchinm, 2021-04-07:
@tjpetz got it :slightly_smiling_face: 
You can still use the mbed USBSerial but its header must be included explicitely via `#include <USBSerial.h>` .
Its removal from default includes allowed us to get rid of the depencency from `Callback.h` (which is by far the heaviest header I've ever seen in the embedded world and was slowing down compilation considerably).

tjpetz, 2021-04-07:
@facchinm Thanks for the response.  I'll probably stick mostly with the Arduino implementation and good to know I can use the mbedOS implementation if necessary.

Yikes Callback.h certainly seems to address every possible variant on the types being fed into a callback.  I can certainly see how it would significantly impact compile times.

Thanks again for all the great work you and the team are doing!

facchinm, 2021-04-08:
We just went live, core version is 2.0.0 (release note here https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.0.0 ); the actual problem and the workaround is described in the final section :slightly_smiling_face:

closed 2021-04-08 by facchinm