arduino/ArduinoCore-mbed #176: arduino:dfu-util@0.9.0-arduino2 not available for arm64

arpit6232, 2021-04-05:
[image]

Unable to install arduino:mbed core on arm64 architectures. How to explicitly install this.

facchinm, 2021-04-06:
Hi @arpit6232 ,
we are going to fix the missing package with the upcoming core (which will bundle `dfu-util-0.10.0-arduino1` from https://github.com/arduino/dfu-utils-cross

arpit6232, 2021-04-08:
What would be the apprximate sprint or Date around which this would be integrated. ?

facchinm, 2021-04-08:
We just went live, core version is `2.0.0` (release note here https://github.com/arduino/ArduinoCore-mbed/releases/tag/2.0.0 )