arduino/ArduinoCore-mbed #179: Board nano33ble (platform mbed, package arduino) is unknown

witzed1, 2021-04-06:
I have been using a new Nano 33 BLE Sense for several weeks.  It has suddenly stopped working.  By that I mean that I get this error whenever I download any sketch.  I have even resorted to trying to download Blink without success.  Compile fails with this error, Board nano33ble (platform mbed, package arduino) is unknown.  I have installed all of the applicable updates from board manager specific to Nano sense.  What happened?

sbailey128, 2021-04-06:
Try uploading the boot loader again. Same thing happened to me...boot loader got corrupt and I could not do anything after that.

facchinm, 2021-04-06:
It's not about the bootloader but most likely you have two packages installed and/or an old version of the IDE. Would you mind sharing the verbose output of the compilation?
I'm adding some debugging info just in case it's the first case: navigate to `$arduino15` folder (`C:\yourusername\AppData\Local\Arduino15` if on Windows, `~/.arduino15` on Linux, `~/Library/Arduino15` on Mac), then continue to `packages/arduino/hardware/mbed/`. If there are two folders, delete the older one and then retry compiling.

witzed1, 2021-04-06:
It seems like bootloader is halted rather than corrupt.  I pusher the reset
button twice rapidly and the pin 13 led started to pulse slowly.  I the
recompiled and downloaded ant it worked.  I found this trick in a post but
nothing associated with the arduino authority.  I find this very strange.

witzed1, 2021-04-06:
I followed this path (somewhat corrected) and under arduino15 I found one
directory labeled 1.3.2.  under that are two directories, nano33ble and
potenta h7.  I suppose these are the two that you mean.  In my search for a
solution to my problem I found reference to potenta.  Both appear to
contain bootloader code.  I going to delete potenta and see what happens.

sebromero, 2021-04-07:
>  I pusher the reset
button twice rapidly and the pin 13 led started to pulse slowly.  I the
recompiled and downloaded ant it worked.

@witzed1 If there is something in your sketch that makes the board immediately crash by the means of a runtime error, it can no longer respond and you won't be able to upload a sketch anymore. Double pressing the reset button puts it into bootloader mode where it just waits for a new sketch to be uploaded without running the previous sketch. This is the official suggestion of dealing with this.

witzed1, 2021-04-08:
Thanks this process works well.  I couldn't find this official suggestion is any arduino documentation.

closed 2021-04-09 by sebromero