arduino/ArduinoCore-mbed #182: mbed tools not found during mbed compile

tylerhartwig, 2021-04-09:
Hi,

I'm attempting to build via the README, and am getting the following error. (Line 135)

```
Compiling Mbed Application...[mbed] ERROR: The mbed tools were not found in "/private/tmp/mbed-os-program". 
       You can run "mbed deploy" to install dependencies and tools.
```

I am building on macOS -- I know that's not officially supported, I made some modifications to get it to this point. Is this related to building on macOS, or a problem with my mbed setup?

pennam, 2021-04-09:
Hi, the error should be due to there is no global configuration for mbed tools. To solve this you can move into /private/tmp/mbed-os-program directory and type

pennam, 2021-04-09:
Hi, the error can be due to mbed is not finding mbed-os sources but just mbed-os.lib file. Make sure that running mbed-os-to-arduino script you are pointing the source dir of mbed-os and not to the parent folder. An mbed program folder structure looks like this:
```
mbed-program
|_ mbed_app.json
|_ mbed_os   <- mbed-os-to-arduino script use this folder
|   |_ ...
|_ mbed_os.lib
|_ ...
```

tylerhartwig, 2021-04-09:
@pennam maybe I understood what mbed-os-to-arduino did.

Should I be pointing to the mbed-os repo? or to an existing mbed application?

I thought the script compiled the ArduinoCore-mbed library and needed a reference to mbed-os to do so

tylerhartwig, 2021-04-11:
@pennam 

After some more debugging I made a lot more progress on this. However, it seems that a file is missing that was supposed to be created somewhere in the process.

```
Generating libs..../mbed-os-to-arduino: line 201: ./BUILD/ARDUINO_NANO33BLE/GCC_ARM/.link_options.txt: No such file or directory
```

Do you know what is supposed to generate this?

tylerhartwig, 2021-04-11:
The above issue occured when the compile failed -- I was looking at the wrong error.

For anyone interested in running the latest version on macOS, these are the changes I needed:

- Use/install latest bash (anything over 4.x I believe should be fine)
- Use gcp in the script rather than cp -- can be installed with `brew` under the package `coreutils`
- Install `jq` also with `brew`

closed 2021-04-11 by tylerhartwig